nfc-tools/libnfc #291: pn53x_initiator_select_passive_target with hard coded timeout = 0

RobertMitschke, 2015-04-14:
The function pn53x_initiator_select_passive_target has a timeout of 0 hard coded. It is the only publicly accessible function for this task. Since the timeout is set to 0 and not -1 the timeout cannot be modified using a property either. 

I would ideally welcome an API alternative where I can set the timeout explicitly or a way to set the timeout using a property.

I right now have to do a modification of the source as the only option available to me. Ideally I would like to use a vanilla version of libnfc and would therefore kindly ask for this timeout to be configurable for me.

neomilium, 2015-08-19:
Have you any patch to handle this correctly ?

masterbraind, 2015-09-09:
Hi, I experienced the same problem. 

If I get a TAG with nfc_initiator_poll_target() then remove the physical TAG before call nfc_initiator_select_passive_target() it blocks the thread until you place the same previous phisical TAG.

This happens very often when the user present and remove the card too fast.

My quick fix was HARDCODE a 50mS Time Out. Now I have no more problems in the way I implemented the detection and read.

int
pn53x_initiator_select_passive_target(struct nfc_device *pnd,
                                      const nfc_modulation nm,
                                      const uint8_t *pbtInitData, const size_t szInitData,
                                      nfc_target *pnt)
{
  return pn53x_initiator_select_passive_target_ext(pnd, nm, pbtInitData, szInitData, pnt, 50);
}

I think at least will be good to hardcode a Time Out i.e. 150mS as the constant used for the Polls.