nfc-tools/libnfc #294: update readme: See the file 'INSTALL' for configure, build and install details.

ozett, 2015-05-12:
"See the file 'INSTALL' for configure, build and install details."...

but there is not any INSTALL....

and configure give me an error:

" ./configure
-bash: ./configure: Datei oder Verzeichnis nicht gefunden
root@raspberrypi:/usr/src/libnfc-master# mc
"

smortex, 2015-05-13:
I'll take it!

We are in the process of modernizing the nfc-tools project infrastructure, and with the switch to GitHub with the ability to have awesome markdown info in the repository, build instructions from http://nfc-tools.org/index.php?title=Libnfc#Development_version will eventually reach the root of the repository.

neomilium, 2015-05-13:
AFAICR, INSTALL file is generated by "autoreconf -is" command.

ozett, 2015-05-13:
thanx, i will try. (if ./configure & make) only will do....

another thing, while i am trying to install: lot of websources have links to version 1.7.0 and 1.7.1
but you moved to github, but there are no version branches. 
it is also difficult to find the right link to a version package.
i went from http://nfc-tools.org/index.php?title=Libnfc to bintray-repository, while here on github the master exists... better if it could it all in one place...? confusion for noobs like me...(and other)
thankx a lot...

smortex, 2015-05-13:
On Wed, May 13, 2015 at 03:40:59AM -0700, ozett wrote:
> another thing,
In the future, please open new issues in order to help us tracking each
problem more easily (1 problem per issue).  Thanks!

> while i am trying to install: lot of websources have links to version
> 1.7.0 and 1.7.1 but you moved to github, but there are no version
> branches. it is also difficult to find the right link to a [...]

Good point, I attached the few tarballs I had on my personal mirror to
the tags:

* libnfc-1.3.0.tar.gz
* libnfc-1.3.2.tar.gz
* libnfc-1.3.3.tar.gz
* libnfc-1.4.0.tar.gz
* libnfc-1.4.2.tar.gz
* libnfc-1.7.1.tar.bz2

Unfortunately, I don't have any other tarball I can trust.

ozett, 2015-05-13:
i  had a brandnew "2015-02-16-raspbian-wheezy.img" image, naked.
apt-get update &&  apt-get install autoconf libtool libpcsclite-dev libusb-dev
downloaded the master-branch today,

autoreconf -is
./configure --prefix=/usr --sysconfdir=/etc
make
make install

  sudo mkdir /etc/nfc/devices.d
  sudo cp /contrib/libnfc/pn*.sample /etc/nfc/devices.d/pn*.conf

  sudo mkdir /etc/nfc
  sudo cp libnfc.conf.sample /etc/nfc/libnfc.conf

...!!tadaahh!! it worked ! 

thanks a lot
( now i try nfc-eventd, and possibly come with a new issue... ;-))))

closed 2015-07-22 by neomilium