nfc-tools/libnfc #298: API breakage mitigation

fuzxxl, 2015-06-05:
Hello!

I'm the author of the Go bindings for the libnfc and I saw that you plan to change parts of the API for the upcoming release. Could you please reconsider this? API breakages are a huge PITA for each user of your library (as each user needs to change his code and the code can't work with both the new and the old library version without some preprocessing tricks) and I want to avoid them wherever possible.

Could you consider to add a new call `nfc_get_supported_baud_rate_with_mode` that implements the new behaviour, leaving `nfc_get_supported_baud_rate` unchanged? That's at least what I plan to do in my wrapper if you don't want to.

smortex, 2015-06-08:
Change was introduced in e57dc8a746d5d4257a70f87aaae80bab87f20b8a by @doegox. He may want to comment.

doegox, 2015-06-08:
I don't like breaking the API neither but in this case nfc_get_supported_baud_rate() was just not returning the proper list of supported baud rates.

I made a branch with a proposal:
https://github.com/nfc-tools/libnfc/tree/nfc_get_supported_baud_rate_proposal

There nfc_get_supported_baud_rate() returns baud rates for initiator mode (which is the one people use 99% of the time) and I added a
nfc_get_supported_baud_rate_target_mode()

What do you think?
Phil

fuzxxl, 2015-06-08:
That looks better. I can live with that. Does that change what `nfc_get_supported_baud_rate` returns with respect to what it returned in 1.7.0?

doegox, 2015-06-08:
Yes, as shown in example given in e57dc8a746d5d4257a70f87aaae80bab87f20b8a this is now more correct (for initiator mode at least).
E.g. before, for a PN533 for TypeA it was returning only 106 and now it's returning 847, 424, 212 and 106.

fuzxxl, 2015-06-08:
That's ok for me. I hope this decision is going to be well-received with users.

smortex, 2015-06-08:
:+1:

Does it makes sense to provide 2 functions (target mode / initiator mode) and deprecate the existing one with a ` __attribute__ ((deprecated));` to help users fix their code?

fuzxxl, 2015-06-08:
That would be useful but please wrap `__attribute__((deprecated))` in a macro like this so the code compiles with compilers other than gcc:

    #ifdef __GNUC__
    # define DEPRECATED __attribute__((deprecated))
    #else
    # define DEPRECATED
    #fi

There are some other problems with respect to portability (like the needless use of `#pragma pack`)  but fixing them would inhibit compatibility so there's little one can do.

fuzxxl, 2015-06-08:
You could also just make a function with an extra mode parameter as originally planned instead. If you give this function a new name (and not `nfc_get_supported_baud_rate_for_mode`), keeping the original function intact, then that would be fine, too, and the best choice in my opinion.

doegox, 2015-06-08:
A new name like nfc_get_really_supported_baud_rate_because_the_other_one_is_bogus() ?
:-)
If we want to keep the original one as such, I prefer to make some sense of it and make clear why there is a second almost similar function in the API... because results may be wrong for target mode. Therefore the proposal in the branch.

fuzxxl, 2015-06-08:
Yeah, something like that. If you think the proposal in the branch is ok for you, than go for it. It's ok from a stability point-of-view, too. I just thought that a function with an extra mode argument might be more useful than having one function for each mode. Of course, `nfc_get_supported_baud_rate` would default to initiator mode.

fuzxxl, 2015-06-18:
10 days have passed since the last comment. Could you merge the commits from branch `nfc_get_supported_baud_rate_proposal` so I can go on and adopt the changes in my code or is there any reason to wait longer?

doegox, 2015-06-18:
As nobody opposed to the proposed change, I merged it.
Thanks for your report fuzxxl

closed 2015-06-18 by doegox

fuzxxl, 2015-06-18:
It was a pleasure to me!