nfc-tools/libnfc #299: Semantic versui

smortex, 2015-06-08:


fuzxxl, 2015-06-08:
Maybe you could just try harder not to break the API every other version? It's not that difficult.

neomilium, 2015-06-09:
@fuzxxl we are trying to not break API but sometimes its needed... How your comment could be useful?

fuzxxl, 2015-06-09:
@neomilium  Well, in the last case (cf. issue #298) it wasn't actually needed to break the API but you still did. Breaking the API is required much less often then it seems on first glance. You could do a great service to your users by trying harder not to break the API, even when the result might be slightly less beautiful than a broken API; that's what I tried to say with my comment.

smortex, 2017-04-03:
Hello, it's 2017! [SemVer](http://semver.org/) FTW!

closed 2017-04-03 by smortex