nfc-tools/libnfc #303: Remove pcre dependency

Maxhy, 2015-07-20:
Regex are only used to parse config file but because of that libnfc depends of pcre.
Which is kind of a nightmare to maintain on Windows.

We recently [added libnfc support](https://github.com/islog/liblogicalaccess-libnfc) on our RFID library [LibLogicalAccess](https://github.com/islog/liblogicalaccess) (sort of LibFreefare library initially developed around PC/SC) and it is working great ; thank you for your work here. Everything ok on Linux, but providing **clean** Windows release is more complicated partially because of this dependency.

Is removing regex to avoid pcre dependency and adding manual string parsing something acceptable for you?
Let me know, if positive I or another liblogicalaccess maintainer will handle it later. Thanks!

neomilium, 2015-07-22:
Yes, I think its acceptable :)

Feel free to PR !

Thanks

neomilium, 2015-08-04:
PR merged, thanks :)

closed 2015-08-04 by neomilium