nfc-tools/libnfc #305: nfc-emulate-tag example - SIGINT hang up on windows

thnimacio, 2015-07-31:
Abording the emulation of an nfc tag (via STR+C) leads to the SIGINT cennected function "intr_hdlr", calling "nfc_abort_command", "nfc_close" and "nfc_exit".
The" nfc_close" command leads to a windows API call "PurgeComm" (in /contrib/win32/libnfc/buses/uart.c -> "uart_flush_input") that does not return.

Tested with libnfc 1.7.0 rc6, rc7 and libnfc master (pulled last week) on windows XP and windows 7, all giving the same issue.

neomilium, 2015-08-03:
Any patch to make it work ? (Nobody is working on Windows port in core team, ATM)

AxisRay, 2015-08-18:
I‘m afraid that I can't find anything wrong with it……
It works well on my computer.
Could you tell me  the more detailed information?

thnimacio, 2015-08-18:
The example "nfc-emulate-tag.exe" is calling the function "nfc_target_init" to start the tag emulation. Using a PN532 NFC-Shield connected via uart/serial the driver calls the windows API function "ReadFile" with zero timeout (in "uart.c", "uart_receive"-function). I am able to scan the tag with a smartphone. But if I want to abort the tag emulation pressing CTR+C (SIGINT) the program hangs up while calling "nfc_close". It hangs up because the windows API call "PurgeComm" (called in "uart_flush_input"-function) does not return. I'm not familar with windows API programming but maybe calling PurgeComm while another ReadFile function is blocking the same handle is not allowed.

I created a post in the developer forum where I tested a simple but ugly workaround before creating this issue that can be found here:
http://forums.nfc-tools.org/topic/1410/pn532-uart-hang-up-on-windows-using-nfcemulatetag-example/

AxisRay, 2015-08-19:
It's really interesting.
In fact,I found that in MSDN and I think it is the reason for the problem.
I'm still trying to solve it...
>Because signal-handler routines are usually called asynchronously when an interrupt occurs, your signal-handler function may get control when a run-time operation is **incomplete** and in an **unknown state**. The following list summarizes the restrictions that determine which functions you can use in your signal-handler routine.
> * Do not issue low-level or STDIO.H I/O routines (for example, printf or fread).
> * Do not call heap routines or any routine that uses the heap routines (for example, malloc,_strdup, or _putenv). See malloc for more information.
> * Do not use any function that generates a system call (for example, _getcwd or time).
> * Do not use longjmp unless the interrupt is caused by a floating-point exception (that is, sig is SIGFPE). In this case, first reinitialize the floating-point package by using a call to _fpreset.
> * Do not use any overlay routines.

neomilium, 2017-04-03:
This issue is old and and have no recent comment. The code may have been fixed upstream.
Please reopen if relevant.

closed 2017-04-03 by neomilium