nfc-tools/libnfc #311: contrib/win32/nfc.def

BourgeoisLab, 2015-08-21:
I went through nfc.def and compared it to nfc.h. I found some missing entries and one unnecessary entry.

Here the nfc.def content how it should be:
```
LIBRARY libnfc
VERSION 1.7

EXPORTS
  nfc_init
  nfc_exit
  nfc_register_driver
  nfc_open
  nfc_close
  nfc_abort_command
  nfc_list_devices
  nfc_idle
  nfc_initiator_init
  nfc_initiator_init_secure_element
  nfc_initiator_select_passive_target
  nfc_initiator_list_passive_targets
  nfc_initiator_poll_target
  nfc_initiator_select_dep_target
  nfc_initiator_poll_dep_target
  nfc_initiator_deselect_target
  nfc_initiator_transceive_bytes
  nfc_initiator_transceive_bits
  nfc_initiator_transceive_bytes_timed
  nfc_initiator_transceive_bits_timed
  nfc_initiator_target_is_present
  nfc_target_init
  nfc_target_send_bytes
  nfc_target_receive_bytes
  nfc_target_send_bits
  nfc_target_receive_bits
  nfc_strerror
  nfc_strerror_r
  nfc_perror
  nfc_device_get_last_error
  nfc_device_get_name
  nfc_device_get_connstring
  nfc_device_get_supported_modulation
  nfc_device_get_supported_baud_rate
  nfc_device_set_property_int
  nfc_device_set_property_bool
  iso14443a_crc
  iso14443a_crc_append
  iso14443b_crc
  iso14443b_crc_append
  iso14443a_locate_historical_bytes
  nfc_free
  nfc_version
  nfc_device_get_information_about
  str_nfc_modulation_type
  str_nfc_baud_rate
  str_nfc_target

```

neomilium, 2015-08-21:
Could provide a patch or fork the repo and made a push request, please ?

neomilium, 2017-04-03:
This issue is old and and have no recent comment. The code may have been fixed upstream.
Please reopen if relevant.

closed 2017-04-03 by neomilium