nfc-tools/libnfc #316: Unable to claim USB interface (Device or resource busy) MAC OSX

luli-isa, 2015-09-04:
Hi, I have a nfc reader/writer ACR122U. I would like to use libnfc on my mac book with Yosemite 10.10.04. I follow all steps on 

http://nfc-tools.org/index.php?title=Libnfc

I have installed all dependencies required for mac using "mac port" with succes but when try "nfc-list" command I have an error

>nfc-list

nfc-list uses libnfc libnfc-1.7.1-52-g48c271e
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Resource busy)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:003:022-072f-2200-00-00

On FAQ wiki of libnfc, I have read the section

http://nfc-tools.org/index.php?title=Libnfc:FAQ

"Unable to claim USB interface (Device or resource busy)

You did not read the README!
Udev rules have not been installed..."

I don't understand if I must setup some rules for the device, and I don't know how to do it on my mac OSX.

grossjonas, 2015-09-05:
Did you try running it as root?

luli-isa, 2015-09-08:
Yes, i try with sudo command

sudo nfc-list

I have the same error

grossjonas, 2015-09-20:
Sorry, I do not own macs anymore.

grossjonas, 2016-01-06:
Since this question came up again I tried a better answer in #330. Please refer to that issue if the problem persists.

neomilium, 2017-04-03:
This issue is old and and have no recent comment. The code may have been fixed upstream.
Please reopen if relevant.

closed 2017-04-03 by neomilium

da0201451, 2017-05-18:
I am actually having the same issue:

error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)

unknowndomain, 2017-05-30:
I'm having this issue also

Chaaang, 2018-03-16:
Anything NEW?

nospam2000, 2019-12-08:
> Anything NEW?

I have written a detailed description how to fix the issue on OSX [here](https://stackoverflow.com/questions/57876097/nfc-reader-acr122u-and-libnfc-driver-acr122-usb-unable-to-write-to-usb-result-t/59230125#59230125)

Linzdigr, 2022-08-08:
@nospam2000 This is no longer working on new MacOS releases (from BigSure)