nfc-tools/libnfc #319: How to configure multiple device?

andrea-spotsoftware, 2015-10-13:
Hello everyone.
I'm experiencing some issues about configuring and running libnfc to work with an Adafruit pn532 reader (via UART) and an ACR122U together.

At the moment, I can only configure the Adafruit Reader, but not the ACR122.

Which are correct commands to run?
This is what i did:

```
./configure --sysconfdir=/etc --prefix=/usr
make
make install
```

Then i created a file in the /etc/nfc/devices.d/pn532.conf with the following content for the UART device:
```
name = "PN532 board via UART"
connstring = pn532_uart:/dev/ttyAMA0
allow_intrusive_scan = true
```

everything works just fine for the first device, but running nfc-list i cannot get the ACR122.
Then i tried to create another file in devices.d directory
```
name = "ACR122"
connstring = acr122_usb:001:004
```
But now, running nfc-list tells me that it can't open the device.
```
root@spot-raspberrypi ~/libnfc-1.7.0/examples # nfc-list
nfc-list uses libnfc 1.7.0
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:004
NFC device: pn532_uart:/dev/ttyAMA0 opened
```
What I'm missing? Is the connection string correct? lsusb gives me Bus 001 and device 004.
```
Bus 001 Device 004: ID 072f:2200 Advanced Card Systems, Ltd 
```

I don't wanna use pcsc drivers for ACR122u, and I already blacklisted nfc and pn533 on linux.

Thank you so much

neomilium, 2015-10-14:
First, allow_intrusive_scan is an option for main file only, do not use it
in devices.d/ files. Additionally, you do not need it if you specify serial
devices as you made.

Next, please use the latest version.

Finally, use nfc-scan-device -v to have the correct device connstring.

neomilium, 2015-10-14:
Please use the mailing list or StackOverflow for questions/help, where a wider community will be able to help you. We reserve the issues tracker for issues only.

closed 2015-10-14 by neomilium

andrea-spotsoftware, 2015-10-15:
Okay, 
however thank you!