nfc-tools/libnfc #321: nfc-list "No device found" but pcscd up and working

harseld, 2015-10-25:
Hello!

I have a SDI011 card reader and I installed libnfc 1.7.1, and pcsc-tools-1.4.24. When I run pcscd in debug mode on foreground, I can see that my reader is detected. When I run  "pcsc_scan" (nfc-tools) and I present one card to the reader, I have the output below (output of pcscd -d -a -f) :
----------------------------------------------------------------------------------------------------------------------------
14244407 winscard_msg_srv.c:230:ProcessEventsServer() Common channel packet arrival
00000041 winscard_msg_srv.c:242:ProcessEventsServer() ProcessCommonChannelRequest detects: 7
00000040 pcscdaemon.c:111:SVCServiceRunLoop() A new context thread creation is requested: 7
00000085 winscard_svc.c:301:ContextThread() Thread is started: dwClientID=7, threadContext @0x1ef4a20
00000038 winscard_svc.c:319:ContextThread() Received command: CMD_VERSION from client 7
00000021 winscard_svc.c:331:ContextThread() Client is protocol version 4:3
00000011 winscard_svc.c:351:ContextThread() CMD_VERSION rv=0x0 for client 7
00000126 winscard_svc.c:319:ContextThread() Received command: ESTABLISH_CONTEXT from client 7
00000020 winscard.c:193:SCardEstablishContext() Establishing Context: 0x4722BCA3
00000008 winscard_svc.c:412:ContextThread() ESTABLISH_CONTEXT rv=0x0 for client 7
00000114 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00000156 winscard_svc.c:319:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 7
00000029 winscard_svc.c:319:ContextThread() Received command: CMD_STOP_WAITING_READER_STATE_CHANGE from client 7
00000017 winscard_svc.c:393:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 7
00000096 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00000153 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00000074 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00000106 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00059855 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00000071 winscard_svc.c:319:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 7
54743766 winscard_svc.c:311:ContextThread() Client die: 7
00000039 winscard.c:204:SCardReleaseContext() Releasing Context: 0x4722BCA3
00000011 winscard_svc.c:933:MSGCleanupClient() Thread is stopping: dwClientID=7, threadContext @0x1ef4a20
00000007 winscard_svc.c:939:MSGCleanupClient() Freeing SCONTEXT @0x1ef4a20
------------------------------------------------------------------------------------------------------------------------------

But when I use nfc-list (with sudo or not), I get 
"nfc-list uses libnfc 1.7.1"
"No NFC device found."

However, the debug output of pcscd is

-------------------------------------------------------------------------------------------------------------------
17903129 winscard_msg_srv.c:230:ProcessEventsServer() Common channel packet arrival
00000045 winscard_msg_srv.c:242:ProcessEventsServer() ProcessCommonChannelRequest detects: 7
00000011 pcscdaemon.c:111:SVCServiceRunLoop() A new context thread creation is requested: 7
00000105 winscard_svc.c:301:ContextThread() Thread is started: dwClientID=7, threadContext @0x1ef4a20
00000065 winscard_svc.c:319:ContextThread() Received command: CMD_VERSION from client 7
00000023 winscard_svc.c:331:ContextThread() Client is protocol version 4:3
00000015 winscard_svc.c:351:ContextThread() CMD_VERSION rv=0x0 for client 7
00000108 winscard_svc.c:319:ContextThread() Received command: ESTABLISH_CONTEXT from client 7
00000033 winscard.c:193:SCardEstablishContext() Establishing Context: 0x649E0536
00000014 winscard_svc.c:412:ContextThread() ESTABLISH_CONTEXT rv=0x0 for client 7
00000075 winscard_svc.c:319:ContextThread() Received command: CMD_GET_READERS_STATE from client 7
00000095 winscard_svc.c:319:ContextThread() Received command: RELEASE_CONTEXT from client 7
00000028 winscard.c:204:SCardReleaseContext() Releasing Context: 0x649E0536
00000016 winscard_svc.c:427:ContextThread() RELEASE_CONTEXT rv=0x0 for client 7
00000060 winscard_svc.c:311:ContextThread() Client die: 7
00000039 winscard_svc.c:933:MSGCleanupClient() Thread is stopping: dwClientID=7, threadContext @0x1ef4a20
00000024 winscard_svc.c:939:MSGCleanupClient() Freeing SCONTEXT @0x1ef4a20
---------------------------------------------------------------------------------------------------------------------

So, why is nfc-list, and any other command from libnfc not working ?????

neomilium, 2015-10-26:
libnfc does not relies on PCSC.
PCSC is a different project.

SDI011 is PCSC-compliant but not libnfc-compliant.

neomilium, 2015-10-26:
Please use the mailing list or StackOverflow for questions/help, where a wider community will be able to help you. We reserve the issues tracker for issues only.

closed 2015-10-26 by neomilium