nfc-tools/libnfc #329: Format Mifare Classic with nfc-mifclassic FAIL

pablito12, 2015-12-22:
Hi!
I need to format some Mifare Classic cards and I am using the command "nfc-mfclassic". I have two dump files but I get the following error:

usr@ubuntu:~$sudo nfc-mfclassic w b blankChinese.mfd 1KClassic.mfd
NFC reader: SCM Micro / SCL3711-NFC&RW opened
Expected MIFARE Classic card with UID starting as: 5afe8448
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): e4  3a  dd  50  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |.nfc_initiator_transceive_bytes: Timeout
nfc_initiator_transceive_bytes: Timeout
failed to write trailer block 7 
xnfc_initiator_transceive_bytes: Timeout
nfc_initiator_transceive_bytes: Timeout
failed to write trailer block 11 
x.........nfc_initiator_transceive_bytes: Timeout
nfc_initiator_transceive_bytes: Timeout
failed to write trailer block 51 
xnfc_initiator_transceive_bytes: Timeout
nfc_initiator_transceive_bytes: Timeout
failed to write trailer block 55 
xnfc_initiator_transceive_bytes: Timeout
nfc_initiator_transceive_bytes: Timeout
failed to write trailer block 59 
xnfc_initiator_transceive_bytes: Timeout
nfc_initiator_transceive_bytes: Timeout
failed to write trailer block 63 
x|
Done, 40 of 64 blocks written.

Then I use a Proxmark device to get the the keys and that's what I get:

Iterations count: 15
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|001|  6338a371c0ed  | 1 |  243f160918d1  | 1 |          
|002|  f124c2578ad0  | 1 |  9afc42372af1  | 1 |          
|003|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|004|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|005|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|006|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|007|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|008|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|009|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|010|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|011|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|012|  63f17a449af0  | 1 |  82f435dedf01  | 1 |          
|013|  c4652c54261c  | 1 |  0263de1278f3  | 1 |          
|014|  d49e2826664f  | 1 |  51284c3686a6  | 1 |          
|015|  3df14c8000a1  | 1 |  6a470d54127c  | 1 |          
|---|----------------|---|----------------|---|

I don't undertend why it works on just some sectors. I really need that this can works in all. Please somebady help me!!

joanbono, 2016-02-23:
If you have this keys:

> |012| 63f17a449af0 | 1 | 82f435dedf01 | 1 |
|013| c4652c54261c | 1 | 0263de1278f3 | 1 |
|014| d49e2826664f | 1 | 51284c3686a6 | 1 |
|015| 3df14c8000a1 | 1 | 6a470d54127c | 1 |

You could do a nested attack with your reader with libnfc

`user@local:~# mfoc -k 6a470d54127c -k 3df14c8000a1 [...] -O mifaredump.mfd `

However, this is not a place to ask, you must [forum](http://forums.nfc-tools.org/forum/3/questions-and-requests/) instead. This is a place for **issues in libnfc**.

Regards and hope it works

neomilium, 2016-02-24:
@joanbono :+1:

closed 2016-02-24 by neomilium