nfc-tools/libnfc #330: Unable to claim USB interface (Permission denied)

RodrigezJson, 2015-12-29:
i run nfc-list on terminal Mac Yosemite
But the error appears:
```
nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
```
Please guide me to solve this problem.

grossjonas, 2015-12-29:
Did you try doing this as root, e.g. ```sudo nfc-list```?

RodrigezJson, 2015-12-30:
Ya but same problem.so do u think i miss out somwthing during instllation.

grossjonas, 2016-01-01:
did you install libusb?

RodrigezJson, 2016-01-01:
Yes absolutely. Still cant solve the problem

grossjonas, 2016-01-03:
Sorry, I remember having this problem with my last Mac - but I sold that one 2+ years ago.
Maybe you could try with a higher log level, like mentioned [here](http://nfc-tools.org/index.php?title=Libnfc:Troubleshooting)
```
LIBNFC_LOG_LEVEL=3 nfc-list
```
and post the console output.

alex-chan, 2016-01-04:
I also meet this problem:

```
# LIBNFC_LOG_LEVEL=3  nfc-list                                                                                                        !6590
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 253 Device 004 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:253:004"
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
debug	libnfc.general	Unable to open "acr122_usb:253:004".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:253:004
```

Also:

```
  # LIBNFC_LOG_LEVEL=3  nfc-mfclassic W b w100_1k.mfd w100_1k.mfd f                                                           !6591
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 253 Device 004 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:253:004"
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
debug	libnfc.general	Unable to open "acr122_usb:253:004".
nfc-mfclassic: ERROR: Error opening NFC reader
```

logless, 2016-01-06:
Having the exact same problem... any solution?

grossjonas, 2016-01-06:
First of all this seems to be a duplicate of #316 .

The Proxmark guys seem to have the same problem [here](http://www.proxmark.org/forum/viewtopic.php?id=986) and fixed it by writing a custom dummy kernel extension, which prevents apples usb driver claiming the interface (for more info see [this stackoverflow post](http://stackoverflow.com/questions/20253350/unable-to-claim-usb-interface-with-c-libusb-on-mac-os-x) or [this apple archive](http://lists.apple.com/archives/usb/2010/Nov/msg00039.html)).

As temporary solution you could try to kill pcscd([as mentioned here](http://alon21034.github.io/blog/2013/11/29/acr122-on-macosx/)).
Which brings us to the next approach:
```
debug   libnfc.general  1 device(s) found using acr122_usb driver
```
suggests you use the ```usb driver```. According to this [link](https://www.mail-archive.com/search?l=pcsclite-muscle@lists.alioth.debian.org&q=subject:%22[Pcsclite-muscle]+Authenticate+on+OSX+with+NFC%22&o=newest&f=1) you might succeed by using the ```pcsc driver```. I think this is an config option( ```./configure --help```), but I can't verify this right now.

Which brings us to the another approach that actually should have been the first one:
The [README](https://github.com/nfc-tools/libnfc) states something about a ```.plist``` file that needs to be changed:
```
ACR122:
-------
Using an ACR122 device with libnfc and without tag (e.g. to use NFCIP modes or
card emulation) needs yet another PCSC-lite tweak: You need to allow usage of
CCID Exchange command.  To do this, edit libccid_Info.plist configuration file
(usually /etc/libccid_Info.plist) and locate "<key>ifdDriverOptions</key>",
turn "<string>0x0000</string>" value into 0x0001 to allow CCID exchange or
0x0005 to allow CCID exchange and bogus devices (cf previous remark) and
restart pcscd daemon.

Warning: if you use ACS CCID drivers (acsccid), configuration file is located
in something like: /usr/lib/pcsc/drivers/ifd-acsccid.bundle/Contents/Info.plist
```

Sorry that I can't try these approaches my self, since I do not Macs anymore. Please be so kind to report back on your success trying these approaches.

logless, 2016-01-07:
I have solved the problem on MacOS El Capitan (the newer the MacOS the harder is the solution).

First, as already mentioned, you need to create a codeless dummy Kernel extension (kext) in /System/Library/Extensions/DUMMY.kext/Contents/Info.plist

Here's an example:

`<?xml version="1.0" encoding="UTF-8"?>
    <!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
    
    
    
    
    <plist version="1.0">
    <dict>
        <key>CFBundleDevelopmentRegion</key>
        <string>English</string>
        <key>CFBundleIconFile</key>
        <string></string>
        <key>CFBundleIdentifier</key>
        <string>com.ACR122U.dummy.kext</string>
        <key>CFBundleInfoDictionaryVersion</key>
        <string>6.0</string>
        <key>CFBundlePackageType</key>
        <string>KEXT</string>
        <key>CFBundleSignature</key>
        <string>????</string>
        <key>CFBundleVersion</key>
        <string>1.0.0d1</string>
        <key>IOKitPersonalities</key>
        <dict>
            
            <key>ACR122U</key>
            <dict>
                <key>CFBundleIdentifier</key>
                <string>com.ACR122U.dummy.kext</string>
                <key>IOClass</key>
                <string>com.ACR122U.dummy.kext</string>
                <key>IOProviderClass</key>
                <string>IOUSBInterface</string>
                <key>bConfigurationValue</key>
                <integer>1</integer>
                <key>bInterfaceNumber</key>
                <integer>0</integer>
                <key>idProduct</key>
                <integer>XXXXXXXXXXXXXXXXXXXXX</integer>
                <key>idVendor</key>
                <integer>XXXXXXXXXXXXXXXXXXXXX</integer>
            </dict>
        </dict>
        <key>OSBundleLibraries</key>
        <dict>
            <key>com.apple.iokit.IOUSBFamily</key>
            <string>1.8</string>
        </dict>
    </dict>
</plist>
`

Replace product and vendor ID (XXXXXs) with your hardware info (get this info using the following command in terminal).

`system_profiler SPUSBDataType`

IMPORTANT! Don't forget to convert the hex IDs to integers and then place them in the Info.plist file. 

In order to load the unsigned driver in recent OSX versions, you will need to change system settings. In Recovery Mode, run the following command:

`csrutil disable `

This disables Rootless System Integrity Protection.

Now you are ready to load the driver. Use the following command:

`sudo kextload -verbose  /System/Library/Extensions/DUMMY.kext  # or whatever you named the kext`

You should get a message that the driver is loaded. 

Now reattach the reader. If using the ACR122U, the LED will no longer light up red and will no longer change to green when you place a card. Don't worry, it is working!

Now run your NFC commands. If it says NFC device not found, reattach and run the command several times immediately after reattaching until it runs successfully.

grossjonas, 2016-01-07:
Wow, that's really hard. Congrats on getting it to work.
These steps should be documented somewhere.
SInce there is a ```README-WIndows.txt```, would a ```README-MacOs.txt``` be the appropriate place?

oiooj, 2016-02-21:
I have the same problem on my MAC

oiooj, 2016-02-21:
```
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.config	Unable to open directory: /usr/local/Cellar/libnfc/1.7.1/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 020 Device 011 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	1 device(s) found using acr122_pcsc driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:020:011"
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
debug	libnfc.general	Unable to open "acr122_usb:020:011".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:020:011
debug	libnfc.driver.acr122_pcsc	Attempt to open ACS ACR122U
debug	libnfc.driver.acr122_pcsc	PCSC connect failed
debug	libnfc.general	Unable to open "acr122_pcsc:ACS ACR122U".
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U
```

pvieito, 2016-02-28:
@logless Hi, I'm trying to use the Dummy Kext trick but I haven't been able to reproduce it. My Info.plist code is below:
 
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<!-- This is a dummy driver which binds to ACR122U. It -->
<!-- contains no actual code; its only purpose is to -->
<!-- prevent Apple's USBHID driver from exclusively -->
<!-- opening the device. -->
<plist version="1.0">
<dict>
	<key>CFBundleDevelopmentRegion</key>
	<string>English</string>
	<key>CFBundleIconFile</key>
	<string>
	</string>
	<key>CFBundleIdentifier</key>
	<string>com.ACR122U.dummy.kext</string>
	<key>CFBundleInfoDictionaryVersion</key>
	<string>6.0</string>
	<key>CFBundlePackageType</key>
	<string>KEXT</string>
	<key>CFBundleSignature</key>
	<string>????</string>
	<key>CFBundleVersion</key>
	<string>1.0.0d1</string>
	<key>IOKitPersonalities</key>
	<dict>
		<!-- The ACR122U USB interface -->
		<key>ACR122U</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.ACR122U.dummy.kext</string>
			<key>IOClass</key>
			<string>com.ACR122U.dummy.kext</string>
			<key>IOProviderClass</key>
			<string>IOUSBInterface</string>
			<key>bConfigurationValue</key>
			<integer>1</integer>
			<key>bInterfaceNumber</key>
			<integer>0</integer>
			<key>idProduct</key>
			<integer>8704</integer>
			<key>idVendor</key>
			<integer>1839</integer>
		</dict>
	</dict>
	<key>OSBundleLibraries</key>
	<dict>
		<key>com.apple.iokit.IOUSBFamily</key>
		<string>1.8</string>
	</dict>
</dict>
</plist>
```

Can you send me your kext to try it? Thank you!

anderssonjohan, 2016-03-01:
@logless @pvieito I have tried this and have the identical Info.plist file as @pvieito and when running ```sudo nfc-list``` I get the following output:

**Having the ACR122U disconnected**

```
nfc-list uses libnfc 1.7.1
No NFC device found.
```

**Having the ACR122U plugged in**

```
nfc-list uses libnfc 1.7.1
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:250:005
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U PICC Interface
```

anderssonjohan, 2016-03-01:
@logless @pvieito @RodrigezJson I just solved the issue on my mac running El Capitan. Hooray! :cake: 

It is the SmartCardServices service, which uses [PCSCLite](https://pcsclite.alioth.debian.org/pcsclite.html), that claims the access to the device.
I made a backup of that driver's Info.plist file, then edited the file and commented out the ACR122U product ID. After a reboot the device led does not lit up and now I get the following output:

```
$ sudo nfc-list
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
```

See [my edited Info.plist file for SmartCardServices](https://gist.github.com/anderssonjohan/4ae8fe09100224ce119c).

Please observe the path to the file, ```/usr/libexec/SmartCardServices/drivers/ifd-ccid.bundle/Contents/Info.plist```

pvieito, 2016-03-02:
I got it to work without modifying any OS X system file, compiling `libnfc` with only the `acr122_pcsc` driver:

```
./configure --with-drivers=acr122_pcsc
make install
```

ollym, 2016-03-04:
@pvieito I get this now:

```
$ nfc-list
nfc-list uses libnfc 1.7.1
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U
``

ollym, 2016-03-04:
And logs:

```
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.general	1 device(s) found using acr122_pcsc driver
debug	libnfc.driver.acr122_pcsc	Attempt to open ACS ACR122U
debug	libnfc.driver.acr122_pcsc	PCSC connect failed
debug	libnfc.general	Unable to open "acr122_pcsc:ACS ACR122U".
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U
```

neomilium, 2016-03-06:
acr122_pcsc is an obsolete driver and should not be used anymore. (and will be dropped in future)

anderssonjohan, 2016-03-07:
@neomilium Good to know, thanks for the info! :+1:

pvieito, 2016-03-07:
So, is there any solution that doesn't use the `acr122_pcsc` driver without modifying OS X and thus having to disable the System Integrity Protection?

neomilium, 2016-03-07:
If goal is to daily use libnfc under OSX, you should find a way to bypass
PCSC.

If you just want to hack something using libnfc, you should prefer using a
virtual machine or dual boot with linux.

wincss, 2016-03-11:
@anderssonjohan 
I edited my /usr/libexec/SmartCardServices/drivers/ifd-ccid.bundle/Contents/Info.plist to prevent pcsc-lite access my acr122u reader. And it works.

But it work ONLY ONCE.

After I plug the reader:

First time:
```
mac15:~ wincss$ LIBNFC_LOG_LEVEL=3 nfc-scan-device
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-scan-device uses libnfc 1.7.1
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 020 Device 014 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using acr122_pcsc driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
1 NFC device(s) found:
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:020:014"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 14
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00
debug	libnfc.general	"ACS / ACR122U PICC Interface" (acr122_usb:020:014) has been claimed.
- ACS / ACR122U PICC Interface:
    acr122_usb:020:014
debug	libnfc.driver.acr122_usb	ACR122 Abort
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 11 00 00 00 00 00 00 00 00 ff 00 00 00 0c d4 06 63 02 63 03 
63 0d 63 38 63 3d
debug	libnfc.driver.acr122_usb	RX: 80 09 00 00 00 00 00 00 81 00 d5 07 00 00 00 00 00 90 00
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and 
framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and 
framing during receiving)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 08 63 02 80 63 
03 80
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00
debug	libnfc.chip.pn53x	InRelease
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 52 00
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 53 00 90 00
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
```
but it doesn't work next time:
```
mac15:~ wincss$ LIBNFC_LOG_LEVEL=3 nfc-scan-device
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-scan-device uses libnfc 1.7.1
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 020 Device 014 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using acr122_pcsc driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
1 NFC device(s) found:
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:020:014"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00
debug	libnfc.general	Unable to open "acr122_usb:020:014".
nfc_open failed for acr122_usb:020:014
```

it won't work before I unplug & replug the usb cable.

Do you know why and how to resolve the problem? Thank you.

anderssonjohan, 2016-03-11:
@wincss I had similar problems when having the reader connected to my Macbook using an USB hub. I have no issues when it's connected directly to a port on the computer.

I also experienced this when re-enabling the System Integrity Protection (```csrutil enable```)

abdulazizalmass, 2016-04-24:
@anderssonjohan  I am using usb hub. it is really a challenge to be honest. The only solution which worked for me is this:
```
git clone https://code.google.com/p/libnfc/
cd libnfc
autoreconf -iv
./configure --with-drivers=acr122_pcsc
make
```

but worked only when I run it one time after the usb is plugged.

Still looking for efficient solution

abdulazizalmass, 2016-04-24:
One solution to try is through @logless answer. If you could please upload the kext, then I can test your scenario. Unfortunately, apple docs are not up to date. Otherwise, I would have created one from [ here](https://developer.apple.com/library/mac/documentation/Darwin/Conceptual/KEXTConcept/KEXTConceptKEXT/kext_tutorial.html)

I

ishangirdhar, 2016-07-13:
Hi @logless, 

Thank you for sharing detailed steps. I followed all your steps and each step was succeed as it should, problem is it card reader works as it was before like LED's are working and sound is coming. However, there is a service that starts every time I connect my reader with Mac OS x, Service name is "ifdreader", I kill is forcefully and then I tried nfc-list and then nfc-jewel r dump.jwd, abruptly it worked twice and then it stopped working as well.

I am using Mac OS x El Capitan 10.11.5, I don't understand what is it that is still using nfc reader.

Any help will be highly appreciated.

-Ishan

sigsergv, 2017-03-10:
I've managed to disable ifdreader:

~~~~~
sudo launchctl remove com.apple.ifdreader
sudo launchctl stop com.apple.ifdreader
~~~~~

Now I can execute nfc-list without errors (using sudo).

sigsergv, 2017-03-10:
But unfortunately nfc-list works just once after plugging in ACR122, system log:

~~~~
10/03/17 20:35:36,000 kernel[0]: 060311.403523 ACR122U PICC Interface@14100000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
~~~~

sigsergv, 2017-03-10:
Complete log (maybe it will help):

```
% sudo LIBNFC_LOG_LEVEL=3 nfc-list -v
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.config	Unable to open directory: /usr/local/Cellar/libnfc/1.7.1/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 020 Device 015 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using acr122_pcsc driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:020:015"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00
debug	libnfc.general	Unable to open "acr122_usb:020:015".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:020:015
```

Nic062, 2017-03-12:
Hello, 

Same problem, as @sigsergv and @abdulazizalmass, I can only run it one time after the usb is plugged. It's the same on Kali 2.0 :(

teonji, 2017-03-15:
Hello, 
same problem i guess, but my log look a little bit different.
Permission problem?
```
sudo LIBNFC_LOG_LEVEL=3 nfc-list -v

debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.config	Unable to open directory: /usr/local/Cellar/libnfc/1.7.1/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.driver.acr122_usb	device found: Bus 020 Device 011 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	1 device(s) found using acr122_pcsc driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:020:011"
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
debug	libnfc.general	Unable to open "acr122_usb:020:011".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:020:011
debug	libnfc.driver.acr122_pcsc	Attempt to open ACS ACR122U PICC Interface
debug	libnfc.driver.acr122_pcsc	PCSC connect failed
debug	libnfc.general	Unable to open "acr122_pcsc:ACS ACR122U PICC Interface".
nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U PICC Interface
```

smortex, 2017-04-03:
Please use [Stack Overflow](http://stackoverflow.com/questions/tagged/lib-nfc/) for questions/help, where a wider community will be able to help you. We reserve the *issues* tracker for **issues** only.

closed 2017-04-03 by smortex

charafsalmi, 2017-04-12:
Did anybody solve this ? This topic isn't reproduced on the website @smortex gave us.

Nic062, 2017-04-12:
Yes, by stopping to use VMware on my Mac OS and by installing natevely Kali on another computer

doegox, 2017-04-12:
@smortex didn't tell you the topic was on Stack Overflow, he told you to ask the question over there, to reach a larger audience able to help you. Here we're only dealing with bugfixes in libnfc. Thank you for your comprehension.

loganbr, 2017-07-13:
Guys after I do this:

sudo launchctl remove com.apple.ifdreader
sudo launchctl stop com.apple.ifdreader

The nfc-list works but after few minutes not ... I reboot my Mac and my USB port (Same port I used for NFC Reader) not working more.. I put my iPhone and another USB devices e not working. anybody know anything about this ?

Thanks

rbe1733, 2017-08-02:
Same problem on macOS Sierra.
Had a look on StackOverflow but no luck.

I tried to install libnfc 1.7.1 with MacPorts (doesn't work) and with HomeBrew (works once).
Then I compiled libnfc 1.7.1 with acr122_pcsc driver only.
nfc-list works once then failed until disconnect/reconnect NFC Reader.

Can't make mfoc works even once ...

Works well on CentOS under Parallels Desktop. It's a workaround but I don't understand why it bugs this way under MacOS X.

Has someone found a way to make the ACR122U work under Sierra without disabling SIP ?

Thanks

ronnyandre, 2017-08-08:
I am having the same issues. Tried everything from macOS Sierra, Kali Linux (in VirtualBox) and even natively on Raspberry Pi. Nada.

uni3orns, 2017-09-07:
rebuild libnfc, mfoc by handwork worked for me on macOs Sierra 10.12.6

dbhorst, 2017-09-07:
@uni3orns A bit more information would be great.
Did you just ./configure and make?

uni3orns, 2017-09-08:
@dbhorst here is detail
for libnfc
`
autoreconf -iv
./configure --with-drivers=acr122_pcsc
make
make install
`
for mfoc
`
autoreconf -iv
./configure
make
make install
`
after all run as root

rbe1733, 2017-09-09:
@uni3orns 

Thanks for the details.
It works better this way.

I had to delete manually some libnfc stuff installed by homebrew but not deleted when uninstalled.
That's why the compiled libnfc was not working well in my previous tests.

It's working better but it's not perfect.

Here is my protocol test 👍 
1) Connect ACR122U reader to usb => device light is red : nfc-scan-device find the card reader
2) Put a card on the reader => beep and device light is green : nfs-list reads the card
3) Remove the card from the reader ==> device light is still green

Then you can put and remove the card, and it is still working.

After a while, the ACR122U reader doesn't respond and I have tu unplug/plug it again to make it work.

With the command ioreg -p IOUSB -l -w , you have the following result : 

+-o ACR122U PICC Interface@14200000  <class AppleUSBDevice, id 0x1002b3560, registered, matched, active, busy 0 (6 ms), retain 20>
        {
          "sessionID" = 470332701745077
          "iManufacturer" = 1
          "bNumConfigurations" = 1
          "idProduct" = 8704
          "bcdDevice" = 532
          "Bus Power Available" = 250
          "USB Address" = 12
          "bMaxPacketSize0" = 8
          "iProduct" = 2
          "iSerialNumber" = 0
          "bDeviceClass" = 0
          "Built-In" = No
          "locationID" = 337641472
          "bDeviceSubClass" = 0
          "bcdUSB" = 272
          "USB Product Name" = "ACR122U PICC Interface"
          "PortNum" = 2
          "non-removable" = "no"
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "bDeviceProtocol" = 0
          "IOUserClientClass" = "IOUSBDeviceUserClientV2"
          "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=4,"CapabilityFlags"=32768,"MaxPowerState"=4,"DriverPowerState"=4}
          "kUSBCurrentConfiguration" = 1
          "Device Speed" = 1
          "USB Vendor Name" = "ACS"
          "idVendor" = 1839
          "IOGeneralInterest" = "IOCommand is not serializable"
          "IOClassNameOverride" = "IOUSBDevice"
        }

I wonder if there is a way to "reset" the attached USB port without plug/unplug.
For the moment, I can't find what to unload (kextunload) with kextstat result.

Do you have the same behavior or not ?

Thanks for your help

rbe1733, 2017-09-10:
I ran a few extra tests :

- When a card is put on the device and left in place, the device keeps working indefinitely.

- When you put off the card, the card reader is no longer operational after a few seconds if you don't put another card (the green light turns off then turns on to red). 

If you can put a card in this state, after a few seconds, you get a beep and the light switch to green. The card reader is operational again.

So no need to unload kext, just have to wait.
Using the results of previous ioreg command, it seems to be possible to unload the kext used by the card reader, i.e IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle

The command is : sudo kextunload -v 2 -class com.apple.iokit.IOUSBFamily
If the command works, the return is : All instances of class com.apple.iokit.IOUSBFamily terminated.

mac-zhou, 2017-12-21:
@sigsergv Thanks

Chaaang, 2018-03-16:
Well, It's not that well in my 10.13

jacopo-j, 2018-05-22:
I can confirm that unloading the `ifdreader` daemon allows `libnfc` to control the device and use it *just once* (macOS High Sierra). After that, the reader stops responding and it only works again after unplugging and plugging back in, even if there are no permissions/ownership issues preventing `libnfc` from accessing the device. Maybe reopen the issue?

Also, editing SmartCardServices as @anderssonjohan showed doesn't seem to work anymore on High Sierra, at least not for me.

JiaJian, 2018-08-31:
Same issue as those who experienced it in High Sierra. Any idea how to make it work more than once?

bramus, 2019-01-28:
Still an issue. Fixes mentioned don't work.

I've noticed that there are drivers available on https://www.acs.com.hk/en/products/3/acr122u-usb-nfc-reader/ … is there a way to make `nfc-tools` use these (they are installed in `/usr/local/libexec/SmartCardServices/drivers`)?

bettse, 2019-06-13:
I found success on Mojave with linbfc (installed using homebrew) by setting the `LIBNFC_DEVICE` environment variable (Insead of trying to disable ifdreader).

before, testing with `mifare-desfire-info` from libfreefare:
> $ mifare-desfire-info
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
mifare-desfire-info: nfc_open() failed.

> $ LIBNFC_DEVICE="acr122_pcsc" mifare-desfire-info
===> Version information for tag 047e1aa2654080:
UID:                      0x047e1aa2654080
Batch number:             0xba65189f10
Production date:          week 48, 2015
...TRUNCATED FOR READABILITY...
Free memory: 128 bytes
Use random UID: no
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
mifare-desfire-info: nfc_open() failed.

So although an error is still printed, it works.

nospam2000, 2019-12-07:
@anderssonjohan thanks for this solution:
> It is the SmartCardServices service, which uses [PCSCLite](https://pcsclite.alioth.debian.org/pcsclite.html), that claims the access to the device.
> I made a backup of that driver's Info.plist file, then edited the file and commented out the ACR122U product ID.

I have written a detailed description how to fix the issue on OSX [here](https://stackoverflow.com/questions/57876097/nfc-reader-acr122u-and-libnfc-driver-acr122-usb-unable-to-write-to-usb-result-t/59230125#59230125)

maxgfr, 2020-03-08:
> I got it to work without modifying any OS X system file, compiling `libnfc` with only the `acr122_pcsc` driver:
> 
> ```
> $ ./configure --with-drivers=acr122_pcsc
> $ make install
> ```

It works for me. 

```
git clone https://github.com/nfc-tools/libnfc.git
cd libnfc 
autoreconf -is 
./configure --with-drivers=acr122_pcsc
make install
 ```

nospam2000, 2020-03-08:
> I got it to work without modifying any OS X system file, compiling `libnfc` with only the `acr122_pcsc` driver:

This is no new information. When you read above in this thread you will find this from neomilium :
_"acr122_pcsc is an obsolete driver and should not be used anymore. (and will be dropped in future)"._

That means your are lucky when it works with today's version, but it might not work tomorrow if you don't convince the developers not to drop the support for acr122_pcsc.

@neomilium: are there still plans to remove the acr122_pcsc driver? It really causes big problems to Mac Users if you remove them.

sam1902, 2020-04-29:
> I've managed to disable ifdreader:
> 
> ```
> sudo launchctl remove com.apple.ifdreader
> sudo launchctl stop com.apple.ifdreader
> ```
> 
> Now I can execute nfc-list without errors (using sudo).

This solves it for me running macOs Catalina 10.15.4 (without disabling SIP thanksfuly)

Don't forget to run `brew install pcsc-lite` for the `CCID exchange` stuff.

vkravets, 2021-10-18:
Sorry but it seems it will not work after reboot.... My miss

jvhaarst, 2023-04-14:
I first found this issue before I found the solution here : https://hackerman.medium.com/losing-and-gaining-access-52d31d42fd38

In short : stopping the service enables you to use a command that accesses the RFID reader, but after a while the OS starts the service again, blocking access again.
The simple solution is to just stop it every time you run a command, sleep for a few seconds, and it'll be fine:
`sudo launchctl stop com.apple.ifdreader ; sleep 10 ;nfc-list -v`
 
This isn't a complete solution (this would mean blocking the OS from restarting the service) , but it at least allows to work with the RFID reader.

jeff-pang, 2023-05-19:
I had similar problem. Initially i made the mistake of installing `libnfc` and `mfoc` via `brew install`. I had to uninstall and then compile with `libnfc` with ` --with-drivers=acr122_pcsc` as described above and then also compiled `mfoc` (as already described above)

steps to do that:

first uninstall any versions of libnfc and mfoc then begin compile for `libnfc`

```
git clone https://github.com/nfc-tools/libnfc.git
cd libnfc
autoreconf -iv
./configure --with-drivers=acr122_pcsc
make && make install
```

then compile mfoc

```
git clone https://github.com/nfc-tools/mfoc.git
cd mfoc
autoreconf -is
./configure
make && make install
```

replace `make install` with `sudo make install` if necessary (e.g make && sudo make install)

leeren, 2023-08-04:
2023 and still get this issue - anyone have a fix?

leeren, 2023-09-19:
Any pointers on how to do go about doing that @Maxigame99 ? Really feels like NFC programming is not possible on macs these days

Maxigame99, 2023-09-19:
I ended up to open VM using UTM (on macOS) with Linux Ubuntu Server Minimal and run all the scripts right there.

[1 more reply]