nfc-tools/libnfc #331: Variable LIBNFC_ROOT_DIR does not exist

mtangoo, 2016-02-03:
I followed instructin to build library here: http://libnfc.googlecode.com/svn/trunk/README-Windows.txt
After configuring everything I get errors below.
Is there something that am missing?
TIA

```
 Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Found LIBUSB: C:/TestNFC/libusb-1.0.20/MinGW32/dll/libusb-1.0.dll.a C:/TestNFC/libusb-1.0.20/include
Found LIBUSB: C:/TestNFC/libusb-1.0.20/MinGW32/dll/libusb-1.0.dll.a C:/TestNFC/libusb-1.0.20/include
Year for copyright is 2016
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.4/Modules/CMakePackageConfigHelpers.cmake:258 (message):
  Variable LIBNFC_ROOT_DIR does not exist
Call Stack (most recent call first):
  CMakeLists.txt:195 (configure_package_config_file)

Configuring incomplete, errors occurred!
See also "C:/TestNFC/libnfc-build/CMakeFiles/CMakeOutput.log".
```

Nightshinra, 2016-04-18:
I have the same problem

CMake Warning (dev) at cmake/modules/UseDoxygen.cmake:94 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "doc".
Call Stack (most recent call first):
  CMakeLists.txt:70 (INCLUDE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Found LIBUSB: C:/libusb-win32-bin-1.2.6.0/lib/gcc/libusb.a C:/libusb-win32-bin-1.2.6.0/include
Found LIBUSB: C:/libusb-win32-bin-1.2.6.0/lib/gcc/libusb.a C:/libusb-win32-bin-1.2.6.0/include
Year for copyright is 18/04/2016 
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.5/Modules/CMakePackageConfigHelpers.cmake:258 (message):
  Variable LIBNFC_ROOT_DIR does not exist
Call Stack (most recent call first):
  CMakeLists.txt:194 (configure_package_config_file)

Nightshinra, 2016-04-18:
I found the solution

just click "Add Entry" in cmake-gui and add LIBNFC_ROOT_DIR and put it to path and inset as example C:/Tools/libnfc

But I have now 

CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.5/Modules/CPack.cmake:423 (message):
  CPack package description file: "C:/libnfc-1.7.0-rc7/README" could not be
  found.

And I don't now how to change the dir file

closed 2016-04-19 by mtangoo

Zamiell, 2016-05-06:
Rename `README.md` to `README`.