nfc-tools/libnfc #335: contrib sample is now incorrect

jbdatko, 2016-03-03:
I *believe* the contrib example `contrib/libnfc/pn532_via_uart2usb.conf.sample` is incorrect. It looks like the sytanx for the conf file is now `device.name` and `device.connstring` ? 

First, I just copied the the conf sample to `/etc/nfc/libnfc.conf` and got:

```
$ LIBNFC_LOG_LEVEL=3 nfc-list -v
debug	libnfc.config	key: [name], value: [Adafruit PN532 board via UART]
info	libnfc.config	Unknown key in config line: name = Adafruit PN532 board via UART
debug	libnfc.config	key: [connstring], value: [pn532_uart:/dev/ttyUSB1]
info	libnfc.config	Unknown key in config line: connstring = pn532_uart:/dev/ttyUSB1
debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc libnfc-1.7.1-66-g24979c6
No NFC device found.
```

But then I ran:

```
$ LIBNFC_LOG_LEVEL=3 nfc-scan-device -i -v
debug	libnfc.config	key: [name], value: [Adafruit PN532 board via UART]
info	libnfc.config	Unknown key in config line: name = Adafruit PN532 board via UART
debug	libnfc.config	key: [connstring], value: [pn532_uart:/dev/ttyUSB1]
info	libnfc.config	Unknown key in config line: connstring = pn532_uart:/dev/ttyUSB1
debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-scan-device uses libnfc libnfc-1.7.1-66-g24979c6
debug	libnfc.driver.pn532_uart	Trying to find PN532 device on serial port: /dev/ttyUSB1 at 115200 baud.
debug	libnfc.bus.uart	Serial port speed requested to be set to 115200 baud.
```

Which was odd. So I looked at [conf.c](https://github.com/nfc-tools/libnfc/blob/master/libnfc/conf.c#L217) and it looks like it wants `device.name` now, so I changed it:

```
$ LIBNFC_LOG_LEVEL=3 nfc-list -v
debug	libnfc.config	key: [device.name], value: [Adafruit PN532 board via UART]
debug	libnfc.config	key: [device.connstring], value: [pn532_uart:/dev/ttyUSB1]
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	1 device(s) defined by user
debug	libnfc.general	  #0 name: "Adafruit PN532 board via UART", connstring: "pn532_uart:/dev/ttyUSB1"
nfc-list uses libnfc libnfc-1.7.1-66-g24979c6
debug	libnfc.driver.pn532_uart	Attempt to open: /dev/ttyUSB1 at 115200 baud.
debug	libnfc.bus.uart	Serial port speed requested to be set to 115200 baud.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
```

I spent like all of 15 minutes playing around with libnfc (read as I'm not very familiar with libnfc), so if this does look like the bug, I'll submit a PR to fix the conf sample.

WildDylanX, 2016-06-15:
Hi jbdatko-
Thanks for posting this!  I'm trying to troubleshoot connecting a Raspberry Pi 2 board up to a pn352 nfc card using UART (following directions at https://learn.adafruit.com/adafruit-nfc-rfid-on-raspberry-pi/overview) but unfortunately these directions seem to have been written for libnfc-1.7.0... I'm pretty new to this, so just trying to follow whatever clues I can to get it going.  I've been getting similar errors to you in the debug (though I've been trying to connect via serial: ttyAM0 per the tutorial).  I noticed that it seems to be looking for the syntax device.connstring (while I think libnfc-1.7.0 only looked for connstring).  How did you finally change the .conf file to deal with the new syntax?

jbdatko, 2016-06-15:
@WildDylanX Yeah, if you change the conf file to have `device.X` that worked for me. If you have multiple USB devices plugged into your pi, you might want to make a udev rule so that the adafruit devince consistently shows up as `/dev/adafruit-nfc` or something and then use *that* in your conf file.

WildDylanX, 2016-06-16:
I'm a bit lost - this is the .conf file I have been using, which part do I need to change to device.X?
[image]

jbdatko, 2016-06-16:
@WildDylanX Check out the pull request. Try:

```
device.name = 
device.connstring = 
```

WildDylanX, 2016-06-16:
Hmmm.  That's funny - that was the first thing I tried (that's how I had interpreted your first message), but then the errors changed from saying "unknown key device.connstring" to "unknown key device.device.connstring".  I wonder if because the .conf file is in the etc/nfc/devices.d directory... The way I have it set-up right now, I have a libnfc.conf file in the etc/nfc directory and a pn352_uart_on_rpi.conf file in the etc/nfc/devices.d directory.  The pn352 conf file looks just like the one in the pull request commit, but when I add "devices." in front of the name or connstring, I just end up with devices.devices.X in the unknown key error

jbdatko, 2016-06-16:
Odd. This is what I have in mine:

```
~ $ tree /etc/nfc 
/etc/nfc
├── devices.d
└── libnfc.conf

1 directory, 1 file
~ $ cat /etc/nfc/libnfc.conf 
## Typical configuration file for PN532 board (ie. microbuilder.eu / Adafruit) device
device.name = "Adafruit PN532 board via UART"
device.connstring = pn532_uart:/dev/nfcshield
```

WildDylanX, 2016-06-16:
Ok.  I moved the pn532 conf file that I had inside the devices.d folder out to the etc/nfc level and overwrote the libnfc.conf file I had put there before.  Now the unknown key errors are gone, but still getting an error on libnfc.driver.pn532_uart saying Invalid serial port /dev/ttyAMA0
Any idea on that one?

jbdatko, 2016-06-16:
Not really besides the obvious. Are you sure that's the correct path? Is that what `dmesg` says? The `AMA` looks odd to me, I would have expected `/dev/ttyACM0` or similar. Also, do you have appropriate permission set?

WildDylanX, 2016-06-16:
Yeah... I'm thinking there's something funky going on there - I've been following the directions at: https://learn.adafruit.com/adafruit-nfc-rfid-on-raspberry-pi/building-libnfc but they are a few years old and based on Raspbian Wheezy and libnfc-1.7.0.  I'm working from a clean install of Raspbian Jessie on a Raspberry Pi 2, but when I got to the step about downloading libnfc from code.google.com, it redirected me to github and I figured out how to git clone libnfc-1.7.1 over.  The screenshots in the tutorial show dev/ttyAMA0 and that's what the pn532_uart_on_rpi.conf.sample file has in there, but it doesn't seem to be working.  I'm not too familiar with what ports are available to use... What is ACM0?

jbdatko, 2016-06-16:
Unplug the FTDI cable. Plug it back in, Then do `dmesg | tail` Here's mine:

```

WildDylanX, 2016-06-16:
Ok!  Now I'm getting somewhere.  Thank you!  I didn't know about the dmesg command before.  I did like you suggested and it didn't show anything different when I unplugged or plugged in the FTDI cable so even went back and looked at the whole dmesg log (not just the tail).  I realized that the device doesn't even seem to be detecting.  I'm not sure if it's a faulty board or maybe I didn't solder the headers on well enough... I've checked the wiring and I was fooled since the red power light came on and when I did nfc-list it said I had 1 device... But I think that was just saying that I had one registered.

neomilium, 2017-04-03:
Contrib sample is correct… but should be placed in `devices.d` directory.

closed 2017-04-03 by neomilium