nfc-tools/libnfc #340: libnfc Installation: Build/compile error problem in OSX El Captain

abdulazizalmass, 2016-03-31:
I am trying to install libnfc-1.7.1 as RFID Reader Kiosk for (ACR122U).
It worked on raspberry [pi,](http://chocolatechips.me/how-to-raspberry-pi-rfid-reader-kiosk-acr122u/) but  libnfc installation was not successful on my mac.

### I executed these commands on terminal:
```
sudo port -v selfupdate        // stack overflow 
sudo port upgrade outdated // suggested by terminal 
sudo port install libusb-legacy //source libnfc community website

wget https://bintray.com/artifact/download/nfc-tools/sources/libnfc-1.7.1.tar.bz2
tar -xvzf libnfc-x.x.x.tar.gz
cd libnfc-x.x.x
./configure --prefix=/usr -sysconfdir=/etc
make
brew update && brew upgrade
sudo make install 

```
### a warning showed up while executing "make":
```
arygon.c:94:22: warning: unused variable 'arygon_error_incomplete_command'
      [-Wunused-const-variable]
static const uint8_t arygon_error_incomplete_command[] = "FF0C0000\x0d\x0a";

```
I even tried:
`make clean && make`

## later on "sudo make install" errored:
```
~/Downloads/libnfc-1.7.1$ sudo make install
Making install in libnfc
Making install in chips
make[3]: Nothing to be done for `install-exec-am'.
make[3]: Nothing to be done for `install-data-am'.
Making install in buses
make[3]: Nothing to be done for `install-exec-am'.
make[3]: Nothing to be done for `install-data-am'.
Making install in drivers
make[3]: Nothing to be done for `install-exec-am'.
make[3]: Nothing to be done for `install-data-am'.
Making install in .
test -z "/usr/lib" || .././install-sh -c -d "/usr/lib"
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   libnfc.la '/usr/lib'
libtool: install: /usr/bin/install -c .libs/libnfc.5.dylib /usr/lib/libnfc.5.dylib
install: /usr/lib/libnfc.5.dylib: Operation not permitted
make[3]: *** [install-libLTLIBRARIES] Error 71
make[2]: *** [install-am] Error 2
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1
```
it is simialr to this issue [https://lists.nongnu.org/archive/html/openexr-user/2006-08/msg00001.html](url)

### The user admin privileges were verified. 

#Please help me get the libnfc installation sorted out as RFID Reader Kiosk if possible.

pulimento, 2016-07-12:
Hey @abdulazizalmass , I know maybe it's too late, but I will write my solution, for the sake of completness. Those are the updated steps to install libusb on Mac OS X succesfully:

Requirements:
- XCode, any will do
- Brew
- Packages to install from brew: libtool autoconf automake pkgconfig glib libusb

Steps:
1) Clone repo
2) On the directory just cloned, and after install the dependencies, run those commands:

```
autoreconf -vis
./configure
make
sudo make install
````

That's it, I think I don't miss anything. Tested on Mac OS X 10.10 Yosemite and 10.11 El Capitan.

abdulazizalmass, 2016-07-12:
We'll be testing this and will reply back to you. Thanks

closed 2017-04-03 by smortex