nfc-tools/libnfc #342: Read ISO 14443-4B with PN532_uart

Hayrisson, 2016-04-07:
Hello,

Using libnfc-1.7.1, I am able to read ISO 14443-4B cards with a ACR122U through USB (which is a PN532 chipset if I'm right) but not with a PN532 through UART.
Could you help me?

Thank you in advance for any help.

Mathis

nianqing126, 2016-07-14:
i dont know

smortex, 2017-04-03:
Can you please provide a sample application that reproduces the error?

l29ah, 2017-04-09:
Just hit that with PN5321. I'm using `LIBNFC_DEVICE="pn532_uart:/dev/ttyUSB0:115200" nfc-poll` and it doesn't seem to see my biometric passport. `strace` doesn't see anything going on when i try to scan it.

lincichen, 2018-01-26:
hello !
ghost
I have the same problem as you.Have you solved your previous problem?Can you give me some advice?

lincichen, 2018-01-30:
我找到使用的方法了，就不打英文了。
我的环境是ubuntu14.04 lTS ,串口已被系统识别 设备好ttyUSB0
1.cd /libnfc/examples
2.root@ubuntu:/home/tmp/libnfc/examples# pn53x-tamashell 
>3205000002
>4a010300
>05000071ff
通过上述操作你可以获得到B卡的应答。接下来参考下面提及的文档与B卡进行交互
嗯，没错，就是中文。
值得一提的是，B卡的操作需要使用底层的命令，例如：REQB,ATTRIB等，你可以参考《fcd-14443-3.pdf》。
没错，就是用中文。

closed 2021-01-04 by smortex

l29ah, 2021-01-05:
Sup?

smortex, 2021-01-05:
We closed a bunch of old issues…  I closed this one because:
* Reporter deleted their account;
* "ACR122U" (known to be crap)
* No feedback from the reporter after asking more details
* Using a PN5321 with the pn532_uart driver?
* Text I can't read with "ubuntu14"

In order for problems to be fixed, people have to send us Pull-Requests, and I am afraid that this discussion is not moving in that direction :-/