nfc-tools/libnfc #352: PN532 sees only Mifare Classic cards with 4 byte UIDs

hsanjuan, 2016-05-20:
I have a very weird problem. I have two very similar PN532 readers in both the Elechouse (http://www.elechouse.com/elechouse/index.php?main_page=product_info&cPath=90_93&products_id=2242) and Adafruit versions (https://www.adafruit.com/products/364). I use them in UART mode.

libnfc sees them and describes them exactly the same:

```
nfc-scan-device uses libnfc 1.7.1
1 NFC device(s) found:
- pn532_uart:/dev/ttyUSB0:
    pn532_uart:/dev/ttyUSB0
chip: PN532 v1.6
initator mode modulations: ISO/IEC 14443A (106 kbps), FeliCa (424 kbps, 212 kbps), ISO/IEC 14443-4B (106 kbps), Innovision Jewel (106 kbps), D.E.P. (424 kbps, 212 kbps, 106 kbps)
target mode modulations: ISO/IEC 14443A (106 kbps), FeliCa (424 kbps, 212 kbps), D.E.P. (424 kbps, 212 kbps, 106 kbps)
```

However, the Elechouse version is unable to see any tags other than Mifare Classic cards (1K/4K) which happen to have 4 byte UIDs (not sure if related). The Adafruit version sees all the expected cards (I tested with Desfire, Jewel, Ultralight...). There is no error when polling the tags, they are simply treated as if they weren't there at all.

I find this very weird because they are the same in the end (ISO/IEC 14443A (106 kbps) targets) and I have no idea why the Elechouse reader only sees some cards and not others.

Can someone tell me if this might be libnfc related (perhaps not configuring the chip correctly if somehow it had different defaults or came different from factory?). Additional insights on possible causes are also welcome.

punkadiddle, 2016-05-21:
I noticed the same thing. Also the smaller board takes a bit longer to recognize tags. The Elechouse board is the only one with the electronics within the antenna. May guess was that the antenna design might not be according to the design book.

hsanjuan, 2016-05-21:
Oh it's good to know I'm not alone. Pardon my ignorance, but if Mifare Classic use ISO/IEC 14443A (106 kbps), shouldn't this mean that the antenna can communicate with all tags compatible with that and that it shouldn't matter if I there's a Desfire or an Ultralight below with use the same modulation? 

Also, for the price I paid on Ebay, I think I might have gotten the Chinese fake Elechouse clone they warn about in the website. Pretty weird this thing...

kikiorg, 2016-06-06:
Are you also using pynfc and the mifareauth.py example/library?  I had this problem and solved it, but the solution was in the mifareauth.py file, not libnfc -- look for these two lines:

            if nt.nti.nai.szUidLen == 4:
                uid = "".join([chr(nt.nti.nai.abtUid[i]) for i in range(4)])

And change them to this:

            if nt.nti.nai.szUidLen >= 1:
                uid = "".join([chr(nt.nti.nai.abtUid[i]) for i in range(nt.nti.nai.szUidLen)])

I'm not really sure why they hardcoded the 4 in place and didn't use szUidLen from the start.

After I made that change, I was able to read a number of other kinds of tags.  I'm able to read my transit card, an EV charge card and some tiny Adafruit RFID tags (all with 7 digits.)

Kiki

AdamLaurie, 2016-06-07:
On 06/06/16 21:57, kikiorg wrote:
> Are you also using pynfc and the mifareauth.py example/library? I had
> this problem and solved it, but the solution was in the mifareauth.py
> file, not libnfc -- look for these two lines:
>
> |if [...]

Hi Kiki,

Good catch! Can I suggest you submit a pull request to pynfc?

cheers,
Adam
-- 
Adam Laurie                         Tel: +44 (0) 20 7993 2690
Suite 7
61 Victoria Road
Surbiton
Surrey                              mailto:adam@algroup.co.uk
KT6 4JX                             http://rfidiot.org

hsanjuan, 2016-06-07:
I am not using pynfc, I'm using plain libnfc's `nfc-poll` for example... libnfc works well with UIDS longer than 4 bytes so I kind of doubt it's related. However I don't have a working 8 byte UID tag for this reader so I cannot be 100% sure.

kikiorg, 2016-06-23:
Thanks Adam.  Give me a few days -- I'm still learning my way around github, after a very long programming hiatus. :)

Kiki

kikiorg, 2016-06-23:
@hsanjuan  -- don't know if this is helpful, but here's the user manual for your reader:

http://www.elechouse.com/elechouse/images/product/PN532%20NFC%20Module/PN532%20NFC%20RFID%20Module%20Manual.pdf

I noted this when I searched for Uid:

`if(sta && buf[0] == 4) {`

This is their sample Arduino code, but I can see that "4" is hardcoded in their own sample.  You might check to see if you can make a similar change that I did for the pynfc library.  (The file is a PDF and does not allow me to copy and paste text, but this snippet is the second find of "Uid".)

Then submit a pull request if it works.  Then tell me how to do that, haha. ;) (j/k -- I'll figure it out when I'm done with my own code. :) )

Best of luck everyone!

Kiki

hsanjuan, 2016-06-23:
@kikiorg thanks for the suggestions, but I'm not using the Arduino code. My problem is that two identical readers behave differently (using only libnfc) and `nfc-poll` only sees one type of card with one of them, despite both being the same chip. It's probably not even libnfc's problem. I think I have a fake elechouse reader anyway.

What really puzzles me is that it doesn't make sense that it fails to see cards using ISO/IEC 14443A (106 kbps), which is the same as the Mifare classic which are seen without problem.

ya0guang, 2016-08-18:
Oh I have the same problem. I bought PN532 of Elechouse, a red board from Taobao, and it can only read Mifare 1k and 4k card by using nfc-list. So that may be a hardware shortcoming. So what kind of PN532 board is better?

hsanjuan, 2016-08-25:
@ya0guang on the elechouse website for the product there is a guide to distinguish their board from some unofficial clones on the market... can you check if your device is a clone? Mine looks like it...

ya0guang, 2016-08-25:
@hsanjuan I found the product website of PN532 on elechouse and I think my chip is different from their product. The font of "ElECHOUSE" is not italic, and there is no mark on serveral pores. So I think I bought a fake one.

doegox, 2017-02-16:
Sorry to hear you got a clone :(
I'm closing this ticket now as there is nothing we can do about it.

closed 2017-02-16 by doegox