nfc-tools/libnfc #361: How to correct 'undefined reference'

rhyaam, 2016-08-05:
../libnfc/.libs/libnfc.so: undefinded reference to 'pn532_spi_driver'
../libnfc/.libs/libnfc.so: undefinded reference to 'acr1222_driver'
../libnfc/.libs/libnfc.so: undefinded reference to 'arygoni_driver'
../libnfc/.libs/libnfc.so: undefinded reference to 'acr122_usb_driver'
../libnfc/.libs/libnfc.so: undefinded reference to 'pn532_i2c_driver'
../libnfc/.libs/libnfc.so: undefinded reference to 'pn532_usb_driver'
collect2: error: ld returned 1 exit status
Makefile:525: recipe for target 'nfc-emulate-forum-tag4' failed

mindsolve, 2017-01-19:
This does not only affect you, I can recreate this problem.
System:
Ubuntu 16.10; x64
`gcc (Ubuntu 6.2.0-5ubuntu12) 6.2.0 20161005`
Kernel: `4.8.0-32-generic`

For me it worked by running 
```
sudo ./configure
sudo make
sudo make install-recursive
```
maybe this is completely wrong, but it works now
(before trying to do this, run `sudo make clean`)

doegox, 2017-02-17:
I could compile, install and execute in a Docker Ubuntu 16:10 with:
```
docker run -it ubuntu:16.10
cd
apt-get update
apt-get install git autoconf libtool libusb-dev libglib2.0-dev
git clone https://github.com/nfc-tools/libnfc.git
cd libnfc/
autoreconf -vis
./configure
make
make install
ldconfig
nfc-list
```
Of course the last command fails because there is no USB in my Docker image.

deancsmith, 2017-03-28:
I am a relative newbie when it comes to compiling from source and I had the same issue on a fairly vanilla install of Ubuntu 16.04 with:

Kernel: x86_64 Linux 4.4.0-66-generic
gcc version: 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4)

The above from doegox helped me get it working.

Could steps similar to the above be added to the Readme.md or elsewhere? It appears to be a little out of date, e.g. referencing an INSTALL file that no longer seems to exist.

smortex, 2017-04-03:
> Could steps similar to the above be added to the Readme.md or elsewhere? It appears to be a little out of date, e.g. referencing an INSTALL file that no longer seems to exist.

Feel free to submit a PR, the nfc-tools folks may merge them on a best-effort basis 😄

closed 2017-04-03 by smortex