nfc-tools/libnfc #364: SmartMX with MIFARE 1K emulation

Xoib, 2016-09-12:
Hello,

With a SCL3711 (PN533), I am trying to deal with a `SmartMX with MIFARE 1K emulation`.
Of course I obtain an error when using mifare tools because it doesn't recognize how to deal with the mifare part.
Is there anyone that found a solution already?

Thanks,

sniperle, 2016-09-25:
Since the last version I have problems with my PN533 too. I can't read tags successfully.

$ mfoc -O mfc1k.mfd 
error   libnfc.chip.pn53x       Unexpected PN53x reply!
ISO/IEC 14443A (106 kbps) target:
[...]
Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Classic 1K
* MIFARE Plus (4 Byte UID or 4 Byte RID) 2K, Security level 1
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:

Try to authenticate to all sectors with default keys...
Symbols: '.' no key found, '/' A key found, '\' B key found, 'x' both keys found
[Key: ffffffffffff] -> [xnfc_initiator_mifare_cmd: Timeout

doegox, 2017-02-17:
I tried nfc-mcflassic on a JCOP J3A081 + MFC 1k and nfc-mfclassic worked as expected.
Are you sure your card has MFC or is it just based on nfc-list output?

TonyPeng-cyber, 2019-07-16:
Hi, I meet the same problem when trying to write/read block for Mifare Emulator in Mifare Smart  MX based card. Can you help share more experience on how you solve the problem?