nfc-tools/libnfc #365: Correct Detection of multiple MIFARE 'Magic' tag

quantum-x, 2016-09-28:
-  Multiple'MAGIC' Mifare cards with the same UIDs are not detected when on the reader - they are considered as one. Presumably this is in: nfc.c - > nfc_initiator_list_passive_targets () 

Is it possible that nfc_initiator_list_passive_targets() can have a switch to list 'duplicates', or badges with the same UID ?

quantum-x, 2016-10-03:
Hi @neomilium 
I noticed you'd authored most of this function, would you have some ideas for the problem?
Specifically, if you have multiple Chinese Magic Tags, that have the same UIDs, nfc-list only detects one tag. Presumably this is probably handled by the deselect function, but when I did some basic hacking of the source code, I couldn't produce a version that showed the correct number of badges.

At worst, even the ability to show that there are multiple tags, or not would be fine.
Any suggestions?

smortex, 2016-10-03:
As far as I can recall, the anti-collision depends on tags having different UIDs… I would be really surprised if a PCD could detect two PICC with the same UID (I don't say it *is* impossible, I say *I am really confident believing it is* impossible… my guess is that both cards are kept active simultaneously and really bad (but quite fun) things may happen).

For me, this is an *invalid* request.

quantum-x, 2016-10-03:
It seems that both tags get sent the same information (that is to say, nfc-mfclassic will write to both tags, etc)

I will dig into the deselect function, to see if we can have a system to at least detect if there are multiple tags with the same ID (even if we can't detect reliably how many).

I've been trying to add more and more coverage to libnfc to keep up with the oddities in Chinese tags, hopefully we can have some succes..

doegox, 2017-02-16:
There is no way for things to behave correctly in this situation.
Detection of multiple cards occur in the reader firmware when both cards send their UID an two bits differ. As bits are Manchester-encoded, the reader can detect the collision and process with the anti-collision algorithm. If both cards are having the same ID, they will be seen and selected/halted as one, but it becomes worse if you try to authenticate to the card(s) because they will presumably send two different nonces and collisions will occur. (unless the magic MFC use a very bad RNG and reply both with the same nonce, in which case you could write to both cards simultaneously, but again if they contain different data and you read them, collisions will occur). Have fun with your experiments, I'm closing this ticket ;)

closed 2017-02-16 by doegox