nfc-tools/libnfc #369: "Connection timed out" error using ACR122u in a kali virtual machine

ecbec, 2016-11-05:
The command pcsc_scan works great,
but running nfc-list in my kali virtual machine I got the error:

```
# LIBNFC_LOG_LEVEL=3  nfc-list
debug	libnfc.config	Unable to open directory: /etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 003 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:003"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00 
error	libnfc.driver.acr122_usb	Unable to write to USB (Connection timed out)
debug	libnfc.general	Unable to open "acr122_usb:001:003".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:003
```

Please, can anyone help me to fix this?
Thanks

_**Extra infos:**_

1) OS version:
```
# uname -a
Linux kali 4.7.0-kali1-amd64 #1 SMP Debian 4.7.8-1kali1 (2016-10-24) x86_64 GNU/Linux
```

2) The error happens in virtualbox and vmware machines

3) It doesn't happen in a physical machine ( the same version of kali running in a notebook)

neomh, 2016-11-29:
Same issue here.

PSANTOS86, 2016-12-27:
Hello,

Anyone solve it?

Tks!

justyb0, 2016-12-27:
hi solved what
compiled the windows versiopn ?

neomh, 2017-01-24:
It's a bug on Virtual machines, it works on hard installations :) just boot kali from usb and it will work

closed 2017-02-17 by doegox

JorgeLiu, 2018-01-28:
Anyone solve it?

Raphdeumax, 2018-08-30:
UP, have you find a solution ?