nfc-tools/libnfc #373: LibUSB ./configure error

Azef1, 2016-11-23:
Hi, 

When i do `./configure` to install lib-nfc 1.7.1 i have this error:
```
checking which drivers to build... acr122_usb
checking for documentation request... no
./configure: line 14026: syntax error near unexpected token `libusb,'
./configure: line 14026: `        PKG_CHECK_MODULES(libusb, libusb, HAVE_LIBUSB=1, HAVE_LIBUSB=0)'

```
So, i did `apt-get install libusb-dev` 
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-dev is already the newest version (2:0.1.12-30).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
But i always have the same error when i do ./configure

Could-you help me. My NFC reader is ACR122U and i work on kali linux

BuptPeter, 2016-11-24:
mayby you can try use:
autoconfig -vis
(you have to install autoconf at first.)

doegox, 2017-02-17:
You got the error
```
./configure: line 14026: syntax error near unexpected token `libusb,'
./configure: line 14026: `        PKG_CHECK_MODULES(libusb, libusb, HAVE_LIBUSB=1, HAVE_LIBUSB=0)'
```
because you're missing `libglib2.0-dev`. Install it and try again:
```
apt-get install libglib2.0-dev
autoreconf -vis
./configure
make
```

smortex, 2017-04-03:
Closing as @doegox gave a solution.  Please reopen if this does not fix your problem.

closed 2017-04-03 by smortex

Krazubu, 2018-03-28:
I'm getting the same error with 1.5.1, however doing
```
apt-get install libglib2.0-dev
autoreconf -vis 
```
does not help

symdeb, 2022-08-27:
From clean Ubuntu I had to install
build-essentials
autoconf
libusb-1.0
libudev-dev
libglib2.0-dev

bosgnoom, 2025-03-23:
> libusb-compat.

Installing libusb-compat and libusb-compat-devel did it for me on fedora 41