nfc-tools/libnfc #384: Nfc-emulate-forum-tag2 / Problems with some phones

jbatea, 2016-12-15:
Hi everyone,

I try to emulate a type2 tag, with one of the example's functions of libnfc : nfc-emulate-forum-tag2.

That's work everytime with most of phones, but sometimes i have reading problems with a Samsung galaxy s6. 

My setup:
-Orange pi pc.
-Adafruit Pn532.
-Samsung s6.
-./my_emulate_tag (copy of nfc-emulate-forum-tag2 program, i just add debug)

Expected behaviour:
What happening with most of phones and sometimes (~1/10) with a samsung s6:

```
root@orangepipc:~/my_emulate# ./my_emulate_tag 
NFC device: pn532_uart:/dev/ttyS3 opened
Emulating NDEF tag now, please touch it with a second NFC device
My_emulate_target
My_target_init

In:   
30  00  
Out:  
00  00  00  00  00  00  00  00  00  00  ff  ff  e1  10  06  0f  
In:   
30  04  
Out:  
03  21  d1  02  1c  53  70  91  01  09  54  02  65  6e  4c  69  
In:   
30  08  
Out:  
62  6e  66  63  51  01  0b  55  03  6c  69  62  6e  66  63  2e  
In:   
30  0c  
Out:  
6f  72  67  00  00  00  00  00  00  00  00  00  00  00  00  00  
In:   
50  00  
HALT sent
```

Unexpected behaviour:
What happening with a samsung s6 in most of cases (~9/10) :

```
root@orangepipc:~/my_emulate# ./my_emulate_tag 
NFC device: pn532_uart:/dev/ttyS3 opened
Emulating NDEF tag now, please touch it with a second NFC device
My_emulate_target
My_target_init
```

This problems seems to came from RATS command, because nothing happen after the call of "my_target_init", and as said in function documentation:

> Warning:
> Be aware that this function will wait (hang) until a command is received that is not part of the anti-collision. The RATS command for example would wake up the emulator. After this is received, the send and receive functions can be used.

Does someone knows this problem, and maybe a way to solve it?

Thanks.

jbatea, 2017-01-13:
Hi everyone, i don't solve my problem, but i have new infos.

I supposed this problem came from the uid of the tag i tried to emulate, let me explain :

Tag i emulate is a 7 bytes uid, but what happened is the reader (with all phones) only see a 4 bytes uid (check with Nxp Tag Reader and Nfc Pro Tools).
I emulate the tag with an Adafruit pn532, what i understand is it's impossible to emulate an 7 bytes uid with this board : 

http://nfc-tools.org/index.php?title=PN53x
http://manpages.org/nfc-emulate-forum-tag2
(The emulated target has only a 4-byte UID while most devices assume a Tag Type 2 has always a 7-byte UID (as a real Mifare Ultralight tag))

So, first question it's is really impossible to emulate a 7 bytes uid with libnfc and Adafruit pn532?

Then, if yes, do a samsung s6 (Samsung S3FWRN5P Nfc Controller) can correctly handle a 4 bytes uid tag?
 
And if it can't, there is another existing board that can emulate 7 bytes uid tag?

Thanks.

doegox, 2017-01-13:
PN532 can't emulate 7-byte UID.

nfc-emulate-uid.c could probably be modified to emulate 7-byte anticollision but this type of emulation is very fragile anyways (hard to setup and non-standard timings) so you're risking even more problems.

> And if it can't, there is another existing board that can emulate 7 bytes uid tag?

ChameleonMini can, Proxmark too probably.

jbatea, 2017-01-14:
Ok, thanks for your response, i will try something with nfc-emulate-uid, and if i fail to do it, i will take a look on others boards.

closed 2017-02-16 by doegox