nfc-tools/libnfc #385: NetBSD support

manu0401, 2016-12-22:
Here are two patch to build libnfc on NetBSD:

[patch-libnfc_drivers_acr122_pcsc.c.txt](https://github.com/nfc-tools/libnfc/files/668022/patch-libnfc_drivers_acr122_pcsc.c.txt)
[patch-libnfc_buses_uart.c.txt](https://github.com/nfc-tools/libnfc/files/668023/patch-libnfc_buses_uart.c.txt)

manu0401, 2016-12-23:
The two previous patches were against 1.7.1. Here are patches against git
[0001-NetBSD-definition-for-CCID-ioctl.patch.txt](https://github.com/nfc-tools/libnfc/files/670274/0001-NetBSD-definition-for-CCID-ioctl.patch.txt)
[0002-NetBSD-serial-ports.patch.txt](https://github.com/nfc-tools/libnfc/files/670273/0002-NetBSD-serial-ports.patch.txt)

doegox, 2017-02-16:
Thank you for your contribution, your patched have now been pushed

closed 2017-02-16 by doegox