nfc-tools/libnfc #390: NFC-Tools website disappeared

Davideddu, 2017-02-03:
Your website is not working, I don't know any other way to contact you other than opening an issue.
The website disappeared today, it's been replaced with this weird aviation megastore thing: https://postimg.org/image/b3vf0yqpl/
Just in case you weren't aware.
Have a nice day

12458, 2017-02-03:
I am also experiencing the same problem. Maybe the domains got mixed up?

doegox, 2017-02-03:
Thanks for the notification, I'm trying to reach the owner. DNS seems ok, looks more like defacing via redirection (err 301).

Xoib, 2017-02-03:
If the administrator needs help, I can offer some sysadmin skills.

smortex, 2017-02-03:
@doegox, it seems to be fixed, as far as I can see.

Davideddu, 2017-02-03:
@smortex I can confirm that.

closed 2017-02-03 by Davideddu

smortex, 2017-02-03:
Excellent, thank you for your quick report ;-)