nfc-tools/libnfc #397: Can't read ISO14443B TAG

MMM1986, 2017-03-10:
Hi,

I can read 14443A TAGs without problem, but I can't read an 14443B TAG. I have tried using nfc-list, nfc-poll and my own code (down below). I have read the same 14443B TAG with a Samsung phone and the app "NFC Reader", so the TAG IS working.

The output of the Samsung phone app was:

Tag ID (hex): bc 88 e3 d9
Tag ID (dec): 3163087833
ID (reversed): 3655567548
Technologies: IsoDep, NfcB

The TAG ID changed everytime the phone readed the TAG.

I have also tried to use a nfc_initiator_transceive_bytes function to initiate a BAC identification (the tag is supposed to be compatible with ISO/IEC 7816-4) with the code below and I get an "RF Transmission Error". I can't figure out why.

Any hints of why the reader can't detect the 14443B smartcard? Thank you.

libnfc version: libnfc-1.7.1-154-gb7ae7cb.
System information: Raspbian GNU/Linux 8.0 (jessie) on a raspberry pi2 model B+.

Reader:

- itead 13.56mHz PN532 NFC SDcard module (PN532 based used via UART)
- elechouse PN532 NFC RFID Module (PN532 based used via UART)

---------------------------------

[nfctest_log.txt](https://github.com/nfc-tools/libnfc/files/832979/nfctest_log.txt)
[nfc-list_log.txt](https://github.com/nfc-tools/libnfc/files/832978/nfc-list_log.txt)

---
my code is:
------------
	#include <stdlib.h>
	#include <nfc/nfc.h>

	static void print_hex(const uint8_t *pbtData, const size_t szBytes)
	{
		size_t  szPos;
		for (szPos = 0; szPos < szBytes; szPos++) {
			printf("%02x  ", pbtData[szPos]);
		}
		printf("\n");
	}

	int main(int argc, const char *argv[])
	{
		nfc_device *pnd;
		nfc_target nt;
	
		// Allocate only a pointer to nfc_context
		nfc_context *context;
	
		// Initialize libnfc and set the nfc_context
		nfc_init(&context);
		if (context == NULL) {
		printf("Unable to init libnfc (malloc)\n");
		exit(EXIT_FAILURE);
		}
	
		// Display libnfc version
		const char *acLibnfcVersion = nfc_version();
		(void)argc;
		printf("%s uses libnfc %s\n", argv[0], acLibnfcVersion);
	
		pnd = nfc_open(context, NULL);

		if (pnd == NULL) {
			 printf("ERROR: %s\n", "Unable to open NFC device.");
		 	exit(EXIT_FAILURE);
		}
	
		const char * delviceName = nfc_device_get_name(pnd);
		const char * connstring = nfc_device_get_connstring(pnd);
	
		printf("Device Name is %s\n", delviceName);
		printf("Device Connection String is %s\n", connstring);
	
		// Set opened NFC device to initiator mode
		if (nfc_initiator_init(pnd) < 0) {
		//if (nfc_initiator_init_secure_element(pnd) < 0) {
			nfc_perror(pnd, "nfc_initiator_init");
			exit(EXIT_FAILURE);
		}

		int conf_res = 0;

		conf_res = nfc_device_set_property_bool(pnd,NP_ACTIVATE_FIELD,true);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );

		conf_res = nfc_device_set_property_bool(pnd,NP_INFINITE_SELECT,false);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_int(pnd,NP_TIMEOUT_COMMAND,1000);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_int(pnd,NP_TIMEOUT_ATR,200);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_int(pnd,NP_TIMEOUT_COM,200);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_FORCE_ISO14443_A,false);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_FORCE_ISO14443_B,true);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_FORCE_SPEED_106,true);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_EASY_FRAMING,false);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_AUTO_ISO14443_4,false);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_HANDLE_PARITY,true);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		conf_res = nfc_device_set_property_bool(pnd,NP_HANDLE_CRC,true);
		if(conf_res<0)
			printf("Error configuring property: %d, %s\n", conf_res, nfc_strerror(pnd) );
	
		printf("NFC reader opened\n");
	
		// Poll for a ISO14443B (CI Chilena) tag
		const nfc_modulation clpci = {
			.nmt = NMT_ISO14443B,
			.nbr = NBR_106, //NBR_106, NBR_212, NBR_424, NBR_847
		};
	
		uint8_t pbtInitData[2] = {0x00,0x01};
	
		int res_sel = 0;
	
		printf("Trying NMT_ISO14443B ... \n");
		res_sel = nfc_initiator_select_passive_target( pnd, clpci, pbtInitData, 2, &nt);
	
		if(res_sel>0)
		{
			printf("The following (NFC) ISO14443B tag was found:\n");
			printf("Application Data: ");
			print_hex(nt.nti.nbi.abtApplicationData, 4);
			printf("Protocol Info: ");
			print_hex(nt.nti.nbi.abtProtocolInfo, 3);
			printf("PUPI Info: ");
			print_hex(nt.nti.nbi.abtPupi, 4);
			printf("Card ID: %d", nt.nti.nbi.ui8CardIdentifier);
			//print_hex(nt.nti.nbi.ui8CardIdentifier, 1);
		}
		else
		{
			printf("No Tag Found. Last error code is : %d, %s\n", res_sel, nfc_strerror(pnd));
			//printf("ERROR: %s\n", nfc_strerror(pnd));
		}
	
		uint8_t cla = 0;	//getchallenge
		uint8_t ins = 0x84; //getchallenge
		uint8_t p[2] = {0, 0};

		uint8_t commandHeader[4] = {cla,ins,p[0],p[1]};

		uint8_t le = 9;

		uint8_t pbtTx[5] = {commandHeader[0],commandHeader[1],commandHeader[2],commandHeader[3],le};

		uint8_t pbtRx[10] = {0,0,0,0,0,0,0,0,0,0};

		int receivedBytesCount = nfc_initiator_transceive_bytes(pnd, pbtTx, 5, pbtRx, 9, -1);

		printf("%d bytes received.\n", receivedBytesCount);
		for(int i=0; i<le;i++)
		{
			printf("b%d: %d\n", i, pbtRx[i]);
		}
		
		printf("Last error code is : %d, %s\n", receivedBytesCount, nfc_strerror(pnd));
		// Close NFC device
		nfc_close(pnd);
		
		// Release the context
		nfc_exit(context);
		exit(EXIT_SUCCESS);
		
		}

-------------
[nfc-list_log.txt](https://github.com/nfc-tools/libnfc/files/832974/nfc-list_log.txt)

[nfctest_log.txt](https://github.com/nfc-tools/libnfc/files/832975/nfctest_log.txt)

Piket564, 2017-03-16:
I've same Problem with this tag:

`1 ISO14443B passive target(s) found:
ISO/IEC 14443-4B (106 kbps) target:
               PUPI: ********* 
   Application Data: 00  00  00  00  
      Protocol Info: b3  71  71 `

Nothing application of NFC tool like mfoc, mfcuk or other start with this Tag

neomilium, 2017-04-03:
Some hardware are not able to communicate with ISO14443B tags due to antenna or amplifier.
In this case, there is no way to fix it using software, sorry.

closed 2017-04-03 by neomilium