nfc-tools/libnfc #398: ACR122U "No NFC Device Found" --with-drivers=acr122_pcsc

joswr1ght, 2017-03-11:
I'm using libnfc 2d4543673e9b76c02679ca8b89259659f1afd932 on Debian 8 with an ACR122U-A9. I have `pcscd` installed and running, `pcsc_scan` correctly identifies the reader and tags, but `nfc-list` reports `No NFC Device Found.`.

This seems like a common issue, but I've been unable to figure out a resolution. I built libnfc with `--with-drivers=acr122_pcsc`, edited the `/etc/libccid_Info.plist` file, set `ifdDriverOptions` to different recommended values (restarting `pcscd` each time) but I still get the `No NFC Device Found` error.

```
# (./configure --prefix=/usr --with-drivers=acr122_pcsc && make && make install) >/dev/null
# echo $?
0
# utils/nfc-list 
/home/jwright/libnfc/utils/.libs/lt-nfc-list uses libnfc libnfc-1.7.1
No NFC device found.
# ps -ef | grep pcscd
root      41011      1  0 06:52 ?        00:00:00 /usr/sbin/pcscd --foreground --auto-exit
root      41017  31589  0 06:52 pts/0    00:00:00 grep pcscd
# pcsc_scan
PC/SC device scanner
V 1.4.23 (c) 2001-2011, Ludovic Rousseau <ludovic.rousseau@free.fr>
Compiled with PC/SC lite version: 1.8.11
Using reader plug'n play mechanism
Scanning present readers...
0: VMware Virtual USB CCID 00 00

Sat Mar 11 06:52:31 2017
Reader 0: VMware Virtual USB CCID 00 00
  Card state: Card inserted, 
  ATR: 3B 81 80 01 80 80

ATR: 3B 81 80 01 80 80
+ TS = 3B --> Direct Convention
+ T0 = 81, Y(1): 1000, K: 1 (historical bytes)
  TD(1) = 80 --> Y(i+1) = 1000, Protocol T = 0 
-----
  TD(2) = 01 --> Y(i+1) = 0000, Protocol T = 1 
-----
+ Historical bytes: 80
  Category indicator byte: 80 (compact TLV data object)
+ TCK = 80 (correct checksum)

Possibly identified card (using /usr/share/pcsc/smartcard_list.txt):
3B 81 80 01 80 80
	RFID - ISO 14443 Type A - NXP DESFire or DESFire EV1
	"Reiner LoginCard" (or "OWOK", how they name it) - they have been distributed by a german computer magazine ("Computer BILD")
	https://cardlogin.reiner-sct.com/
	Belgium A-kaart (Antwerp citycard)
	Oyster card - Transport for London
	http://en.wikipedia.org/wiki/Oyster_card
```

godfreychung, 2017-03-15:
According to the reader name "VMware Virtual USB CCID 00 00", you are using shared mode to connect the reader to VMware. Please use USB passthrough mode to connect it so that ACR122U is detected correctly.

smortex, 2017-04-03:
Closing the issue since @godfreychung seems to have a solution for this problem. Please reopen if this does not solve your problem.  Thanks!

closed 2017-04-03 by smortex

ronnyandre, 2017-08-08:
I have the exact same problem on my Raspberry Pi, not running through any VM.

ArcadeHustle, 2020-01-17:
Here is how to solve when not running in a VM. https://stackoverflow.com/questions/31131569/unable-to-claim-usb-interface-device-or-resource-busy

NotOlly02, 2020-10-02:
I've the same @smortex problem, i want to try @godfreychung solution, but how i can active USB passthrought mode?