nfc-tools/libnfc #402: Unable to claim USB interface (Device or resource busy)

frustreated, 2017-03-28:
**0x001:Envirment**
amchine:hp6515b
cpu:amd64
system:ubuntu16.0.4/16.0.10(two systems all have been tested and the have the same response)

**0x002:the errors occured**

output_1:
~$ sudo nfc-list
[sudo] password for frust: 
nfc-list uses libnfc libnfc-1.7.1-161-g32bb627
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Device or resource busy)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:003:003

output_2:
~$ nano /etc/modprobe.d/blacklist-libnfc.conf
blacklist nfc
blacklist pn533

output_3:
~$ modprobe -r pn533 nfc
modprobe: FATAL: Module pn533 is in use.

output_4:
~$ sudo LIBNFC_LOG_LEVEL=3 nfc-list -v
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc libnfc-1.7.1-161-g32bb627
debug	libnfc.driver.acr122_usb	device found: Bus 003 Device 003 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:003:003"
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Device or resource busy)
debug	libnfc.general	Unable to open "acr122_usb:003:003".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:003:003

**0x003:the operations i did**
all of my operations:
git the git ,and install 
the configure command is 

libtoolize
aclocal
autoheader
autoreconf -vis
./configure --with-drivers=acr122_usb
(without any specified drivers command also tested :./configure)
and i fond ubuntu also can install nfc-list from apt-get
initially i install from the github from source, and then i reinstalled ubuntu to test the software nfc-list that installed from apt-get, the same error occured

**0x004:3Q for master's reading and your help,god bless**
3Q for your help!
for Convenient ,i kept some notes for the installation of libnfc , because its really a little hard for me as a beginner,and i installed this tool by some glimpse of that notes ,the address is 
https://shimo.im/doc/LUnKNKmkzL0xLpIm/「nfc-install」

doegox, 2017-03-28:
On recent kernels you nee to blacklist pn533_usb too, cf
https://github.com/nfc-tools/libnfc/blob/master/contrib/linux/blacklist-libnfc.conf

frustreated, 2017-03-28:
it works!!
great 3q for your help!!
this cost me lots time ..
thanks again!! @doegox

zhangguanzhang, 2022-04-19:
I have managed to solve this issue by adding blacklist `pn533_usb` to:
```
/etc/modprobe.d/blacklist-libnfc.conf
```
so its content is:
```
blacklist nfc
blacklist pn533
blacklist pn533_usb
```
then run `sudo modprobe -rf pn533_usb`