nfc-tools/libnfc #404: Forum is not reachable anymore

neomilium, 2017-04-03:
I updated wiki page, we need to update README.md too.

neomilium, 2017-04-03:
For users questions, do we accept them on GitHub issues or should we send users somewhere else ? (e.g. StackOverFlow, GoogleGroups)

neomilium, 2017-04-03:
IMHO, developers could go in GH issue tracker. Are you agree ?

doegox, 2017-04-03:
The forum was a good place.
The issue with the forum being broken has been reported to Roel on 27/03, no reaction so far.

doegox, 2017-04-03:
Developer questions in GH, yes no prob.

anik786, 2017-05-07:
Why is the forum not reachable anymore: http://forums.nfc-tools.org/

It just says:

> This PHP environment doesn't have SQLite support built in. SQLite support is required if you want to use a SQLite database to run this forum. Consult the PHP documentation for further assistance.

If it's no longer being used anymore and using StackOverflow instead, can we update the README file already.

ildar, 2018-04-17:
https://gitter.im/nfc-tools/libfreefare is good?

closed 2021-01-04 by smortex