nfc-tools/libnfc #407: undefined reference to `pn53x_transceive'

Tiwas, 2017-04-09:
Hi.

I'm trying to install libnfc on my Windows box, but I'm having problems compiling it. After running the cmake gui, configure it and generate the makefile I proceed to run mingw32-make. This seems to work well for a long time until I get the following

> [ 90%] Building C object examples/CMakeFiles/pn53x-diagnose.dir/pn53x-diagnose.c.obj
> [ 91%] Building RC object examples/CMakeFiles/pn53x-diagnose.dir/__/windows/pn53x-diagnose.rc.obj
> [ 92%] Linking C executable pn53x-diagnose.exe
> CMakeFiles\pn53x-diagnose.dir/objects.a(pn53x-diagnose.c.obj): In function `main':
> D:/CygStuff/NFC/libnfc-master/examples/pn53x-diagnose.c:106: undefined reference to `pn53x_transceive'
> D:/CygStuff/NFC/libnfc-master/examples/pn53x-diagnose.c:117: undefined reference to `pn53x_transceive'
> D:/CygStuff/NFC/libnfc-master/examples/pn53x-diagnose.c:126: undefined reference to `pn53x_transceive'
> collect2.exe: error: ld returned 1 exit status
> examples\CMakeFiles\pn53x-diagnose.dir\build.make:117: recipe for target 'examples/pn53x-diagnose.exe' failed
> mingw32-make[2]: *** [examples/pn53x-diagnose.exe] Error 1
> CMakeFiles\Makefile2:1018: recipe for target 'examples/CMakeFiles/pn53x-diagnose.dir/all' failed
> mingw32-make[1]: *** [examples/CMakeFiles/pn53x-diagnose.dir/all] Error 2
> Makefile:148: recipe for target 'all' failed
> mingw32-make: *** [all] Error 2

Could someone help me with how to get past this?

Cheers!

wooyunfeng, 2017-04-19:
I also encountered the same problem, did you solve it?

Tiwas, 2017-04-19:
Yeah. Kind of. It hurt less to repartition my laptop, install linux on it and build all the apps :)

Sorry - wish I could give you a better answer, and if you can make it work for Windows I'd really appreciate it. Until then I'll just have to remote control my linux box...

rlaraujo, 2017-04-24:
Oops. Same situation here. Exactly the same. All the rest is working for me. PLEASE HELP!

MegatronsCannon, 2017-04-24:
Same situation here. Note: I'm on a windows 7 laptop. Also, I do not yet have the PN532 board. Is this needed prior to the build process? Soooo close ... yet so far.

FSDGFSDGSG, 2017-06-29:
who know the answer?
thanks

doegox, 2017-06-29:
Sorry, we, the usual maintainers, don't use Windows.
Please try asking for help on stackoverflow.

FSDGFSDGSG, 2017-06-29:
@doegox 
i tried to understand how to install libNfc with all the dependency on linux but i don't get it
can you explain how to do this?
witch kind on linux is recommended 
or i can download a complete image with libNfc work fine

thanks!

peacepenguin, 2017-10-06:
If you've reached the point that you're getting undefined reference related to the pn53x_xxx functions, just make the change noted in my fork:
https://github.com/peacepenguin/libnfc/commit/b7d08bff41824255ef7f96535a8bf0b6450f1d14
It just removes the culprit files from the build process, doesn't actually "fix" the root problem. But this change does allow the rest of the utilities and examples to build successfully in windows.

I was able to get a basic compile done in windows using that fork, haven't confirmed it works yet, but here's my progress so far if anyone wants to test in windows. I'm not sure where the config file is supposed to be stored by default in windows.
https://github.com/peacepenguin/libnfc-unofficialbuilds

There's detailed instructions there on how I was able to build on windows for anyone that wants to try to get to the root problem quickly.

peacepenguin, 2017-10-18:
@Tiwas 
I've got the fixes done for this problem, pull request is pending acceptance:

https://github.com/nfc-tools/libnfc/pull/457

Try compiling the fork that has those changes using the instructions I've given here: https://github.com/peacepenguin/libnfc/blob/master/README-Windows.md

The changes allow building in mingw64 natively for 64-bit windows, with an up-to-date gcc/g++ tool-chain.

Hope that helps!

-Ben