nfc-tools/libnfc #408: Fix unresponsive SCL3711

manu0401, 2017-04-18:
After a DESfire operation SCL3711 will sometimes enter a state where
libusb-0.x is unable to fill config field in struct usb_device. The
USB device has to be power-cycled to restore a sane state.

This introduce a work around, by using hardcoded values for endpoints
and packet size when they are unavailable.

[0001-Workaround-missing-USB-config.patch.txt](https://github.com/nfc-tools/libnfc/files/931363/0001-Workaround-missing-USB-config.patch.txt)

doegox, 2017-04-19:
Thank you for your contribution, it has been merged.
But the problem you got is the tip of an iceberg...
It's a known bug of PN533 when large commands will corrupt internal XRAM and destroy usb descriptors.
The official vendor recommendation was "_When the computer wants to go in suspend or shutdown state, the routine in the PCSC driver shall store the proper USB data in the PN533’s XRam_" but as we're not playing with PCSC, restoring XRAM every time a libnfc program finishes is a bit overkilling.

But your patch gave me the occasion to think again about this workaround and triggers it *only* when I know a corruption occurred (if the command is > 17 bytes) so I added the following commit: 
38164c49ef085823378716c9c117b3a92ad4c615 (more details there)

closed 2017-04-19 by doegox

doegox, 2017-04-20:
see also 767abe50e2c487958a741ecc4d80276bd9013b09 and c9ac17c9f5e793d1cba877a33e4d8353beff94e6

manu0401, 2017-04-26:
I experienced a situation where dev->config->interface->altsetting was set to a corrupt value (0x3fffffff), hence I came to a more brutal fix:
          if (pn53x_usb_supported_devices[n].uiMaxPacketSize != 0) {
            dev->config = NULL;
          }
And from then always use the hardcoded values instead of usb descriptors when we have them. Do you think it would be an acceptable change?