nfc-tools/libnfc #409: Support of other readers by ACS

dadreamer, 2017-04-19:
I know it's been asked before on the old forums but I've never seen a clear answer...

Why libnfc doesn't support most of ACS readers but ACR122 only? Especially I'm interested in [ACR1281U-C1](http://www.acs.com.hk/en/products/159/acr1281u-c1-dualboost-ii-usb-dual-interface-reader/) reader as I have one (but cannot use it with your tools). If that's a technical issue then what is it?

Thanks.

doegox, 2017-04-19:
At the moment it was also reported to support ACR1222U-C1 and ACR1222L.
The primary reason for a lack of support of the other ones is that the developers don't have a full collection of all ACR readers tp try out. Now, *if* the firmware of another ACR reader is using the same chip PN532 internally and the same mechanism to reach it (Direct Transmit via
PC_to_RDR_XfrBlock / PC_to_RDR_Escape FF000000), then it should be as easy as adding an entry in https://github.com/nfc-tools/libnfc/blob/master/libnfc/drivers/acr122_usb.c#L262-L266 
But looking at the API documentation of ACR1281U-C1 (http://www.acs.com.hk/download-manual/2603/API-ACR1281U-C1-1.07.pdf), this mechanism is absent therefore the ACR1281U-C1 can't be supported by libnfc.

closed 2017-04-19 by doegox

doegox, 2017-04-22:
From issue #285 : what about ACR1251U?
Here again its API documentation (http://www.acs.com.hk/download-manual/4414/API-ACR1251U-1.07.pdf) doesn't mention the required mechanism to talk to the inner PN532, if there is any at all.

dadreamer, 2017-04-22:
@doegox 
Thank you for the clarification. So, I assume there's no way to make those readers work if they aren't able to execute Direct Transmit commands. Except for reflashing its internal firmware, maybe... If there's a correct chip in the reader and such the firmware would be available for the end-users. I wonder how could I determine the chip model and the manufacturer without disassembling the device?..

doegox, 2017-04-22:
We may ask to ACS, or open the device or see what info we can get from the FCC certification:

## ACR1251
Based on a picture of the rear, we can see its FCCID is V5MACR1251 I found the internal pics from FCC but they're not detailed enough...
https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=2049771
Still, here is the manual including a view of the soldering pads (Annex A):
https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=2961818
We know PN532 is in a HVQFN40 package and U1 could be a match...
So there is some hope, but still this would require an adequate firmware...

## ACR1281
There is no room for a HVQFN40 at all:
https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=2559025

dadreamer, 2017-04-22:
@doegox 
When I follow these links, I get "You are not authorized to access this page." message.

doegox, 2017-04-22:
Mmm ok deep liks seem not to be possible.
Follow https://gov.fccid.io/V5MACR1251 and https://gov.fccid.io/V5MACR1281 -> details -> internal photos & user manual

dadreamer, 2017-04-23:
@doegox 
This early manual looks interesting. According to that I see that ACR1281U uses AT90SCR100H controller and RC531 interface chip. There are also the pictures of some firmware upgrade tool but I didn't find it.

amahlaka, 2018-09-14:
i have the ACR1251u and can provide high res pictures of the board if needed

amahlaka, 2018-09-14:
After probing the pins i believe that th 2x5 pads are for avr jtag
(not sure if this is useful info.)

amahlaka, 2018-09-14:
I have equipment to connect to the jtag posrt and have already soldered pins to jtag pads, but i never really done stuff with jtag.
Any help with accessing the chips thru jtag is appreciated
i have openocd installed and using buspirate to access the jtag