nfc-tools/libnfc #412: Is it a bad idea to run nfc-list command on a loop (1 second interval), to poll for new NFC tags?

MikaeI, 2017-05-02:
I have ACR122 hooked up to a Raspberry Pi, and I'm using a python script to continuously run the nfc-list command to check for tags (1 second intervals). If there is a tag, I run some more functionality. Is this a bad idea, and if so, is there a better way to poll for new tags (I only need the UID for my extended functionality)? @neomilium

neomilium, 2017-05-02:
It seems that nfc-eventd (part of nfc-tools) is exactly what you are
looking for.

MikaeI, 2017-05-02:
Thanks for you reply :)
I've looked at nfc-eventd, and I'm considering it. At it stands, my script does work perfectly. I'm just worried I'm doing something harmful by running the nfc-list command at an infinite loop at such short intervals.

smortex, 2017-05-02:
`nfc-list` is intended to discover NFC initiators and targets on them, and as such, does not return a useful exit-status.  While you can run it in a loop and watch stdout to determine what to do, it's a better design to poll directly from your application (using the polling feature of libnfc (see `examples/nfc-poll.c`) or through `nfc-eventd`) rather than wasting resources by spawning processes forever.

BTW, please use [Stack Overflow](http://stackoverflow.com/questions/tagged/lib-nfc/) for questions/help, where a wider community will be able to help you. We reserve the *issues* tracker for **issues** only.

closed 2017-05-02 by smortex