nfc-tools/libnfc #413: Libnfc doesn't detect PN532 Adafruit

sdmaviless, 2017-05-04:
Hello all,
I am a computer science student who is trying to connect PN532  on Linux for a project.
The problem is that I'm doing it throw Arduino using SPI.
I've tried many different ways of doing it but not of them have worked and I'm going crazy trying to fix it.

I'll try to give you all information about my trouble.
- Operating System: Kali Linux (last version and updated)
All necessary programs to configure and install libnfc are installed.
- Shield : Adafruit PN532 v1.3 (SEL1 and SEL0 already soldered to close them to connect by SPI )
  All connections are ok because Arduino detects it and can play with it,
- Libnfc Version: I've tried many.. even applying patches (I've though perhaps the problem is that some of them worked but because I installed others before, now they don't) (The first I tried was the last version)
- Arduino Board: Arduino Mega 2560 REV3.

After everything is install, I get No NFC device found error..
Hope anyone can help me, I've been trying to connect it for a couple of weeks and already thinking of giving up.

Thanks in advance!

anik786, 2017-05-07:
libnfc newbie here, but you may be better off using Adafruit's library https://github.com/adafruit/Adafruit_Python_PN532

probably much easier to use as well!

doegox, 2021-01-04:
not related to development of this project I'm afraid. Feel free to ask over https://discord.com/invite/2Y8tRNa94G

closed 2021-01-04 by doegox