nfc-tools/libnfc #416: How to build targeting arm64 arch?

andertsk, 2017-06-09:


smortex, 2017-06-12:
You need to give us more information on how to reproduce this issue, otherwise there is nothing we can do. Thanks!

closed 2017-06-12 by smortex

andertsk, 2017-06-12:
@smortex Thanks for fast reply!
I want to build libnfc for arm64 arch but don't know how to do that (and your repo doesn't have configure script). New iOS devices is based on this arch. And one more question - can i use your lib in this use case (using on iPhone, iPhones have nfc chip)?

doegox, 2017-06-12:
iPhones don't use NFC chips compatible with libnfc (unless maybe if you connect a compatible reader via USB OTG)

gyx99, 2017-06-12:
so can i use usb otg ,pn532 and cp2012 and an android phone to use the tools
（my english isn't very goo）

doegox, 2017-06-12:
Some people did it before, I never tried myself. See e.g (probably outdated) https://drive.google.com/file/d/160clTJ-sSgHiNKd-IsFHGzZY8M9HQudYqHnp0mD09NBTPfZmm9mP_NEMV3E3/view
If you need assistance, please ask your questions on Stack Overflow, you will have more chance to get help.
Here, it's only about bugreports.
Thanks