nfc-tools/libnfc #418: undefined reference to pn53x_transceive

JONIDi12P, 2017-06-30:
i got error of undefined reference to pn53x_transceive' on file
https://github.com/nfc-tools/libnfc/blob/master/examples/pn53x-diagnose.c

on line 106+117

but i saw here
https://github.com/nfc-tools/libnfc/blob/master/libnfc/chips/pn53x.h
the header of function 
pn53x_transceive

that not depend on windows/linux
thanks

doegox, 2017-06-30:
Without any clue about how you proceeded I can't tell anything.
All I can say is that it works fine on a fresh Ubuntu, see https://github.com/nfc-tools/libnfc/issues/361#issuecomment-280784813 for commands

neryaz, 2017-06-30:
@doegox 
i tried on windows but undefined reference is not depend on OS 
(i don't know how your linux handle with that)
can you take a look on the code and maybe try to understand what is the problem?( that seem simple problem)...
thanks

smortex, 2017-06-30:
@JONIDi12P can you please tell us what build system you are using and post the output of make in verbose mode (`make V=1` with the autotools, `make VERBOSE=1` with cmake)?

JONIDi12P, 2017-06-30:
@smortex 
that you tried to help
i use with cmake, according to this tutorial
http://www.mobilefish.com/developer/libnfc/libnfc.html#configure_cmake

smortex, 2017-06-30:
@JONIDi12P please copy and paste here the full output of `mingw32-make VERBOSE=1`.  We really can't help you efficiently without this.  Thanks!

smortex, 2017-06-30:
Wow, reading a bit before build instructions, I see this "tutorial" targets an old version of libnfc.  Please reassure us, are you building from a recent version of git's master branch ?  Also, I would not be confident with the patches listed on this page.  Maybe better not apply them blindly.  Last (but not least), at the very end, copying the build dll into some system's directory looks a terrible idea for somebody like me who did not used Windows for the last ~15 years… :cold_sweat:

JONIDi12P, 2017-06-30:
@smortex 
thanks first.
i not get what i need to copy
do you want i run it `mingw32-make VERBOSE=1` on CMD and paste the output?
the run this command  `mingw32-make` i get this output

```

[ 19%] Built target nfc
[ 20%] Built target win32lib
[ 22%] Built target nfcutils
[ 26%] Built target nfc-list
[ 30%] Built target nfc-emulate-forum-tag4
[ 34%] Built target nfc-jewel
[ 39%] Built target nfc-mfclassic
[ 43%] Built target nfc-relay-picc
[ 48%] Built target nfc-mfultralight
[ 53%] Built target nfc-scan-device
[ 56%] Built target nfc-read-forum-tag3
[ 60%] Built target nfc-anticol
[ 63%] Built target nfc-dep-initiator
[ 67%] Built target nfc-dep-target
[ 71%] Built target nfc-emulate-forum-tag2
[ 74%] Built target nfc-emulate-tag
[ 78%] Built target nfc-emulate-uid
[ 81%] Built target nfc-mfsetuid
[ 85%] Built target nfc-poll
[ 89%] Built target nfc-relay
[ 90%] Linking C executable pn53x-diagnose.exe
CMakeFiles\pn53x-diagnose.dir/objects.a(pn53x-diagnose.c.obj): In function `main':
C:/SUPPORT_TOOLS/libnfc-master/examples/pn53x-diagnose.c:106: undefined reference to `pn53x_transceive'
C:/SUPPORT_TOOLS/libnfc-master/examples/pn53x-diagnose.c:117: undefined reference to `pn53x_transceive'
C:/SUPPORT_TOOLS/libnfc-master/examples/pn53x-diagnose.c:126: undefined reference to `pn53x_transceive'
collect2.exe: error: ld returned 1 exit status
examples\CMakeFiles\pn53x-diagnose.dir\build.make:117: recipe for target 'examples/pn53x-diagnose.exe' failed
mingw32-make[2]: *** [examples/pn53x-diagnose.exe] Error 1
CMakeFiles\Makefile2:954: recipe for target 'examples/CMakeFiles/pn53x-diagnose.dir/all' failed
mingw32-make[1]: *** [examples/CMakeFiles/pn53x-diagnose.dir/all] Error 2
Makefile:150: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
```

smortex, 2017-06-30:
>  do you want i run it `mingw32-make VERBOSE=1` on CMD and paste the output?

Yep! This should show us the complete command line.

JONIDi12P, 2017-06-30:
@smortex 
i sorry i not have this output now
you can't help with the output i put before ? (mingw32-make without VERBOSE=1)
thanks

smortex, 2017-06-30:
@doegox an idea: does'nt Windows require all symbols to be exported?  `pn53x_transceive()` is an internal function, which prototype appears in an [internal header](https://github.com/nfc-tools/libnfc/blob/dcdbff0705c8eabb8f055e8e7b9ef237430d6577/libnfc/chips/pn53x.h#L305).

Using this in an example seems boggus since it's an internal API function.  If it should be public API, maybe the name could be fixed to match `nfc_*` (or a wrapper function could be added) and the function prototype should be moved to  `nfc/nfc.h` after a `NFC_EXPORT`?

I don't have a Windows box available for testing this further.

@JONIDi12P if `pn53x_transceive` is the only symbol causing problem, I guess you would just be unable to build `pn53x-diagnose` and `pn53x-tamashell` without patching.

JONIDi12P, 2017-06-30:
@smortex thanks you!
i see that pn53x_transceive should by internal api and should not be export.
and you say that windows want that all simple will export.

so...
1- i get that pn53x is a chipset , so the libnfc know to working only on this chipset?
2-what i can do to fix that? i don't care even that need to be internal api that it export if it fix that...
i will be happy if you can help to fix that
thanks

doegox, 2017-07-01:
@smortex yes things wouldbe clearer if pn53x_transceive() was officially part of the API. Still it would be nice to make the distinction in the naming between "standard" API and "hacky" API, e.g. nfc_pn53x_transceive() ?

smortex, 2017-07-01:
Hum :thinking: 

In order to understand the intention, does it make sense to "protect" these API functions in order to avoid users for using them in the wild while more "standard" API should be used in place?

* If yes, maybe a new header could be introduced `#include <nfc/nfc-hacks.h>` or the prototype may be visible only if some macro is defined, e.g. `-DNFC_HACKS`;
* If no, we can just add if to `nfc/nfc.h`.

Regarding the name, I understool that you wanted `pn53x` to be kept on the name, right?

* If yes, I agree;
* If your point was something else, can you ellaborate?

I think we can keep the original name which is not in-line with the rest of the API but it's a hacky function so I see no problem; or we can prepend `nfc_` to align with the rest of the API and I am OK with this.

doegox, 2017-07-01:
I'm not really sure if we need to "hide" it or not. I've no strong opinion on that.
But ```pn53x``` must remain in the name because it will only work with those chipsets (ok ok I know the whole libnfc only supports pn53x for now but you see what I mean).