nfc-tools/libnfc #419: card reader support

jobika1lP, 2017-07-01:
Witch kind of card reader libnfc support?
Or that only important that the card will have the pn53x chip

and
I looking for example(or if you can write) witch function i need to call to read the data on the card

Thanks

jobika1lP, 2017-07-11:
@smortex @doegox

doegox, 2017-07-11:
You'll find this type of information on the website
http://nfc-tools.org/index.php?title=Category:Tags

closed 2017-07-11 by doegox

jobika1lP, 2017-07-12:
@doegox 
that only show the type of card  that good gor libnfc?(that not write on the page)
but even it is, no write there the type of card reader that support with libnfc
thanks!

doegox, 2017-07-12:
http://nfc-tools.org/index.php?title=Category:Devices

jobika1lP, 2017-07-13:
@doegox 
1-that only the card reader that libnfc know to work with them?
2-i see there only ACR122 or ACR122U(that both fix?)
3-so i can read all the card like
ISO14443
ISO14443A
ISO14443B
ISO14443B'
ISO15693
with 
ACR122U reader
with libnfc?

4- http://nfc-tools.org/index.php?title=PN53x
according to this i get that i can only read catd with PN53x
so if i connect the reader with usb i can't read all the card that have PN532 cheap
correct?

thanks

doegox, 2017-07-13:
1-yes
2-yes
3-no, not ISO15693, this is written here http://nfc-tools.org/index.php?title=ISO15693
4-I don't understand your question

jobika1lP, 2017-07-15:
@doegox 
i sorry that i was unclear on my last reply, i will try now to be more clearly
1- is libnfc support only on card reader ACR122U ?
this the only reader i saw here
http://nfc-tools.org/index.php?title=Category:Devices

2- i see on the implementation that libnfc know to work only with card that has PN53x cheap,
and i saw here
http://nfc-tools.org/index.php?title=PN53x
that if i connect my reader (according to 1. that only ACR122U ) to my pc with usb 
so i will not have a Connectivity with  card with PN532 cheap(so i can't read from them via libnfc)
and even i have Connectivity  via usb i can't read data if the card has PN531 cheap and support on ISO14443-B

thanks!

doegox, 2017-07-15:
1 no, on that page you'll also find http://nfc-tools.org/index.php?title=Devices_compatibility_matrix

2 almost any reader based on PN532 or PN533 is supported by libnfc. PN531 is somehow supported too  but this is a very old chip. But you need to configure libnfc according to your reader.

jobika1lP, 2017-07-15:
@doegox 
so thank you!
so if i want to use on ACR122U206 so it have PN532 cheap so if i want to connect via usb according to http://nfc-tools.org/index.php?title=PN53x i will not have a Connectivity so i can't read from card

2- you said that i need to configure libnfc according to your reader ,
where i need to do this?
thanks!

doegox, 2017-07-15:
start by telling exactly which reader you have

jobika1lP, 2017-07-15:
@doegox 
i use with windows and and compail all to dll (i install libusb)
but now i see that ACR122U206  not need lib usb it need PCSC so i need to try install it on windows

so let's say i will use scl3711 , how i need to configure  libnfc to my reader?

thanks

doegox, 2017-07-15:
under linux scl3711 doesn't need any configuration, just make sure pn533 kernel module is blacklisted.
On Windows I've no idea.

jobika1lP, 2017-07-18:
@doegox 
is there any image on linux that can i get with all in ?(libnfc libusb pn533 anable and all of this?)

doegox, 2017-07-18:
Not that I know of but any Debian should do it, you'll just need to ```apt-get install libnfc``` in it and make sure the kernel module pn533 is not in your way. 
Hmm well Kali has already libnfc but it's probably too specialized for you.

jobika1lP, 2017-07-24:
@doegox 
thank you about all!
1)on kali is there all in without i need to install anything?(from witch version of  kali?)
is there inside kali the PCSC, because if i want to connect and read with the reader ACR122U via usb i need PCSC on the os

2) on http://nfc-tools.org/index.php?title=PN53x
i always connect the reader to my pc with usb (how can i connect via UART?)
so as i see i don't understand
i see the the if the reader has PN532 cheaps so i don't have a Connectivity , that mean i can't read or write with that
but when i continue to read there i see that if the card is ISO14443-A i can read even it has PN532 cheap

thanks you!
--