nfc-tools/libnfc #423: compiling issue in myCustom programme in libnfc

Ali-Reza8, 2017-07-20:
Hi,

I've wanted to write my custom programm in c, before that I had tried to compile nfc-poll.c to see if everything is working well or not.

here it showed me this error:
```
/tmp/ccf9WAEC.o: In function `nfcforum_tag4_io':
nfc-emulate-forum-tag4_ME.c:(.text+0x7c): undefined reference to `print_hex'
nfc-emulate-forum-tag4_ME.c:(.text+0x6a4): undefined reference to `print_hex'
/tmp/ccf9WAEC.o: In function `stop_emulation':
nfc-emulate-forum-tag4_ME.c:(.text+0x71c): undefined reference to `nfc_abort_command'
nfc-emulate-forum-tag4_ME.c:(.text+0x730): undefined reference to `nfc_exit'
/tmp/ccf9WAEC.o: In function `main':
nfc-emulate-forum-tag4_ME.c:(.text+0xf8c): undefined reference to `nfc_init'
nfc-emulate-forum-tag4_ME.c:(.text+0xfc0): undefined reference to `nfc_open'
nfc-emulate-forum-tag4_ME.c:(.text+0xff4): undefined reference to `nfc_exit'
nfc-emulate-forum-tag4_ME.c:(.text+0x1018): undefined reference to `nfc_device_get_name'
nfc-emulate-forum-tag4_ME.c:(.text+0x1054): undefined reference to `nfc_emulate_target'
nfc-emulate-forum-tag4_ME.c:(.text+0x1074): undefined reference to `nfc_perror'
nfc-emulate-forum-tag4_ME.c:(.text+0x1084): undefined reference to `nfc_close'
nfc-emulate-forum-tag4_ME.c:(.text+0x1094): undefined reference to `nfc_exit'
nfc-emulate-forum-tag4_ME.c:(.text+0x1138): undefined reference to `nfc_close'
nfc-emulate-forum-tag4_ME.c:(.text+0x1148): undefined reference to `nfc_exit'
nfc-emulate-forum-tag4_ME.c:(.text+0x1160): undefined reference to `nfc_close'
nfc-emulate-forum-tag4_ME.c:(.text+0x1170): undefined reference to `nfc_exit'
collect2: error: ld returned 1 exit status
```

also tried to edit in this way:

```#include "../utils/nfc-utils.h"```
```#include <../utils/nfc-utils.h>```
```#include "nfc-utils.h"```
```#include <nfc-utils.h>```

but the same error appeared again.

what would I do?

tenznhok, 2019-01-17:
I have the same problem. i dont know if anyone get the answer

doegox, 2021-01-04:
you need to link with `-lnfc`

closed 2021-01-04 by doegox