nfc-tools/libnfc #425: nfc-list: ERROR: Unable to open NFC device: acr122_usb:003:002

frank-dspeed, 2017-07-31:
Tryed to use this lib via nodejs it worked now it don't works again
usbreset didn't resolve that

sudo LIBNFC_LOG_LEVEL=3 nfc-list
```
debug   libnfc.config   Unable to open directory: /etc/nfc/devices.d
debug   libnfc.general  log_level is set to 3

debug   libnfc.general  allow_autoscan is set to true
debug   libnfc.general  allow_intrusive_scan is set to false
debug   libnfc.general  0 device(s) defined by user
nfc-list uses libnfc 1.7.1
debug   libnfc.driver.acr122_usb        device found: Bus 003 Device 002 Name ACS ACR122
debug   libnfc.general  1 device(s) found using acr122_usb driver
debug   libnfc.general  0 device(s) found using pn53x_usb driver
debug   libnfc.driver.acr122_usb        3 element(s) have been decoded from "acr122_usb:003:002"
debug   libnfc.driver.acr122_usb        TX: 62 00 00 00 00 00 00 01 00 00
debug   libnfc.driver.acr122_usb        RX: 80 02 00 00 00 00 00 00 81 00 3b 00
debug   libnfc.driver.acr122_usb        ACR122 PICC Operating Parameters
debug   libnfc.driver.acr122_usb        TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00
debug   libnfc.driver.acr122_usb        RX: 80 02 00 00 00 00 00 00 81 00 90 00
debug   libnfc.chip.pn53x       GetFirmwareVersion
debug   libnfc.driver.acr122_usb        TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug   libnfc.driver.acr122_usb        RX: 80 00 00 00 00 00 00 00 fe 00
debug   libnfc.driver.acr122_usb        Command timed out
error   libnfc.driver.acr122_usb        PN532 init failed, trying again...
debug   libnfc.chip.pn53x       GetFirmwareVersion
debug   libnfc.driver.acr122_usb        TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug   libnfc.driver.acr122_usb        RX: 80 00 00 00 00 00 00 00 fe 00
debug   libnfc.driver.acr122_usb        Command timed out
error   libnfc.driver.acr122_usb        PN532 init failed, trying again...
debug   libnfc.chip.pn53x       GetFirmwareVersion
debug   libnfc.driver.acr122_usb        TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug   libnfc.general  Unable to open "acr122_usb:003:002".
nfc-list: ERROR: Unable to open NFC device: acr122_usb:003:002
```

keldnorman, 2019-08-03:
I have the same problem - it works flawlessly when I attach my acr122 reader directly to my raspberry pi zero w via USB but when I add a USB hub in between it fails with that same error you have after some time and can not be recovered without pulling out the reader and re-attaching it. I have tried resetting the USB bus, lowlevel commands and 10 other things even rebooting does not fix the problem. I wanted to avoid physically unplugging it and re-plug but not found a way to do it yet..
I am guessing here but It looks like that whatever is communicating with the acr122 reader using libnfc sends some special command on the USB bus to the reader that does not work if the reader is not alone there and then the reader driver or firmware ends up in "haywire" mode

neomilium, 2021-01-04:
Thanks for the details @keldnorman

closed 2021-01-04 by neomilium