nfc-tools/libnfc #426: Yet Another Unable to claim USB interface

ronnyandre, 2017-08-04:
I am trying the following commando, and it returns permission denied:

    % sudo nfc-list                                                                                                                                                              
    ~
    nfc-list uses libnfc 1.7.1
    error	libnfc.driver.acr122_usb	Unable to claim USB interface (Permission denied)
    nfc-list: ERROR: Unable to open NFC device: acr122_usb:250:006
    nfc-list: ERROR: Unable to open NFC device: acr122_pcsc:ACS ACR122U

I tried to disconnect and reconnect, but same message appear, again and again.

How can I disable the `acr122_usb`?

agebhar1, 2017-10-09:
Try to unload the `pn533` kernel module @ronnyandre, which is suggested on SO [Unable to claim USB interface (Device or resource busy)](https://stackoverflow.com/a/33790442).

> If you have kernel version > 3.5, nfs-list and pcscd will report this error due to the automatic load of pn533 driver.

```
$ lsmod | head -n 2
Module                  Size  Used by
pn533                  40960  0
nfc                   102400  1 pn533
$ nfc-list 
nfc-list uses libnfc 1.7.1
error   libnfc.driver.acr122_usb        Unable to claim USB interface (Device or resource busy)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:003:009
$ sudo rmmod pn533
$ sudo rmmod nfc
$ nfc-list 
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
```

dud225, 2018-02-14:
Hello

I'm getting this issue even though there is proper blacklist rules in /etc/modprobe.d. Don't understand...

maxmbed, 2018-09-08:
Yep, adding pn533_usb in blacklist-libnfc.conf worked for me too.

DavidBerdik, 2019-02-11:
Does making this change cause the LED and detection tone to not work? Making this change seems to have resolved the problem, but neither the LED or tone are working now.

graynk, 2019-04-29:
> Does making this change cause the LED and detection tone to not work? Making this change seems to have resolved the problem, but neither the LED or tone are working now.

I see this effect as well. And pcsc_scan reports card state as "card inserted" in all cases.

antjim, 2019-08-12:
I have modified the line 433 from drivers/acr122_usb.c to "res=0". Then, I have compiled libnfc and now, it's working! 

In fact, I have tried in 3 differents computers. All of computes it's working.

DavidBerdik, 2019-08-12:
@antjim Great! Do you know if this fixes the virtual machine problem too?

brole, 2019-08-12:
How are you compiling the nfc libs?

antjim, 2019-08-13:
> @antjim Great! Do you know if this fixes the virtual machine problem too?

Hi!
sorry, but in virtual machine I don't tried it. Only in a Lenovo ideapad 330 with Linux Mint 19.1, Asus K55A and Raspberry Pi B+ with Raspbian.

DavidBerdik, 2019-08-14:
> Hi!
> sorry, but in virtual machine I don't tried it. Only in a Lenovo ideapad 330 with Linux Mint 19.1, Asus K55A and Raspberry Pi B+ with Raspbian.

No worries. Thanks for your work! Would it be inconvenient for you to try it in a VM and let me know how it goes? Time-permitting, I can do it myself if you are unable to.

kylennn, 2019-10-13:
> > @antjim Great! Do you know if this fixes the virtual machine problem too?
> 
> Hi!
> sorry, but in virtual machine I don't tried it. Only in a Lenovo ideapad 330 with Linux Mint 19.1, Asus K55A and Raspberry Pi B+ with Raspbian.

I tried doing this on a VM without success.

I dont see how Res=0 would help at line 433 as you mention?

`if (res < 0) {
        log_put(LOG_GROUP, LOG_CATEGORY, NFC_LOG_PRIORITY_ERROR, "Unable to set alternate setting on USB interface (%s)", _usb_strerror(res));
        usb_close(data.pudh);
        **res=0;**
        goto free_mem;`

I will try on my Mac computer using virtual box.