nfc-tools/libnfc #427: Libnfc command on kali

jordanmv1c, 2017-08-12:
nfc-list show all the device that connected
Witch more command are usefull?
I want the command that i can type on terminal to read data from card

gyx99, 2017-08-13:
i use mfoc to read and use nfc-mfclassic to wirte(i can also write)

jordanmv1c, 2017-08-13:
@gyx99 could you please explain more?
Can you write here the command you type on the terminal??

(By the way how do you use it? You wrap it with gui?c#?)

Thank

gyx99, 2017-08-14:
i only have used mfoc to crack m1 card (with pn532)，i dont have other card to try
conmand: nfc-mfclassic r a new.dump old.dump
can read card with a current data file(old.dump) but you didnt crack it 
maybe you can ask the author of mfoc about your card
or maybe your card is security ,People can not crack him now

closed 2017-09-24 by jordanmv1c