nfc-tools/libnfc #428: no nfc device found

jobika1lP, 2017-08-13:
i on ubuntu
1) i saw on lsusb the read(ACR)
but on nfc-list i get no nfc device found

what can i do?

2) if i not see the device on lsusb what can i do?

3) i will be happy to get more command like nfc-list and lsusb
witch command i need to read /write from card?

gyx99, 2017-08-14:
i dont have a computer  with a linux now ,so something may be wrong
i used pn532 with cp2012 had same problems
is /dev/ttyUSB0 exist ？
if it is existing ,
vi /etc/libnfc/libnfc.conf
and delete a # which line has "ttyUSB0"
if not ,check your device  for good contact

if the card is m1 card you can use mfoc to read ,("mfoc h")can help you
nfc-mfclassic can read and write

neomilium, 2021-01-04:
No reply from author, closing the issue.

closed 2021-01-04 by neomilium