nfc-tools/libnfc #433: abtAts not correct value

jordanmv1c, 2017-08-15:
When i want to read and done nfc_initiator_poll_target  i get on abtAts not crrect value
But on abtUid i get it well

Why?

smortex, 2017-08-16:
Can you please provide a sample application that reproduces the error?

jordanmv1c, 2017-09-01:
@smortex  like quick_start_example1.c
i get szUidLen & abtUid correct
but szAtsLen is always 0
on abtAts i need to get all the data on the card, no?

doegox, 2017-09-01:
abtAts contains ATS string only if the card emits ATS, i.e. is ISO14443A-4 compatible (the -4 is important).
Try "nfc-list -v"

On a ISO14443-4 you get a line with: (ATS value may vary)
```
* Compliant with ISO/IEC 14443-4
                ATS: 75  77  81  02  80  
```

On an ISO14443-3 you get a line with
```
* Not compliant with ISO/IEC 14443-4
```

jordanmv1c, 2017-09-02:
Ohh sorry...I thought that ats us the data on the card
I will seach what ats mean.

So how can i please get the data on the catd?

doegox, 2017-09-02:
It entirely depends on the type of card you want to read.
E.g. there is nfc-mfultralight to read Ultralight cards, nfc-mfclassic for Mifare Classic cards, etc.

closed 2017-09-02 by doegox