nfc-tools/libnfc #436: sri 512 with libnfc ?

userDevop123, 2017-08-29:
Is libnfc work with sri 512 card?

I follow on this example : https://github.com/nfc-tools/libnfc/blob/master/examples/doc/quick_start_example1.c

but when the code arrive to line 56 it stack  `if (nfc_initiator_init(pnd) < 0) {`

why? libnfc not know how to work with sri 512 ?

thank

doegox, 2017-08-29:
line 56 is just the initialization of the reader.
If it halts there, the problem is not the tag.
quick_start_example.c was written for a tag ISO14443-A, does it work with such a tag on your side?
Then you'll have to change the nfc_modulation as in https://github.com/nfc-tools/libnfc/blob/master/utils/nfc-list.c#L243-L255 and to print the info, cf https://github.com/nfc-tools/libnfc/blob/master/utils/nfc-utils.c#L123-L129

userDevop123, 2017-08-29:
@doegox My card is 14443B .
Can you please show me how to read data from this card if example1 is not good for this type?
thank!

userDevop123, 2017-08-29:
@doegox  I update the issue
line 56 is ok but `nfc_initiator_select_passive_target` stack for me to few seconds...
it iso1444B

doegox, 2017-08-29:
If your card is a SRI then you need to use the modulation I've shown to you in the code above: NMT_ISO14443B2SR

userDevop123, 2017-08-30:
@doegox it tried NMT_ISO14443B2SR.
same problem.
by the way, i use on the newes version,but from witch version my card  suppot to work?
thank

doegox, 2017-08-31:
I forgot to tell you've to use the regular type B first.
Here is a fully working example:
https://gist.github.com/doegox/5205cdad9862e74af75304d599fcbd8f

userDevop123, 2017-09-01:
@doegox 
thank you man, i will try it( i saw the issues you reference to here)
can you tell me on witch card from ISO14443B2SR and on witch reader do you check it?
by the way, the nfc_initiator_init by the source code is init the the device as a reader with iso 14443-A
`res = nfc_device_set_property_bool(pnd, NP_FORCE_ISO14443_A, true
`thank again!

doegox, 2017-09-01:
I used a SCL3711 but any libnfc-compatible reader should work fine.
My code works on all samples I have:
* Seoul Metro tickets
* Brussels transport (STIB) tickets
* Lisbonne transport tickets
* SRIX4k sample
* SRIX4k Onity hotel room card

All UID are apparently 8 bytes long and last 2 bytes are always "02d0".
But I don't know much about these tags.

I'll check the init...

doegox, 2017-09-01:
No, the problem is not the ```nfc_device_set_property_bool(pnd, NP_FORCE_ISO14443_A, true)``` in nfc_initiator_init because writing B there doesn't help and there are already NP_FORCE_ISO14443_B for ISO14443B2SR.
The problem is that some internal registers are configured for B only when the Tama command InListPassiveTarget(B) is used and we don't know yet which ones to set them ourselves, so we've to call InListPassiveTarget(B) before being able to talk to ISO14443B2SR cards

userDevop123, 2017-09-02:
@doegox THANK YOU!
i saw issue 433 and i want to ask about `ISO14443B2SR `card like sri 512 srix4k,
how can I read the data from the card
`nfc-mfultralight` and `nfc-mfclassic` not good for this type of card

doegox, 2017-09-02:
I don't know how to read data from that card.
Find the specs and implement the commands yourself, that's all I can say.

userDevop123, 2017-09-03:
@doegox 
I use your code but nfc_initiator_list_passive_targets stack for 10-15seconds and return 0
Then nfc_initiator_select_passive_target return-20

doegox, 2017-09-03:
Maybe your card is not a ST SRi, maybe it's not working, maybe... Well you're basically on your own now to debug the situation.
You can set LIBNFC_LOG_LEVEL=3 in your environment to get more info and www.nxp.com/docs/en/user-guide/141520.pdf to understand the commands/responses exchanged with the PN532 chip.

userDevop123, 2017-09-04:
@doegox my card is st Sr and it working(i check it on another reader not on libnfc)
i run it on 1.6 , mybe it the problem?(i check on sorce code, i dont think it the problem)  i can't upgdrade fo 1.71 now

closed 2017-09-06 by userDevop123