nfc-tools/libnfc #438: Which Pn532 module ?

aminmrp, 2017-09-01:
Hi ...
i have question ... In libnfc ,There is a driver for PN532 module. Is this driver for the Chinese version of this module or Adafruit version?If this driver is for Adafruit module , can i use it for chinese module?
(Red module is chinese version and blue module is adafruit version )
Thank you for your attention

[image]
[image]

doegox, 2017-09-01:
Drivers talk directly to the PN532 chip so both board will work.
But beware antenna performance may vary a lot.
Antenna of Adafruit is very good (and there are 15$ clones of that one that work equally well).

aminmrp, 2017-09-02:
The NXP module has (I2C , SPI and UART ) . which is the best ? Which is easier to work with?

doegox, 2017-09-02:
They're all good.
It depends on what you want to connect it to.
E.g. for a laptop, UART with a USB-Serial cable is the esiest.
E.g. for a RPi, I2C or SPI on its GPIO port are the easiest.

aminmrp, 2017-09-02:
@doegox Oh sir ... If I want to connect it with SPI to Libnfc, does Libnfc know it as PN532 module or RPi?

aminmrp, 2017-09-02:
@doegox Does the library recognize the NXP PN532 automatically on a laptop??? I use kali linux

aminmrp, 2017-09-02:
@doegox  Oh sir ... i have a more question 😅 if i wanna connect NXP PN532 module to laptop (UART mode), Is there a special cable for connecting it?

doegox, 2017-09-02:
any usb-uart cable should work.
just make sure it's recognized by linux but most cables are.

closed 2017-09-02 by doegox

doegox, 2017-09-02:
to get the board recognized over USB-UART, you've to edit libnfc.conf, see last lines of

https://github.com/nfc-tools/libnfc/blob/master/libnfc.conf.sample