nfc-tools/libnfc #441: uid not correct while read srx4k

Usb1NfcReader, 2017-09-04:
#436 i use [link](https://gist.github.com/doegox/5205cdad9862e74af75304d599fcbd8f) with acr122u
 the card is good.

on` nfc_initiator_list_passive_targets` it takes so much time(like 15 sec.) and return 0
`nfc_initiator_select_passive_target` pass fast and return 1(good )
but abtUid contain only 1 byte that correct(the rest of byte is 0 if i remebmer ok)
and szUidLen is big numner(3000+ if i remember well)

another way to read , i read the datasheet of srx4k i saw the command to get the uid
i saw that before i can ask the uid i need to call select and get the chipId
where on libnfc i get the chipId ?

doegox, 2017-09-04:
Are you using the latest libnfc?

Usb1NfcReader, 2017-09-04:
no my freind. this the first post i post here.
i will be happy if you will answer me on the 2 questions

thanks!

doegox, 2017-09-04:
then please first install the latest libnfc version, there were many changes and it will be much easier to debug if you're using the version I wrote the gist for.

Usb1NfcReader, 2017-09-05:
@doegox 
ok and what about second question

Can i use with libnfc after nfc_open with nfc_initiator_transceive_bytes to make the command:
Initiate()(to get chipid)
then Select(ChipID)
then Get_UID() ?

and not use in nfc_initiator_list_passive_targets and nfc_initiator_select_passive_target ?

doegox, 2017-09-05:
You will need to call nfc_initiator_list_passive_targets with NMT_ISO14443B first anyway because this is the only way we know to set the chip in TypeB mode properly.
Then indeed you can send raw commands yourself instead of using nfc_initiator_select_passive_target with NMT_ISO14443B2SR and do the same as https://github.com/nfc-tools/libnfc/blob/master/libnfc/chips/pn53x.c#L1091-L1108 manually

Usb1NfcReader, 2017-09-05:
thank you about all! @doegox 
if you can help me with last problem
i read the code and read the datasheet,  the init command is 4 bytes 0x06 0x00 crcL crcH
i saw there is a function to calculate the crc of 14443b but i don't understand how to use it
if you can please show me how to "send" to function the 0x06 0x00 and get the crcL and crcH please i will thank you!

doegox, 2017-09-05:
iso14443b_crc() takes a data buffer, its size, and a 2-byte buffer where the function will write the CRC.
iso14443b_crc_append() is usually easier, it directly appends the CRC (provided that the buffer has room for these 2 bytes)

So e.g.
```
uint8_t buf[4]={0x06, 0x00}
iso14443b_crc_append(buf, 2);
// now buf contains 0x06 0x00 CRCH CRCL
```

closed 2017-09-05 by doegox

Usb1NfcReader, 2017-09-13:
@doegox  

> You will need to call nfc_initiator_list_passive_targets with NMT_ISO14443B first anyway because this is the only way we know to set the chip in TypeB mode properly.

i dont want to run all `nfc_initiator_list_passive_targets(B)`
can you show me the exactly code i need to run to config the chip first to Type B?(and then i will call to `nfc_initiator_list_passive_targets(B2sr)`   )