nfc-tools/libnfc #443: question: pn53x_initiator_transceive_bytes when no return buffer

Pnx2Error, 2017-09-06:
[pn53x_initiator_transceive_bytes](https://github.com/nfc-tools/libnfc/blob/master/libnfc/chips/pn53x.c#L1516)
i want to send this function that no expect to return value(like write on tag)
so i send to this function `uint8_t abtRx[0]`
the command is done, but i get on line 1516 res<0 
why? what is to right way to send to pn53x_initiator_transceive_bytes abtRx when i not expect to response?

another way
i don't see when this function clculat the crc that fix to the type of the tag, before it send the command to tag

doegox, 2017-09-06:
You can pass NULL as pbtRx and 0 as szRx if you don't need the response.
Most commands get a response, at least an ack, are you sure the card is not supposed to reply at all?
In such case, check the error, if it's a timeout then it's fine.

CRC gets calculated and appended automatically by the PN533 if NP_HANDLE_CRC is true, which is the case by default.

Pnx2Error, 2017-09-06:
yeh, i sure that my tag not get any response on write command.
i will check if i get -6(timeout)... (what was strange to me is the command done but i get res<0)

about crc, i use on reader on PN532 not pn533, 
so what i looking for is where the code calculate and append the crc and where it check the crc on the response
i can't find if

doegox, 2017-09-07:
About CRC, PN532 and PN533 work the same.
https://github.com/nfc-tools/libnfc/issues/441#issuecomment-327281239

Pnx2Error, 2017-09-08:
@doegox 
1)I can't pass null as `pbtRx`, `pn53x_initiator_transceive_bytes` return -20 
but the command is done.

2) about the crc, that not what i looking for 
`pn53x_initiator_transceive_bytes` get `pbtTx` , that the command without the crc
this function need to
a) append the crc and send the command with crc
b) get the response with crc and check if crc is ok

i not find where this function do each 1 of this .