nfc-tools/libnfc #446: Memory Leak in  quick_start_example1.c

dkotrada, 2017-09-21:
http://www.libnfc.org/api/examples_page.html
Valgrind gives me this output

~~~.cmd
==11496== LEAK SUMMARY:
==11496==    definitely lost: 8 bytes in 1 blocks
==11496==    indirectly lost: 0 bytes in 0 blocks
==11496==      possibly lost: 0 bytes in 0 blocks
==11496==    still reachable: 40,760 bytes in 60 blocks
==11496==         suppressed: 0 bytes in 0 blocks
==11496== 
==11496== For counts of detected and suppressed errors, rerun with: -v
==11496== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 1 from 1)
~~~

doegox, 2017-10-04:
I'm afraid it's not in libnfc.
Rerun with --leak-check=full to see details of leaked memory

doegox, 2021-01-04:
closing by lack of response

closed 2021-01-04 by doegox