nfc-tools/libnfc #455: Unsupported nfc device in lenovo T470

Eik-S, 2017-10-04:
Unfortunately, the nfc module of lenovo T470 is not yet supported.

Neither lsusb is showing the device:
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 138a:0097 Validity Sensors, Inc. 
Bus 001 Device 004: ID 5986:111c Acer, Inc 
Bus 001 Device 002: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Is there any progress on this? 
Thanks.

totakura, 2017-10-26:
I believe the NFC module is connected via I2C bus. Lets see if it is being detected there; can you install `i2c-tools` and run

    i2cdetect -l

Eik-S, 2017-10-26:
Output is:
```
i2c-3	unknown   	DPDDC-A                         	N/A
i2c-1	unknown   	i915 gmbus dpb                  	N/A
i2c-6	unknown   	Synopsys DesignWare I2C adapter 	N/A
i2c-4	unknown   	DPDDC-B                         	N/A
i2c-2	unknown   	i915 gmbus dpd                  	N/A
i2c-0	unknown   	i915 gmbus dpc                  	N/A
i2c-5	unknown   	DPDDC-C                         	N/A
```

totakura, 2017-11-06:
My only suspicion is the Synopsis DesignWare I2C adapter; I wonder what that is.  The rest are I2C inputs for plug-and-play monitors.

Do you also dual boot Windows?  If yes, could you please check the bus under which it is being detected in the device manager?

derTobsch, 2017-11-22:
I have the same problem and my output is

```
i2c-3	unknown   	DPDDC-A                         	N/A
i2c-1	unknown   	i915 gmbus dpb                  	N/A
i2c-6	unknown   	Synopsys DesignWare I2C adapter 	N/A
i2c-4	unknown   	DPDDC-B                         	N/A
i2c-2	unknown   	i915 gmbus dpd                  	N/A
i2c-0	unknown   	i915 gmbus dpc                  	N/A
i2c-5	unknown   	DPDDC-C                         	N/A
```
I do not have a dual boot with windows. Only ubuntu 17.10 on a t470s

derTobsch, 2017-11-27:
if someone needs support or information from the system please ask. It would be nice if there will be a driver for this nfc module.

lbeltrame, 2017-12-13:
The device manager reports the NFC reader as NXP Semiconductors(Proximity) NXP Near Field Proximity Provider. Its "location" is under an I2C controller, "Serial I/O I2C Intel Host Controller - 9D60" (translated from my Language so it may be different from the English version). The "bus number" in the device manager is 0, the "BIOS name" is "\_SB.PCI0.I2C0.NFC1".

The I2C controller listed (Intel Host Controller) is under PCI bus 0, device 21. 

I'll keep Windows here for a couple more days before wiping it, so if need be, I can provide more details.

Roxxor91, 2018-01-17:
Hey! I would like to participate in solving this issue!
From the FRU list (https://download.lenovo.com/parts/ThinkPad/t470_frubom_20170731.pdf) I found that the NFC is this one: https://fccid.io/MCLT77H747/Users-Manual/User-Manual-rev-4-3127142.html
FOXCONN Part No. T77H747.10 
-> NFC (NXP NPC300) Module

tuxflo, 2018-01-20:
The same NFC/smartcard reader is installed in the Lenovo Thinkpad Yoga 260.
Since the driver seems not to work with linux my question is: is the (compiled) Windows version of libnfc able to use the Windows drivers (the device shows status "Usable" in the device manager)?

iSDP, 2018-04-21:
Same story in the T580, also the Foxconn NXP NPC300.

Wish it worked :(

RaphaelWimmer, 2018-06-12:
Some more details and clarifications:
- NXP does not have any info on a "NPC300" chip/solution. It seems to me that this is just another name for an NFC chip from the PN7xxx range (or a name for a combination of chip and antenna).
- NXP does not seem to have support libnfc but provides their own [libnfc-nci](https://github.com/NXPNFCLinux/linux_libnfc-nci) for Linux [according to these interesting Webinar slides](http://www.themobileknowledge.com/wp-content/uploads/2017/09/02-Webinar-slides-How-to-integrate-NFC-controllers-in-Linux.pdf).
- There are some tips on compiling the libnfc-nci kernel driver in the Webinar slides and [NXP application note AN11697](https://www.nxp.com/docs/en/application-note/AN11697.pdf).
- However, there is a major problem: the chip needs to be enabled (activated) through a GPIO pin that drives ``VEN`` high. AFAICT, the device will only appear on the I2C bus if it is activated in this way. Unless you have the schematics for your motherboard, the only way to find out which of the CPUs GPIO pins is connected to the NXP chip is to activate every single one of them and check whether the NXP chip appears on the I2C bus. As lots of other stuff might be connected to these GPIO pins, I'm a little bit hesitant to try this out.
- It seems that the NXP chips usually have an I2C address of 0x28 or 0x29. So, if the chip is enabled, ``i2cdetect`` should detect a device at this address.

The correct way to find out which I2C devices are connected to an I2C bus (and are responding) is the following:

``modprobe i2c-dev      # to make the I2C buses available on /dev
i2cdetect -l     # to find all I2C buses 
i2cdetect 0    # to find out which devices are connected to bus i2c-0``

Despite the warning that "This program can confuse your I2C bus, cause data loss and worse!", probing the I2C bus this way has never caused any problems for me.

Most of the I2C buses you will see with ``i2cdetect -l`` belong to the graphics chip (e.g., everything with "DDC" in the name). The most probable bus to find the NFC chip would be the SMBus (``i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter``) in my case.

pccr10001, 2018-06-24:
Hi. I found something in my DSDT table.
According to the picture, I think `0x0029` is the I2C address, and GPIO lists are the pins definition.
The GPIO pin `0x006C` is active high, is it `VEN` pin?

[image]

RaphaelWimmer, 2018-06-24:
Oh, I didn't know that this can be stored in the DSDT (unfortunately, the DSDT on my HP Elitebook 1030 G2 only contains "0x0000" as pin lists). 
I guess GPIO 0x6C is the connected to the IRQ pin of the NFC module which notifies the kernel that a NFC chip has been detected. GPIO 0x27 and 0x64 are probably connected to VEN and DWL_REQ (enable firmware update mode - potentially dangerous to keep enabled).

jr64, 2018-08-08:
I have a Thinkpad T480s which uses the same NPC300 chip and I partially succeeded getting it to run under Ubuntu 18.04.
Libnfc-nci (https://github.com/NXPNFCLinux/linux_libnfc-nci/) has a mode where it uses /sys/class/gpio to communicate with the hardware. Since that seemed easier to play around with than a kernel module, I decided to start with that.

My DSDT is the same as the one @pccr10001 posted, so I guess the configuration is the same.

I edited the pins in src/halimpl/pn54x/tml/i2c/phTmlNfc_alt.h like this:
```
 #define I2C_BUS         "/dev/i2c-6"
 #define I2C_ADDRESS     0x29
 #define PIN_INT         360+0x6c
 #define PIN_ENABLE      360+0x64
```
I got the bus number from ic2detect -l which looks like this:
```
i2c-3	unknown   	DPDDC-A                         	N/A
i2c-1	unknown   	i915 gmbus dpb                  	N/A
i2c-6	unknown   	Synopsys DesignWare I2C adapter 	N/A
i2c-4	unknown   	DPDDC-B                         	N/A
i2c-2	unknown   	i915 gmbus dpd                  	N/A
i2c-0	unknown   	i915 gmbus dpc                  	N/A
i2c-5	unknown   	DPDDC-C                         	N/A
```
Bus 6 was the only one that made sense to me and indeed it worked.

The i2c address and ports are from the DSDT, finally the offset 360 is from /sys/class/gpio/gpiochip360/base

Afterwards I configured the library like this (not sure if it is a pn7150 or pn7120 but I guess 7150 because according to data sheets I found the chip supports tags of type 5 which the 7120 does not):
```
./configure --enable-pn7150 --enable-alt --sysconfdir=/etc
```
Then I ran make and make install and finally was able to start the demo app:
```
 sudo ./nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	NFC Tag Found

		Type : 		'Type A - Mifare Classic'
		NFCID1 :    	'XX XX XX XX '
		NDEF Content : NO, mode=1, tech=8

		RAW Tag transceive failed
		Type : 		'Type A - Mifare Classic'
		NFCID1 :    	'XX XX XX XX '
		NDEF Content : NO, mode=1, tech=8

		RAW Tag transceive failed
		Type : 		'Type A - Mifare Classic'
		NFCID1 :    	'XX XX XX XX '
		NDEF Content : NO, mode=1, tech=8

		RAW Tag transceive failed
		Type : 		'Type A - Mifare Classic'
		NFCID1 :    	'XX XX XX XX '
		NDEF Content : NO, mode=1, tech=8

		RAW Tag transceive failed
	NFC Tag Lost

Waiting for a Tag/Device...
```
As you can see it correctly detected the Mifare classic card I tested. I'm not sure if the RAW Tag transceive failed means that something is still not working. Maybe my card is not a standard Mifare Classic but a modified version, I'm not sure and don't have anything else to test. 

I guess the next step would be to try to get the proper kernel module to run, it shouldn't be too hard with the configuration already known. 

One thing I noticed is that the touchpad does not seem to play well with the nfc driver. whenever I start it, the touchpad becomes laggy, the mouse pointer sometimes jumps around and acceleration seems to stop working. This persists for a few seconds, then everything works again and then it starts being buggy again. This even persists after I stop the nfc demo until I reboot the system. I don't have much experience with i2c so if anyone could point me in the right direction that would be great.

Also I have never worked with NFC before, am I correct in assuming that libnfc and libnfc-nci are not compatible so even if we got it to work, most of the software out there would have to be heavily modified to work on our machines?

mvdnes, 2018-08-08:
@jr64 I can confirm this also works on my regular T480 with Arch Linux. Same DSDT file, and the i2c bus has the same name altough it is sometimes on `/dev/i2c-0` and sometimes on `/dev/i2c-1`.

The demo app works with poll, and write on a small tag, and share also works with my Android phone.

I would also like to point out the path `/sys/bus/i2c/devices/i2c-NXP1001:00/`, which is present on my system. Maybe that can be used to determine the i2c bus more reliably.

pccr10001, 2018-08-09:
@jc64 This solution works on my X1C and card emulation works too, I am using Ubuntu 18.04.1.

jr64, 2018-08-09:
Update: I just modified the i2c kernel driver (https://github.com/NXPNFCLinux/nxp-pn5xx) to support ACPI and autoconfigure itself. Seems to work fine to me. I'll post the code as soon as I have cleaned it up.

However, this still does not fix the touchpad issues I am experiencing. Does anyone else have this problem? Any ideas how to fix? I'm assuming the touchpad is also connected via i2c and the driver is somehow interfering with it?

jr64, 2018-08-10:
For anyone interested, I just uploaded the first version of my modified kernel driver: https://github.com/jr64/nxp-pn5xx

pccr10001, 2018-08-10:
@jr64 Here is the solution of the trackpoint problem.
My trackpoint is working now.

https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_X1_Carbon_(Gen_6)#TrackPoint_and_Touchpad_issues

jr64, 2018-08-10:
@pccr10001 Thanks, the seemed really promising but unfortunately it looks like they put an Elan touchpad and not a Synaptics in the T480s.

To quote from https://patchwork.kernel.org/patch/10330857/:

"- the T480s is using an Elan touchpad, so that's a different story"

Luckily it looks like there is work being done on Elan too (https://patchwork.kernel.org/patch/10324633/) but it's not yet included in the kernel I'm running. 

Anyways, at least I'm fairly certain now that the problem is with the touchpad driver and not with the code I wrote for the NFC module.

jr64, 2018-08-10:
Oh and another funny detail: the only way I found to fix the slow trackpad without rebooting the system is to close and open the notebook. Reloading the psmouse driver seems to do nothing, but putting the device to sleep and waking it up again fixes it permanently (well at least until I use NFC again).

spielkind, 2018-08-13:
```
I'm still a bit confused, I compiled the DemoApp with the manual GPIO changes, and it worked:
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...

	NFC Tag Found

        Type :         'Type A'
        NFCID1 :    	'04 22 2B C2 B6 46 80 '
		NDEF Content : NO, mode=1, tech=1

	Not a MIFARE card
	NFC Tag Lost

Waiting for a Tag/Device...
```

I'm on a X280: 
`/sys/bus/i2c/devices/i2c-NXP1001:00 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-7/i2c-NXP1001:00`

But why would I need the kernel module adjusted by @jr64 , and how can I confirm that it worked after loading. Also are there any client tools, beside the demoapp, which work right now?

jr64, 2018-08-13:
@spielkind The kernel module has two advantages:
1. it configures itself from ACPI so no need to manually find out the correct GPIO ports
2. driving GPIO pins from userland through /sys/class/gpio is kind of "hacky", a kernel module is the "proper" variant for such a task. But of course, at the end of the day, both work fine.

As for how to test: if you configure the library with "--enable-i2c" instead of "--enable-alt" it will only work if the kernel modules is loaded.

blasty, 2018-11-02:
Thanks @jr64 and others for the progress on this. I've gotten this to work (sort of) on a T480s with @jr64's kernel module and `libnfc-nci-R2.2`. However, with the `nfcDemoApp` (in `poll` mode) I have a very poor detection rate of tags, and haven't seen any succesfull NDEF record reading yet. Has anyone experienced similar issues?

dlezcano, 2018-12-04:
@jr64 : did you plan to upstream the driver ? I tried it on a x280 and it works as far as I can tell. That would be an improvement of the x280 support on Linux.

dlezcano, 2018-12-04:
I have been throught the upstream kernel and found the following commit:

commit 0a5942c8e1480db4b8ee7a8d643e4945ef2f8fed
Author: Robert Dolca <robert.dolca@intel.com>
Date:   Mon Jan 26 13:13:37 2015 +0200

    NFC: Add ACPI support for NXP PN544
    
    Currently there is no support for ACPI.
    This patch uses the following configuration:
            - Device id: NXP5440
            - Pin mapping:
                    - 0 IRQ pin
                    - 1 enable pin
                    - 2 firmware pin
    
    Signed-off-by: Robert Dolca <robert.dolca@intel.com>
    Signed-off-by: Samuel Ortiz <sameo@linux.intel.com>

So at this point I assume the ACPI is supported in mainline since 2015. But if we look closely at the ACPI table and match hid we have in this driver: 

static const struct acpi_device_id pn544_hci_i2c_acpi_match[] = {
       {"NXP5440", 0},
       {}
};

But on my x280, the ACPI table is:
[ ... ]
Name (_HID, "NXP1001")  // _HID: Hardware ID
[ ... ]
Except I'm missing something, it should be NXP5440.

dlezcano, 2018-12-04:
dsdt table looks buggy:
 acpidump > acpidump
 acpixtract -a acpidump
 iasl -d dsdt.dat
 iasl -tc dsdt.dsl

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20180105
Copyright (c) 2000 - 2018 Intel Corporation

Compiler aborting due to parser-detected syntax error(s)
dsdt.dsl    264:     Name (SS4, One)
Error    6126 -        ^ syntax error, unexpected PARSEOP_NAME

ASL Input:     dsdt.dsl - 34418 lines, 1065629 bytes, 15154 keywords
Hex Dump:      dsdt.hex - 230 bytes

Compilation complete. 1 Errors, 0 Warnings, 0 Remarks, 0 Optimizations

dlezcano, 2018-12-04:
And even worse, run:
  sudo fwts syntaxcheck -
and admire the result on my beautiful ultra expensive x280.

Test           |Pass |Fail |Abort|Warn |Skip |Info |
---------------+-----+-----+-----+-----+-----+-----+
syntaxcheck    |   11|   26|     |     |     |     |
---------------+-----+-----+-----+-----+-----+-----+
Total:         |   11|   26|    0|    0|    0|    0|
---------------+-----+-----+-----+-----+-----+-----+

WTF!

vvug, 2019-02-11:
@jr64 Same question here. Do you think your driver can be upstreamed? Thanks!

dileks, 2019-05-13:
Just want to let you know...

With the patch series "[PATCH v2 00/12] NFC: nxp-nci: clean up and support new ID" (see [1]) I was able to get my NFC chip NXP NFC300 ("NXP1001") on my Lenovo ThinkPad T470 run with the NFC nxp-nci driver shipped with Linux v5.1.1.

Furthermore, I can acces, list and poll with neard (daemon) v0.16 and neard-tools from Debian/buster. 

There is no need for nxp-pn5xx driver and libnfc-nci from NXP.

Before this I expermented with jr64 nxp-pn5xx (acpi support) and libnfc-nci from NXP and I saw in the logs: 

NxpHal:     phNxpNciHal_deriveChipType Hw Version (0x68) Not Found: Setting Default Chip Type : PN548C2
NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2

Hope this helps.

[1] https://patchwork.kernel.org/project/linux-wireless/list/

mweinelt, 2019-05-15:
Cool, I applied the patchset against Linux 5.1.2 and the nfc adapter on my X1 C6 came to live.

With `nfctool --poll -d nfc0` it recognizes there is a token (tag0, tag1, ..., tagN), but there is little more information.

I have neard installed but it cannot currently find the default bluetooth adapter to pair a bluetooth headset, but at least it's trying to do that.

dileks, 2019-05-16:
@mweinelt 
Thanks for testing. Andy is preparing a v3 of his patchset.

What says...?
```
# rfkill --output ID,TYPE
ID TYPE
 0 bluetooth
 1 nfc
 2 wlan
 3 bluetooth

# rfkill list nfc
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
```

mweinelt, 2019-05-16:
Out is the same as yours. Looking forward to v3, but it looks like cleanups only.

mweinelt, 2019-05-17:
Not much luck with actually reading NFC tags.

```
neard[2300]: neard[2300]: NEAR daemon version 0.16
neard[2300]: NEAR daemon version 0.16
neard[2300]: neard[2300]: Could not get Bluetooth default adapter Method "DefaultAdapter" with signature "" on interface "org.bluez.Manager" doesn't exist
neard[2300]: Could not get Bluetooth default adapter Method "DefaultAdapter" with signature "" on interface "org.bluez.Manager" doesn't exist
neard[2300]: neard[2300]: Unknown EIR x03 (len: 17)
neard[2300]: neard[2300]: bt_get_default_adapter failed
neard[2300]: neard[2300]: ndef parsing failed
neard[2300]: Unknown EIR x03 (len: 17)
neard[2300]: bt_get_default_adapter failed
neard[2300]: ndef parsing failed
neard[2300]: neard[2300]: Error while reading NFC bytes
neard[2300]: Error while reading NFC bytes
```

mweinelt, 2019-05-18:
@dileks does reading nfc tags work for you?

makkarpov, 2019-05-31:
Hi!

I have Lenovo ThinkPad Carbon X1 6th Gen, and I tried to bring up NFC module according to the instructions above. Unfortunately, with no luck. DSDT shows same information as on above screenshots, but module does not work. 

I have Fedora 30, and first of all - there is no `/sys/class/gpio*` subtree. I tried to load modules named like `gpio*` found on my system, but with no luck. I had tried to compile and load module `pn5xx_i2c` listed above --- it compiled and loaded successfully, determined GPIO pins:

```
[   53.944640] pn54x_get_pdata: request irq_gpio 468
[   53.944773] pn54x_get_pdata: request ven_gpio 460
[   53.946014] pn54x_probe : requesting IRQ 144
```

But NFC demo app (with `--enable-i2c` or `--enable-alt`, it does not matter - same error) shows:

```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

SNEP Client Register Callback Failed
Leaving ...
```

Additionally `./configure` on NFC lib no longer accept `--enable-pn7150`, but I think this is because it is already enabled in all configurations.

dileks, 2019-08-05:
v4 of the patchset was included in net-next Git.

[1] https://git.kernel.org/pub/scm/linux/kernel/git/davem/net-next.git

dileks, 2019-08-20:
@mweinelt 

I am testing the patchset v4 against Linux v5.3-rc5 with Debian neard v0.16:
```
# lsmod | egrep 'nfc|nci|nxp' | sort
nci                    73728  2 nxp_nci,nxp_nci_i2c
nfc                   118784  14 nci,nxp_nci
nxp_nci                16384  1 nxp_nci_i2c
nxp_nci_i2c            16384  0
rfkill                 32768  10 nfc,bluetooth,thinkpad_acpi,cfg80211

# modinfo nxp_nci_i2c
filename:       /lib/modules/5.3.0-rc5-3-amd64-cbl-asmgoto/kernel/drivers/nfc/nxp-nci/nxp-nci_i2c.ko
license:        GPL
description:    I2C driver for NXP NCI NFC controllers
author:         Clément Perrochaud <clement.perrochaud@nxp.com>
author:         Oleg Zhurakivskyy <oleg.zhurakivskyy@intel.com>
vermagic:       5.3.0-rc5-3-amd64-cbl-asmgoto SMP mod_unload modversions 
name:           nxp_nci_i2c
intree:         Y
retpoline:      Y
depends:        nci,nxp-nci
alias:          i2c:nxp-nci_i2c
alias:          of:N*T*Cnxp,nxp-nci-i2c
alias:          of:N*T*Cnxp,nxp-nci-i2cC*
alias:          acpi*:NXP1001:*
alias:          acpi*:NXP7471:*

# cd /usr/lib/neard/tools

# ./test-adapter list

# ./test-adapter powered nfc0 on
# ./test-adapter poll nfc0 on

# ./test-adapter list
 [ /org/neard/nfc0 ]
    Protocols = [ Felica MIFARE Jewel ISO-DEP NFC-DEP ]
    Mode = Idle
    Polling = true
    Powered = true

# ./monitor-near 
{Adapter.PropertyChanged} [/org/neard/nfc0] Mode = Initiator
{Adapter.PropertyChanged} [/org/neard/nfc0] Polling = 0
{Added org.neard.Tag} [/org/neard/nfc0/tag5]
      Protocol = ISO-DEP
      ReadOnly = 0
      Type = Type 4A
      Adapter = /org/neard/nfc0
{Added org.freedesktop.DBus.Introspectable} [/org/neard/nfc0/tag5]
{Added org.freedesktop.DBus.Properties} [/org/neard/nfc0/tag5]
{Removed org.freedesktop.DBus.Properties} [/org/neard/nfc0/tag5]
{Removed org.freedesktop.DBus.Introspectable} [/org/neard/nfc0/tag5]
{Removed org.neard.Tag} [/org/neard/nfc0/tag5]
{Adapter.PropertyChanged} [/org/neard/nfc0] Mode = Idle

# journalctl -u neard.service -f
Aug 20 22:32:54 iniza neard[6158]: Error while reading NFC bytes
```

dileks, 2019-08-20:
Here the same with nfctool command (which I think is a wrapper to above commands):
```
# nfctool --list

# nfctool --enable --device=nfc0

# nfctool --list --device=nfc0
nfc0:
          Tags: [ tag11 ]
          Devices: [ ]
          Protocols: [ Felica MIFARE Jewel ISO-DEP NFC-DEP ]
          Powered: Yes
          RF Mode: Initiator
          lto: 0
          rw: 0
          miux: 0

# nfctool --device=nfc0 --poll=Both --sniff --dump-symm
Start sniffer on nfc0

Start polling on nfc0 as both initiator and target

Targets found for nfc0
  Tags: [ tag11 ]
  Devices: [ ]

# journalctl -u neard.service -f
Aug 20 23:01:15 iniza neard[6158]: Error while reading NFC bytes
```

dileks, 2019-08-20:
Get the full patchset from net-next.git as a single patch...

```
wget -O nfc-nxp-nci-v4.patch "https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/patch/?id=e9e1dcda2e64281a6b66f136b79703763d93771c"

```

[1] https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/patch/?id=e9e1dcda2e64281a6b66f136b79703763d93771c

mweinelt, 2019-08-21:
Patch applied against 5.2.9 on NixOS.

```
nfctool --device=nfc0 --poll=Both --sniff --dump-symm
Start sniffer on nfc0

Start polling on nfc0 as both initiator and target

Targets found for nfc0
  Tags: [ tag5 ]
  Devices: [ device4 ]

 << nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

 >> nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

 << nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

 >> nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

 << nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

 >> nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

 << nfc0: local:0x00 remote:0x00 
    Symmetry (SYMM) 

[...]
```

Aha, no idea what all that means. At the same time:

```
journalctl -f -u neard
-- Logs begin at Wed 2019-07-24 00:19:18 CEST. --
Aug 21 20:38:48 nyx neard[3043]: neard[3043]: Error with Handover server channel
Aug 21 20:38:48 nyx neard[3043]: neard[3043]: Error with VALIDATION_LLCP_CO server channel
Aug 21 20:38:48 nyx neard[3043]: neard[3043]: VALIDATION_LLCP_CL client channel closed
Aug 21 20:38:48 nyx neard[3043]: neard[3043]: Error with VALIDATION_SNEP server channel
Aug 21 20:38:48 nyx neard[3043]: neard[3043]: Error with SNEP server channel
Aug 21 20:38:48 nyx neard[3043]: neard[3043]: Error with NPP server channel
Aug 21 20:38:48 nyx neard[3043]: Error with Handover server channel
Aug 21 20:38:48 nyx neard[3043]: Error with VALIDATION_LLCP_CO server channel
Aug 21 20:38:48 nyx neard[3043]: VALIDATION_LLCP_CL client channel closed
Aug 21 20:38:48 nyx neard[3043]: Error with VALIDATION_SNEP server channel
Aug 21 20:38:48 nyx neard[3043]: Error with SNEP server channel
Aug 21 20:38:48 nyx neard[3043]: Error with NPP server channel
```

So neard integration doesn't seem to be complete here. Also neard does not seem to be too happy with bluez here.

```
# journalctl - u neard
Aug 21 20:33:49 nyx systemd[1]: Started neard service.
Aug 21 20:33:49 nyx neard[3043]: neard[3043]: NEAR daemon version 0.16
Aug 21 20:33:49 nyx neard[3043]: NEAR daemon version 0.16
Aug 21 20:33:51 nyx neard[3043]: neard[3043]: Could not get Bluetooth default adapter Method "DefaultAdapter" with signature "" on interface "org.bluez.Manager" doesn't exist
Aug 21 20:33:51 nyx neard[3043]: Could not get Bluetooth default adapter Method "DefaultAdapter" with signature "" on interface "org.bluez.Manager" doesn't exist
```

KawaiDesu, 2019-08-22:
One more report here.
Patched stock Archlinux kernel `5.2.9-arch1-1-ARCH` using `nfc-nxp-nci-v4.patch`.
```
➜ sudo dmidecode -qH 0x000C
System Information
	Manufacturer: LENOVO
	Product Name: 20KH006JRT
	Version: ThinkPad X1 Carbon 6th
	Serial Number: PF198ZX1
	UUID: <deleted>
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_20KH_BU_Think_FM_ThinkPad X1 Carbon 6th
	Family: ThinkPad X1 Carbon 6th
```
Now I can see NFC available (before patch there wasn't):
```
➜ rfkill list nfc
2: nfc0: NFC
	Soft blocked: no
	Hard blocked: no

➜ nfctool -l
nfc0:
          Tags: [ ]
          Devices: [ ]
          Protocols: [ Felica MIFARE Jewel ISO-DEP NFC-DEP ]
          Powered: No
          RF Mode: None
          lto: 0
          rw: 0
          miux: 0
```
However, libnfc can't see it:
```
➜ nfc-list
nfc-list uses libnfc 1.7.1
No NFC device found.
```
I can power it on and even get some communication with smartphone or read a card:
```
➜ nfctool --enable --device=nfc0
nfc0:
          Tags: [ ]
          Devices: [ ]
          Protocols: [ Felica MIFARE Jewel ISO-DEP NFC-DEP ]
          Powered: Yes
          RF Mode: None
          lto: 0
          rw: 0
          miux: 0

# Read city transport card (MifareClassic, 4k size)
➜ nfctool --device=nfc0 --poll=Both --sniff --dump-symm
Start sniffer on nfc0

Start polling on nfc0 as both initiator and target

Targets found for nfc0
  Tags: [ tag7 ]
  Devices: [ ]

```
But after that it stopped working. There is just no reaction to card or smartphone.  
On the first tests restarting nfctool fixed that problem. But when I repeated tests for this report I got more serious problem:
```
# This is near first tests
[ 3007.514847] llcp: nfc_llcp_recv: err -5
[ 3009.522266] llcp: nfc_llcp_symm_timer: SYMM timeout
[ 3143.245534] llcp: nfc_llcp_recv: err -5
[ 3145.202244] llcp: nfc_llcp_symm_timer: SYMM timeout
[ 3201.777257] llcp: nfc_llcp_recv: err -5
[ 3203.655348] llcp: nfc_llcp_symm_timer: SYMM timeout
# This is last test
[12999.703613] BUG: kernel NULL pointer dereference, address: 0000000000000000
[12999.703618] #PF: supervisor write access in kernel mode
[12999.703620] #PF: error_code(0x0002) - not-present page
[12999.703622] PGD 0 P4D 0 
[12999.703625] Oops: 0002 [#1] PREEMPT SMP PTI
[12999.703628] CPU: 4 PID: 651 Comm: irq/159-nxp-nci Tainted: G           OE     5.2.9-arch1-1-ARCH #1
[12999.703630] Hardware name: LENOVO 20KH006JRT/20KH006JRT, BIOS N23ET55W (1.30 ) 08/31/2018
<cut>
[12999.703653] Call Trace:
[12999.703664]  nci_recv_frame+0x36/0x90 [nci]
[12999.703669]  nxp_nci_i2c_irq_thread_fn+0xd1/0x285 [nxp_nci_i2c]
[12999.703674]  ? irq_forced_thread_fn+0x80/0x80
[12999.703676]  irq_thread_fn+0x20/0x60
[12999.703678]  irq_thread+0xee/0x180
[12999.703681]  ? wake_threads_waitq+0x30/0x30
[12999.703684]  kthread+0xfb/0x130
[12999.703686]  ? irq_thread_dtor+0x90/0x90
[12999.703688]  ? kthread_park+0x80/0x80
[12999.703692]  ret_from_fork+0x35/0x40
<cut>
[12999.703788] note: irq/159-nxp-nci[651] exited with preempt_count 1
[12999.703797] BUG: unable to handle page fault for address: ffff960f4232976e
[12999.703799] #PF: supervisor read access in kernel mode
[12999.703800] #PF: error_code(0x0000) - not-present page
[12999.703801] PGD 44213a067 P4D 44213a067 PUD 44213b067 PMD 43e087067 PTE 0
[12999.703805] Oops: 0000 [#2] PREEMPT SMP PTI
[12999.703807] CPU: 4 PID: 651 Comm: irq/159-nxp-nci Tainted: G      D    OE     5.2.9-arch1-1-ARCH #1
[12999.703809] Hardware name: LENOVO 20KH006JRT/20KH006JRT, BIOS N23ET55W (1.30 ) 08/31/2018
[12999.703812] RIP: 0010:exit_thread+0x12/0xb0
<cut>
[12999.703828] Call Trace:
[12999.703832]  do_exit+0x307/0xb50
[12999.703836]  ? task_work_run+0x93/0xb0
[12999.703838]  ? do_exit+0x2ff/0xb50
[12999.703841]  ? kthread+0xfb/0x130
[12999.703845]  ? rewind_stack_do_exit+0x17/0x20
<cut>
[12999.703909] Fixing recursive fault but reboot is needed!
[13030.425914] nci: __nci_request: wait_for_completion_interruptible_timeout failed 0
[13069.582342] nci: __nci_request: wait_for_completion_interruptible_timeout failed 0
```
System feels bad right now (nfctool, firefox, htop are stuck) so I'm rebooting, copypasting this text to notepad...

dwmw2, 2019-08-22:
I don't even get that far.
```
System Information
	Manufacturer: LENOVO
	Product Name: 20L9S0TW00
	Version: ThinkPad T580
	Serial Number: R90RM6LV
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_20L9_BU_Think_FM_ThinkPad T580
	Family: ThinkPad T580

```
```
$ nfctool --device=nfc0 --poll=Both --sniff --dump-symm
Start sniffer on nfc0

Start polling on nfc0 as both initiator and target

```

At that point I get
```
[  565.414661] nci: nci_ntf_packet: unknown ntf opcode 0x0
```
And no communication seems to work. Hit Ctrl-C, run it again and it says:
```
$ nfctool --device=nfc0 --poll=Both --sniff --dump-symm
Start sniffer on nfc0

Protocol error
```
```
[  663.897627] nci: nci_start_poll: failed to set local general bytes
```

dileks, 2019-08-22:
Thanks for the reports!

Unsure if user-space daemons/tools like `neard{-tools}` and libs like `libnfc{-nci}` needs to be adapted for newer Linux-kernels.

Someone happen to know something about firmware required?

Just looked through my diffs when doing 1st steps with the nxp-pn5xx driver and libnfc-nci from NXP.

Cannot say below was really required as I haven't the logs anymore.

```diff
--- /opt/libnfc-nci/etc/libnfc-nxp-init.conf    2019-05-10 22:54:09.352360774 +0200
+++ /etc/libnfc-nxp-init.conf   2019-05-13 10:22:07.557628486 +0200
@@ -7,25 +7,25 @@
 ###############################################################################
 # Logging Levels. Suggested value for debugging is 0x03.
 #  NXPLOG_GLOBAL_LOGLEVEL - Configuration for Global logging level
-#  NXPLOG_EXTNS_LOGLEVEL - Configuration for extns logging level
+#  NXPLOG_EXTNS_LOGLEVEL  - Configuration for extns logging level
 #  NXPLOG_NCIHAL_LOGLEVEL - Configuration for enabling logging of HAL
-#  NXPLOG_NCIX_LOGLEVEL - Configuration for enabling logging of NCI TX packets
-#  NXPLOG_NCIR_LOGLEVEL - Configuration for enabling logging of NCI RX packets
+#  NXPLOG_NCIX_LOGLEVEL   - Configuration for enabling logging of NCI TX packets
+#  NXPLOG_NCIR_LOGLEVEL   - Configuration for enabling logging of NCI RX packets
 #  NXPLOG_FWDNLD_LOGLEVEL - Configuration for enabling logging of FW download functionality
-#  NXPLOG_TML_LOGLEVEL - Configuration for enabling logging of TML
+#  NXPLOG_TML_LOGLEVEL    - Configuration for enabling logging of TML
 # Logging Levels
 #  NXPLOG_DEFAULT_LOGLEVEL    0x01
 #  NXPLOG_DEBUG_LOGLEVEL      0x03
 #  NXPLOG_WARN_LOGLEVEL       0x02
 #  NXPLOG_ERROR_LOGLEVEL      0x01
 #  NXPLOG_SILENT_LOGLEVEL     0x00
-NXPLOG_GLOBAL_LOGLEVEL=0x00
-NXPLOG_EXTNS_LOGLEVEL=0x00
-NXPLOG_NCIHAL_LOGLEVEL=0x00
-NXPLOG_NCIX_LOGLEVEL=0x00
-NXPLOG_NCIR_LOGLEVEL=0x00
-NXPLOG_FWDNLD_LOGLEVEL=0x00
-NXPLOG_TML_LOGLEVEL=0x00
+NXPLOG_GLOBAL_LOGLEVEL=0x03
+NXPLOG_EXTNS_LOGLEVEL=0x03
+NXPLOG_NCIHAL_LOGLEVEL=0x03
+NXPLOG_NCIX_LOGLEVEL=0x03
+NXPLOG_NCIR_LOGLEVEL=0x03
+NXPLOG_FWDNLD_LOGLEVEL=0x03
+NXPLOG_TML_LOGLEVEL=0x03
 
 ###############################################################################
 # NXP HW Device Node information, when pn5xx_i2c kernel driver configuration is used
```

```diff
--- /opt/libnfc-nci/etc/libnfc-nxp-pn547.conf   2019-05-10 22:54:09.352360774 +0200
+++ /etc/libnfc-nxp-pn547.conf  2019-08-22 16:56:53.657566119 +0200
@@ -79,5 +79,9 @@ NXP_NFC_PROPRIETARY_CFG={05:FF:FF:06:8A:
 # The value is set to 3 by default as it assumes we will discover 0xF2,
 # 0xF3, and 0xF4. If a platform will exclude and SE, this value can be reduced
 # so that the stack will not wait any longer than necessary.
-NXP_NFC_MAX_EE_SUPPORTED=0x00
+##NXP_NFC_MAX_EE_SUPPORTED=0x00
+NXP_NFC_MAX_EE_SUPPORTED=0x03
 
+# Firmware file from NXP NFC Driver for Windows 10 (Version 1803) v12.0.4.0
+NXP_NFC_FW_PATH="/lib/firmware/nxp"
+NXP_NFC_FW_NAME="NXPPN547FW.DAT"
```

```diff
--- /opt/libnfc-nci/etc/libnfc-nxp-pn548.conf   2019-05-10 22:54:09.352360774 +0200
+++ /etc/libnfc-nxp-pn548.conf  2019-08-22 16:59:54.098284278 +0200
@@ -120,4 +120,9 @@ NXP_RF_CONF_BLK_2={ 20, 02, 15, 01,
 # The value is set to 3 by default as it assumes we will discover 0xF2,
 # 0xF3, and 0xF4. If a platform will exclude and SE, this value can be reduced
 # so that the stack will not wait any longer than necessary.
-NXP_NFC_MAX_EE_SUPPORTED=0x00
+##NXP_NFC_MAX_EE_SUPPORTED=0x00
+NXP_NFC_MAX_EE_SUPPORTED=0x03
+
+# Firmware file from NXP NFC Driver for Windows 10 (Version 1803) v12.0.4.0
+NXP_NFC_FW_PATH="/lib/firmware/nxp"
+NXP_NFC_FW_NAME="NXPPN548FW.DAT"
```

That was OK:
```
root@iniza:/opt/libnfc-nci/sbin# ./nfcDemoApp poll
```

To quote myself:
>Before this I expermented with jr64 nxp-pn5xx (acpi support) and libnfc-nci from NXP and had sucess with running the demo-app. I saw in the logs:

>NxpHal: phNxpNciHal_deriveChipType Hw Version (0x68) Not Found: Setting Default Chip Type : PN548C2
>NxpHal: phNxpNciHal_deriveChipType NxpNci > Product : PN548C2

hsanjuan, 2019-10-28:
Have the kernel patches here been upstreamed (I saw something on 5.3.6). With the X1 Carbon 7th Gen I cannot get the reader detected on 5.3.7.

dileks, 2019-10-29:
The relevant patchset got in with Linux v5.4-rc1. As they are no bug-fixes for stable Linux releases you have to get (git cherrypick) them from Linus upstream.

mbarisa, 2019-12-03:
linux 5.4 is now in core, I still can't get nfc reader working.

ikelos, 2019-12-16:
Hiya, I'm on kernel 5.4.3 (which definitely has the patchset changes, I've verified they're there manually) on a Lenovo T490.  The following seem to be automatically loaded at boot:
```
Module                  Size  Used by
nxp_nci_i2c            16384  0
nxp_nci                16384  1 nxp_nci_i2c
nci                    45056  2 nxp_nci,nxp_nci_i2c
nfc                    86016  2 nci,nxp_nci
```
which suggests the device is being automatically detected, but I can't get nfctool, nfc-list or rfkill to list an adaptor, and `nfctool --enable -d nfc0` gives `no such device`, so I'm not sure if I have all the necessary i2c modules loaded?

I have no unusual i2c devices listed under `i2cdetect -l` (notably, no Synopsys Designware entry):
```
i2c-3	unknown   	i915 gmbus dpd                  	N/A
i2c-1	unknown   	i915 gmbus dpc                  	N/A
i2c-6	unknown   	DPDDC-C                         	N/A
i2c-4	unknown   	DPDDC-A                         	N/A
i2c-2	unknown   	i915 gmbus misc                 	N/A
i2c-0	unknown   	i915 gmbus dpb                  	N/A
i2c-7	unknown   	SMBus I801 adapter at efa0      	N/A
i2c-5	unknown   	DPDDC-B                         	N/A
```
Only one contains an entry for 0x28, and it has entries for every value.  It's also DPDDC-A which according to above comments is graphics related:
```
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-4.
I will probe address range 0x03-0x77.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77   
```
The NXP-NCI demo app didn't work with the default configuration, but I also couldn't verify the demo app from the `libnfc-nci` project, but I wasn't sure what address or i2c bus to connect to, so that may not be surprising.

My DSDT also only lists 0x0000 for the various pin lists.

I'd very much like to help debug and get this working for everyone, since it seems a waste to have the hardware but not be able to use it.  Please let me know if there's any additional information I can provide, or further testing I can run to try to figure out the issue...

hsanjuan, 2019-12-28:
My NFC reader started working after upgrading the Synaptics/Fingerprint reader firmware (Prometheus). This is in lvfs-testing which needs to be enabled for fwupdmgr. After upgrading, the reader is now detected by nfctool (Kernel 5.4.6).

hsanjuan, 2019-12-30:
More updates on my side (Lenovo X1 Carbon Gen 7). tl;dr: linux_libnfc-nci works!

The ACPI DSDT table is similar to the one posted above but addresses change:

```
        Device (NFC1)
        {
            Name (_ADR, Zero)  // _ADR: Address
            Name (_HID, "NXP1001")  // _HID: Hardware ID
            Name (_DDN, "NXP NFC For Win10")  // _DDN: DOS Device Name
            Name (_UID, One)  // _UID: Unique ID
            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                Name (RBUF, ResourceTemplate ()
                {
                    I2cSerialBusV2 (0x0029, ControllerInitiated, 0x00061A80,
                        AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                        0x00, ResourceConsumer, , Exclusive,
                        )
                    GpioInt (Level, ActiveHigh, Exclusive, PullNone, 0x0000,
                        "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                        )
                        {   // Pin list
                            0x012A
                        }
                    GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                        "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                        )
                        {   // Pin list
                            0x002F
                        }
                    GpioIo (Exclusive, PullDefault, 0x0000, 0x0000, IoRestrictionOutputOnly,
                        "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                        )
                        {   // Pin list
                            0x0124
                        }
                })
                Return (RBUF) /* \_SB_.PCI0.I2C0.NFC1._CRS.RBUF */
            }
```

i2cdetect -l:

```
i2c-3   i2c             i915 gmbus dpd                          I2C adapter
i2c-1   i2c             i915 gmbus dpc                          I2C adapter
i2c-8   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-6   i2c             DPDDC-C                                 I2C adapter
i2c-4   i2c             DPDDC-A                                 I2C adapter
i2c-2   i2c             i915 gmbus misc                         I2C adapter
i2c-0   i2c             i915 gmbus dpb                          I2C adapter
i2c-9   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-7   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-5   i2c             DPDDC-B                                 I2C adapter
```

This gives i2c-7 and i2c-8 as best candidates. But I also see `/sys/bus/devices/i2c-NXP1001:00` which is a symlink to `/sys/bus/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-7/i2c-NXP1001:00` so I guessed i2c-7 is the right bus.

I grabbed `linux_libnfc-nci` and compiled it (`./configure --prefix=/usr --sysconfdir=/etc --libdir=/usr/lib64 --enable-alt`) with the following in `src/halimpl/pn54x/tml/i2c/phTmlNfc_alt.h`:

```
#define I2C_BUS         "/dev/i2c-7"
#define I2C_ADDRESS     0x29
#define PIN_INT         200+0x12a
#define PIN_ENABLE      200+0x124
```

Where the bus is the one determined above, the I2C address comes from the DSDT, 200 is the offset from `/sys/class/gpio/gpiochip200/base` and the pins are from the DSDT. 

Some people above reported some success at this point but my `nfcDemoApp poll` failed with `Cannot select I2C address (Device or resource busy)` (to see this I needed to enable debugging in `/etc/libnfc-nxp-init.conf`. I also noticed in the logs that it created pin gpio498 (PIN_INT) fine but it actually did not quite manage to create pin 492 (PIN_ENABLE) inside `/sys/class/gpio/`. If I manually ran `echo 492 > /sys/class/gpio/export` I also got `echo: write error: Device or resource busy`. 

Doing `cat /sys/kernel/debug/gpio` actually showed that gpio492 is already registered. So someone was using it. It would seem this was the kernel module `nxp_nci_i2c`. I did `modprobe -r nxp_nci_i2c` and gpio492 got unregistered and the nfcDemoApp started working from that point. However, the nfc0 device dissapeared from rkill and neard/nfctool device list from that point.

I wonder if there is a way to get both a libnfc-nci program and rfkill co-exist.  Pointers from here are appreciated!

crosser, 2020-01-07:
> For anyone interested, I just uploaded the first version of my modified kernel driver: https://github.com/jr64/nxp-pn5xx

@jr64 first, I must report that the driver works on Thinkpad X1 Carbon Gen 7. Thanks for your work!
I have a couple of questions:
1. Are there plans to upstream it (or is it already upstream)? I tested with 5.3.0, the kernel pn544 and nxp-nci_i2c do not work there.
2. How to make smartcard tools (gnupg's scdaemon, pcscd and likes) see the reader?

hsanjuan, 2020-01-09:
@crosser just two messages above I reported it working on 5.4.6 because it was upstreamed already.

crosser, 2020-01-09:
@hsanjuan Thanks! From the comment, it was not clear to me that you were using upstream module.

ikelos, 2020-01-10:
@hsanjuan Can you please provide a few more details?  I got my fingerprint reader upgraded to the firmware from lvfs-testing (although fwupdmgr now isn't displaying it), and I'm on 5.4.10.  I've got all the in-kernel nfc adaptors compiled as modules for testing, but can't find one that actually produces any dmesg output or anything that shows up in libnfc's nfc-list.  As best I can tell, my device isn't accessible directly over I2C, but shows up in /sys as `:acpi:NXP1001:`.

Exactly which driver do you have loaded?  Is it:
* the official NXP one
* the unofficial nxp-pn5xx
* the kernel nxp_nci_i2c 
* the kernel pn544
* some other module

hsanjuan, 2020-01-10:
> @hsanjuan Can you please provide a few more details? I got my fingerprint reader upgraded to the firmware from lvfs-testing (although fwupdmgr now isn't displaying it), and I'm on 5.4.10. I've got [...]

Does `ncf0` show on the rfkill list or the `nfctool -l` list ?

My nfc/nxp/nci related modules:

```
nxp_nci_i2c            20480  0
nxp_nci                16384  1 nxp_nci_i2c
nci                    81920  2 nxp_nci,nxp_nci_i2c
nfc                   139264  14 nci,nxp_nci
rfkill                 28672  8 nfc,bluetooth,thinkpad_acpi,cfg80211
```

They all get loaded automatically, I did nothing special here and did not compile any extra modules. For reasons discussed above, libnfc tooling does not see it, but I could compile `linux_libnfc-nci` and make it see the reader (but I had to unload nxp_nci_i2c first). There's a single line in dmesg: `nfc: nfc_init: NFC Core ver 0.1`.

Mind that NFC needs to be enabled in the UEFI config (I had it disabled because it broke the touchpad before).

dileks, 2020-01-10:
Happy new 2020.

Wow, so many feedbacks :-).

Unfortunately, I have no more the Lenovo T470 notebook as I left the company.

Thanks for the hint with upgrading to a newer firmware via LVFS/testing.
Unfortunately, I had not that much luck getting LVFS running in sense of being able to upgrade any firmware specific stuff This was with Debian/testing AMD64. But that might be better supported theses days.
There was more firmware stuff available via LVFS than offered on the Lenovo support website (Windows binaries). 

My last feedback from the Linux-kernel maintainers: The userspace tools seem to be a culprit to communicate with NFC device.

Happy testing and bonne chance folks.

Regards,
- Sedat -

ikelos, 2020-01-10:
@hsanjuan Ok, so we must have different configurations (mine's detailed higher up).  I do get those modules automatically loaded, but `rfkill list` does not mention NFC.  I verified that NFC is enabled under `security > I/O ports` in my bios.  I also get the `NFC core` line, but if that's the same as you then it's not all that informative.   My rfkill list only returns the following:
```
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: tpacpi_bluetooth_sw: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
```
I haven't tried any non-kernel modules yet, so I haven't yet had the hardware working at all.  What concerns me most is that I can't find the appropriate i2c bus in the list `i2cdetect -l` returns (see above for my full listing), and other than autoloading the nxp_nci_i2c module, there's no other mention of it in relation to i2c, only to acpi (which nxp-pn5xx mentions it should support).  I guess trying the official NXP module or nxp-pn5xx will be my next step.  Thanks for the clarification!  5:)

hsanjuan, 2020-01-10:
@ikelos hmm, I'm on the Thinkpad X1 Carbon Gen 7.

While `i2cdetect -l` shows the Synopsis buses, i2cdetect is not able to find any devices when scanning them though, so it is of little help. I guess your device does not appear on `/sys/bus/devices/i2c-NXP1001:00` either? And what does your acpi DSDT table show?

pbirkants, 2020-01-10:
Hi!

I tried replicating the steps described by @hsanjuan on different hardware - Thinkpad X390 Yoga.

After compiling 5.4.8 kernel with upstream nxp_nci_i2c module enabled (current Debian default config does not include NCI modules) and adjusting the pin offsets according to my DSDT table, I could successfully run the various `nfcDemoApp` modes from `linux_libnfc-nci` without any extra steps.

The device also shows up in `rfkill list` and `nfctool -l`, but is not detected by `nfc-scan-device -v -i`. `nfctool --enable -d nfc0` quits with a timeout, `NFC: Read failed with error -121` (`EREMOTEIO`) is output in kernel log.

I looked around the sources for `nfctool` and the kernel drivers for this error, and it seems to me that powering on the reader fails.

ikelos, 2020-01-10:
@hsanjuan Hmmm, interesting to know the devices don't show up in i2cdetect, but given I don't have the buses it still suggests there's a difference.

Correct, there's no NXP1001:00 under i2c, only under acpi.

For the DSDT table, I tried to follow the instructions in the comment above:
```
plasma ~ # acpidump --output dsdt.aml           
Wrong checksum for FADT!

plasma ~ # acpixtract -a dsdt.aml    

Intel ACPI Component Architecture
ACPI Binary Table Extraction Utility version 20190703
Copyright (c) 2000 - 2019 Intel Corporation

  DSDT -  159706 bytes written (0x00026FDA) - dsdt.dat
  FACS -      64 bytes written (0x00000040) - facs.dat
  FACP -     244 bytes written (0x000000F4) - facp.dat
  SSDT -    6950 bytes written (0x00001B26) - ssdt1.dat
  SSDT -    1389 bytes written (0x0000056D) - ssdt2.dat
  SSDT -   15159 bytes written (0x00003B37) - ssdt3.dat
  SSDT -   12647 bytes written (0x00003167) - ssdt4.dat
  SSDT -    1554 bytes written (0x00000612) - ssdt5.dat
  TPM2 -      52 bytes written (0x00000034) - tpm2.dat
  UEFI -      66 bytes written (0x00000042) - uefi1.dat
  SSDT -    1328 bytes written (0x00000530) - ssdt6.dat
  HPET -      56 bytes written (0x00000038) - hpet.dat
  APIC -     300 bytes written (0x0000012C) - apic.dat
  MCFG -      60 bytes written (0x0000003C) - mcfg.dat
  ECDT -      83 bytes written (0x00000053) - ecdt.dat
  SSDT -    6802 bytes written (0x00001A92) - ssdt7.dat
  SSDT -    8949 bytes written (0x000022F5) - ssdt8.dat
  BOOT -      40 bytes written (0x00000028) - boot.dat
  SSDT -    3326 bytes written (0x00000CFE) - ssdt9.dat
  LPIT -     148 bytes written (0x00000094) - lpit.dat
  WSMT -      40 bytes written (0x00000028) - wsmt.dat
  SSDT -    5735 bytes written (0x00001667) - ssdt10.dat
  DBGP -      52 bytes written (0x00000034) - dbgp.dat
  DBG2 -      84 bytes written (0x00000054) - dbg2.dat
  MSDM -      85 bytes written (0x00000055) - msdm.dat
  BATB -      74 bytes written (0x0000004A) - batb.dat
  DMAR -     168 bytes written (0x000000A8) - dmar.dat
  NHLT -      45 bytes written (0x0000002D) - nhlt.dat
  FPDT -      68 bytes written (0x00000044) - fpdt.dat
  BGRT -      56 bytes written (0x00000038) - bgrt.dat
  UEFI -     298 bytes written (0x0000012A) - uefi2.dat
  XSDT -     268 bytes written (0x0000010C) - xsdt.dat
  RSDP -      36 bytes written (0x00000024) - rsdp.dat
  SSDT -    2736 bytes written (0x00000AB0) - ssdt11.dat
  SSDT -    1532 bytes written (0x000005FC) - ssdt12.dat
  SSDT -     329 bytes written (0x00000149) - ssdt13.dat
  SSDT -    1024 bytes written (0x00000400) - ssdt14.dat
  SSDT -     778 bytes written (0x0000030A) - ssdt15.dat
  SSDT -     791 bytes written (0x00000317) - ssdt16.dat
  SSDT -    1828 bytes written (0x00000724) - ssdt17.dat
  SSDT -    1470 bytes written (0x000005BE) - ssdt18.dat

plasma ~ # iasl -d dsdt.dat

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20190703
Copyright (c) 2000 - 2019 Intel Corporation

File appears to be binary: found 46389 non-ASCII characters, disassembling
Binary file appears to be a valid ACPI table, disassembling
Input file dsdt.dat, Length 0x26FDA (159706) bytes
ACPI: DSDT 0x0000000000000000 026FDA (v02 LENOVO CFL      20170001 INTL 20160422)
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)
[... 67 lines cut ...]
                    CreateWordField (RBUF, 0x60, NFEQ)
                    CreateWordField (RBUF, 0x88, NFON)
                    NFIP = GNUM (0x0404000A)
                    NFEQ = GNUM (0x0401000F)
                    NFON = GNUM (0x04040005)
                    Return (RBUF) /* \_SB_.PCI0.I2C0.NFC1._CRS.RBUF */
                }

                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    If ((\OSYS >= 0x07DF))
                    {
                        If (((\_SB.GGIV (0x04030016) == 0x00) && (\NFCF == 0x00)))
                        {
                            Return (0x0F)
                        }
                    }

                    Return (0x00)
                }
            }
        }
    }
```
I guess the pin lists being 0x0000 is also a problem.  I'm not sure where to go from here, any pointers would be greatly appreciated...

dileks, 2020-01-11:
Thanks @pbirkants 

Just to clarify:
The default Debian/unstable Kernel v5.4.x does not ship the required kernel-modules and options. This is currently linux-image-5.4.0-2-amd64 (5.4.8-1).

Please, open a bug-report in Debian's BTS [1] that required kernel-modules and kernel-configs get enabled.

I would have said please concentrate on the Linux-upstream work and contact the appropriate maintainers [2], but Linux-NFC maintenance is currently marked "orphan" [3].

I am still confused which user-space tools is needed to get stuff run.

[1] https://www.debian.org/Bugs/
[2] https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/MAINTAINERS
[3] https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/MAINTAINERS#n11627

[36 more replies]