nfc-tools/libnfc #456: nfc_initiator_transceive_bytes 14443B stop to work after new enum NMT_BARCODE

etmatrix, 2017-10-11:
I have a simple program that communicates with custom tag 14443B.
Today I tried lastest version and my program stop to work, i got the error "Invalid argument(s)" when I send byte with nfc_initiator_transceive_bytes.
After some tests I discovered the issue is the new item NMT_BARCODE in enum nfc_modulation_type of file include/nfc/nfc-types.h
The issue is the position, when I move NMT_BARCODE to last position after NMT_DEP or after NMT_ISO14443B my program work without problem.
Why the new item NMT_BARCODE is in that position?

doegox, 2017-10-11:
Please check how you include nfc-types.h when you compile your own program.
It sounds to me that you're compiling it against an old version of the libnfc headers.
Check your /usr/include/nfc/nfc-types.h or /usr/local/include/nfc/nfc-types.h

doegox, 2017-10-11:
about the position it's because it's a new TypeA

etmatrix, 2017-10-11:
my fault, the include path was wrong.

closed 2017-10-11 by etmatrix